verilog-to-routing/vtr-verilog-to-routing #979: ODIN_II: No need to check order when instantiating by name

j-b-1-7, 2019-09-23:
Related Issue #848 

#### Expected Behaviour
Odin should not need to check the order of ports that are instantiated by name when instantiating a module

#### Current Behaviour
Odin fails with a netlist error because the ports are out of order
```
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file ../../../Odin-issues/port_order/top.v to parse list
Optimizing module by AST based optimizations
Converting AST into a Netlist. Note this netlist can be viewed using GraphViz (see documentation)
==========================
Detected Top Level Module: 	top
==========================
--------------
Odin has decided you have failed ;)

ERROR (1)::NETLIST_ERROR ../../../Odin-issues/port_order/top.v::4 This module entry does not match up correctly (in2 != out).  Odin expects the order of ports to be the same
output out;
Assertion failed()@[/home/jbrown17/workspace/vtr-verilog-to-routing/ODIN_II/SRC/netlist_create_from_ast.cpp]connect_module_instantiation_and_alias::2328 
[1]    7735 abort (core dumped)  ./odin_II -V ../../../Odin-issues/port_order/top.v
```

#### Possible Solution
No need to check port order if the ports have been instantiated by name

#### Steps to Reproduce
Run Odin II with the verilog code provided by @MohamedElgammal
[top.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3644412/top.v.txt)
Original Repo: https://github.com/MohamedElgammal/Odin-issues/tree/master/port_order
```
./odin_II -V ../../../Odin-issues/port_order/top.v 
```

#### Your Environment
##### uname -a
```
Linux JBpc 5.2.9-arch1-1-ARCH #1 SMP PREEMPT Fri Aug 16 11:29:43 UTC 2019 x86_64 GNU/Linux
```
##### gcc -v
```
gcc version 9.1.0 (GCC) 
```

jeanlego, 2020-03-31:
@Shreyv is this resolved with #1229?

Shreyv, 2020-03-31:
@jeanlego  Yes this one gets resolved with #1229 1229

closed 2020-03-31 by jeanlego