verilog-to-routing/vtr-verilog-to-routing #980: ODIN_II: Segfault when connecting output port to vector element

j-b-1-7, 2019-09-23:
Related Issue #848 

#### Expected Behaviour
Should be able to connect output port to vector element.
If this is not possible, Odin II should at least provide a meaningful error or warning message to help the developer to debug their code.

#### Current Behaviour
Odin II exits with a segfault, with no meaningful error or warning message

#### Steps to Reproduce
Run Odin II with the verilog file provided by @MohamedElgammal
[top.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3644440/top.v.txt)
Original Repo: https://github.com/MohamedElgammal/Odin-issues/tree/master/seg_fault

```
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file ../../../Odin-issues/seg_fault/top.v to parse list
[1]    8722 segmentation fault (core dumped)  ./odin_II -V ../../../Odin-issues/seg_fault/top.v

```
##### Workaround
wire temp1;
wire temp2;
assign out[0] = temp1;
assign out[1] = temp2;

and(temp1, in1[0],in2[0]);
and(temp2, in1[1],in2[1]);
```
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file ../../../Odin-issues/seg_fault/top.v to parse list
Optimizing module by AST based optimizations
Converting AST into a Netlist. Note this netlist can be viewed using GraphViz (see documentation)
==========================
Detected Top Level Module: 	and2
==========================
Performing Optimizations of the Netlist
Performing Partial Map to target device
Outputting the netlist to the specified output format
Successful High-level synthesis by Odin
	Blif file available at default_out.blif
	Ran in 1.3ms
--------------------------------------------------------------------
--------------------------------------------------------------------
Odin ran with exit status: 0
Odin II took 0.00 seconds (max_rss 4.5 MiB)
```

#### Your Environment
##### uname -a
```
Linux JBpc 5.2.9-arch1-1-ARCH #1 SMP PREEMPT Fri Aug 16 11:29:43 UTC 2019 x86_64 GNU/Linux
```
##### gcc -v
```
gcc version 9.1.0 (GCC) 
```

djns99, 2020-03-26:
Ill pick this up

jeanlego, 2020-03-26:
> Ill pick this up

Make sure to: make a WIP PR asap to indicate such. It'll make it easier for us to help you also

PR #1238 (2020-03-30): Fixed Segfault for reference types
Fixes when creating gates with reference types with no identifier name.

#### Related Issue
Fixes #980 

#### How Has This Been Tested?
Added tests for basic vector reference type using both multiple and single gate input types

#### Types of changes
- [x] Bug fix (change which fixes an issue)

#### Checklist:
- [x] I have added tests to cover my changes
- [x] All new and existing tests passed
closed 2020-03-30 by jeanlego