verilog-to-routing/vtr-verilog-to-routing #987: Declaration of wires direction (in/out)

sdamghan, 2019-09-30:
The Odin doesn't recognize the input/output direction of the wires when they defined after the module declaration.

#### Expected Behaviour
The Odin should accept both declaration type whether defining their types in the module name or defining after it separately.

#### Current Behaviour
When the wires' direction define after the module declaration, Odin detects them as twice definition.

#### Possible Solution
Write code for bison part in the way that the declaration of wires' direction as input or output can be detected even after the module declaration.

#### Steps to Reproduce
1. Run Odin by the Verilog codes provided by Mor1kx
  ```
   ERROR (13117)::PARSE_ERROR ../../mor1kx/rtl/verilog/mor1kx_pic.v::28 Module already has input with this name clk
   input clk;

  ```

#### Workaround
  ```
  // works correctly
  module temp (
  input clk
  );
----
  // Odin recognizes it as twice declaration
  module temp (
  clk
  );
  input clk;
  ```

#### Your Environment

* uname -a:
  ```
  Linux casa51 5.0.0-29-generic #31~18.04.1-Ubuntu SMP Thu Sep 12 18:29:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
  ```
* gcc -v:
  ```
  gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1) 
  ```

JakeMercer, 2019-10-21:
I decided to take a look at this one; a simple `top.v` does not fail?

`top.v`:
```
module temp (
clk
);
input clk;
endmodule
```

Output:
```
jmercer@symbiflow:~/Development/vtr-verilog-to-routing$ ./build/ODIN_II/odin_II -V top.v 
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file top.v to parse list
Optimizing module by AST based optimizations
Converting AST into a Netlist. Note this netlist can be viewed using GraphViz (see documentation)
==========================
Detected Top Level Module: 	temp
==========================
Performing Optimizations of the Netlist
Performing Partial Map to target device
Outputting the netlist to the specified output format
Successful High-level synthesis by Odin
	Blif file available at default_out.blif
	Ran in 0.7ms
--------------------------------------------------------------------
--------------------------------------------------------------------
Odin ran with exit status: 0
Odin II took 0.00 seconds (max_rss 4.2 MiB)
```

JakeMercer, 2019-10-21:
I also attempted running against [openrisc/mor1kx](https://github.com/openrisc/mor1kx) but I hit other errors:

```
jmercer@symbiflow:~/Development/mor1kx$ ../vtr-verilog-to-routing/build/ODIN_II/odin_II -V rtl/verilog/mor1kx_pic.v 
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file rtl/verilog/mor1kx_pic.v to parse list
Adding file "mor1kx-defines.v" to parse list
Adding file "mor1kx-sprs.v" to parse list
--------------
Odin has decided you MAY fail ... :

PARSE_ERROR rtl/verilog/mor1kx_pic.v::74 error in parsing: (syntax error, unexpected vIF, expecting vENDGENERATE)
      if (OPTION_PIC_TRIGGER=="EDGE") begin : edge_triggered
PARSE_ERROR rtl/verilog/mor1kx_pic.v::99 error in parsing: (syntax error, unexpected vEND)
      end else if (OPTION_PIC_TRIGGER=="LEVEL") begin : level_triggered
PARSE_ERROR rtl/verilog/mor1kx_pic.v::129 error in parsing: (syntax error, unexpected vEND)
	 end
Optimizing module by AST based optimizations
Converting AST into a Netlist. Note this netlist can be viewed using GraphViz (see documentation)
==========================
Detected Top Level Module: 	mor1kx_pic
==========================
ERROR (4)::NETLIST_ERROR rtl/verilog/mor1kx_pic.v::84 Missing declaration of this symbol irqline
         for(irqline=0;irqline<32;irqline=irqline+1)  begin: picgenerate
Assertion failed()@[/home/jmercer/Development/vtr-verilog-to-routing/ODIN_II/SRC/ast_elaborate.cpp]verify_genvars::2690 
Aborted (core dumped)
```

Could you advise what I'm doing wrong and I'll fix the parser?

jeanlego, 2019-10-22:
### Assessment of the bug
by the looks of it, it is not expecting 
``` if (OPTION_PIC_TRIGGER=="EDGE") ...```
and causes the errors following it to happen (unexpected vEND , etc)
The way to fix this one would be to make sure that verilog_bison.y has a state for if statements during generate blocks, more specifically:

``` 
70:   generate  <- this is where the error starts, must be the cause
71:
72:      genvar 	 irqline; <- this does not get stored for some reason
73:
74:      if (OPTION_PIC_TRIGGER=="EDGE") begin : edge_triggered
75:         reg [31:0] irq_unmasked_r;
```

but to my eyes, the generate keyword is not being handled properly and only goes in generate state for one line 
```
genvar 	 irqline
```
and then comes back, it looks like the parser expects a begin type keyword and an end type keyword to keep parsing in the `generate block` state

### Workaround

I would try to put a begin right after generate and an end right before endgenerate since this looks like a bug in the parser states.

try that and if it works, it should be relatively trivial to fix

JakeMercer, 2019-11-02:
A `begin` and an `end` did not solve/improve the issue, parsing still fails.  I have made the following change which allows the parsing to complete but the elaboration of the AST fails when looking up `irqline`.  The below change allows for more than one statement between `generate` and `endgenerate`.

```diff
diff --git a/ODIN_II/SRC/verilog_bison.y b/ODIN_II/SRC/verilog_bison.y
index 37e6e9b28..011141d29 100644
--- a/ODIN_II/SRC/verilog_bison.y
+++ b/ODIN_II/SRC/verilog_bison.y
@@ -115,7 +115,7 @@ int yylex(void);
 %type <node> specify_block list_of_specify_items
 %type <node> c_function_expression_list c_function
 %type <node> initial_block list_of_blocking_assignment
-%type <node> list_of_generate_block_items generate_item generate_block_item generate loop_generate_construct if_generate_construct 
+%type <node> list_of_generate_block_items list_of_generate_items generate_item generate_block_item generate loop_generate_construct if_generate_construct 
 %type <node> case_generate_construct case_generate_item_list case_generate_items generate_block generate_localparam_declaration generate_defparam_declaration
 
 %%
@@ -194,6 +194,11 @@ list_of_generate_block_items:
        | generate_block_item                                                           {$$ = newList(BLOCK, $1, my_yylineno);}
        ;
 
+list_of_generate_items:
+       list_of_generate_items generate_item    {$$ = newList_entry($1, $2);}
+       | generate_item                                                         {$$ = newList(MODULE, $1, my_yylineno);}
+       ;
+
 generate_item:
        localparam_declaration                          {$$ = $1;}
        | net_declaration                                       {$$ = $1;}
@@ -557,7 +562,7 @@ always:
        ;
 
 generate:
-       vGENERATE generate_item vENDGENERATE    {$$ = $2;}
+       vGENERATE list_of_generate_items vENDGENERATE   {$$ = $2;}
        ;
 
 loop_generate_construct:
```

jeanlego, 2019-11-02:
If you wouldn't mind, I think it would be good if you made a fork and did a WIP pull request out of this. Simplify the work and put the code through the regression test

closed 2025-05-16 by github-actions[bot]