verilog-to-routing/vtr-verilog-to-routing #1004: Inconsistent fasm metadata semantics

leon575777642, 2019-10-12:
As specified [here](https://docs.verilogtorouting.org/en/latest/utils/fasm/), the semantics for different FASM metadata are inconsistent. This is not a bug but somewhat confusing. Specifically:

1. `"="` is used to indicate the bit mapping for LUTs, but `":"` is used for MUXes. I understand that the "direction" of the mapping is different: For LUTs, bits are assigned from the right to the left. For MUXes, features to the right are emitted when the connection to the left is selected. But if you change the "direction" of the mapping and make them consistent, one consistent semantic is feasible and less confusing.

2. For `fasm_features` tag, features are separated with spaces. For `fasm_mux` tag, features are separated with commas. The former is more consistent with VPR's format (space-separated list).

mithro, 2019-10-12:
@litghost

leon575777642, 2019-10-15:
@mithro Any updates? I'm also wondering how to express it if the contents of the LUTs should be mapped into multiple features?

litghost, 2019-10-15:
@leon575777642 While I think your observations are valid, in the case of the LUT/param syntax versus the MUX syntax, you have already identified why the syntax is different.  In addition, the LUT/param syntax is 1:1, versus MUX which is 1:many.

As for changing `fasm_mux` to support space separation, I'm not opposed, but it doesn't seem like a priority, as they are of equivalent expressive power.

> I'm also wondering how to express it if the contents of the LUTs should be mapped into multiple features?

Can you give an example?  The interesting thing about FASM databases/definitions are that you control both the VPR side and the database side.

litghost, 2019-10-15:
That is a 1:many relationship, not 1:1. Anyways, when fusing two LUT4 in your proposed arch, do you require another feature?  Because in the underlying hardware there would be another mux.  In 7-series, the F7MUX and F8MUX are those features, and are explicitly instances as black boxes and timing info, etc.

leon575777642, 2019-10-15:
Oh by 1:1 I mean the bit mapping. But if you think of features, then yeah it's 1:many.

As for the underlying hardware, the F7MUX and F8MUX do not need "programming bits". They can be enabled via some `fasm_features` under a `<mode>` tag. Also, I'm designing my own FPGA, so the underlying hardware can be vastly different (for example, the programming bits of a LUT may not be continuous in the configuration memory address space), but I understand it may not be a general use case so I'll do further processing in the database side.

litghost, 2019-10-15:
@leon575777642 I'm open to you submitting a PR that adds support for a different LUT syntax under a new FASM tag.  Your proposal in https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1004#issuecomment-542274835 is a significant enough departure that it probably should just be a new fasm tag type.  If the new syntax implements nicely, we can deprecate the old FASM LUT syntax and migrate to the new one.

mithro, 2019-10-15:
It is not expected that FASM features map directly to the configuration address bits. IE A fasm feature could enable multiple bits in the configuration space.

It is the fasm assembler which takes the fasm features and turns it into an exact bitstream. This could for example mean splatting the LUT bits all over the place like the Xilinx 7 Series does -> https://symbiflow.github.io/prjxray-db/artix7/tile_int_r.html

mithro, 2019-10-15:
Btw you might want to take a look at the bitstream viewing tool at https://github.com/SymbiFlow/symbiflow-bitstream-viewer which should help connect fasm features to actual bits in a bitstream.

leon575777642, 2019-10-15:
Thanks @mithro. I understand the FASM features don't map directly to the configuration address space. I have an assembler that does most of the bitstream generation as well. Not asking for a huge change, just suggesting a possible improvement that makes the syntax more consistent. Supporting LUT slicing is similar to supporting multiple features emitted from one MUX connection -- it's not necessary, but nicer. The current syntax works for me, so I'm happy to stay with it if the changes are a lot of work.

I'm closing this issue for now.

closed 2019-10-15 by leon575777642

mithro, 2019-10-16:
@leon575777642 - it would be awesome to see if it is possible to extend the bitstream viewer tool I linked above to work with your bitstream / fasm too..

leon575777642, 2019-10-17:
Went through the visualization tool briefly, and it seems too complex for me. I'm using fasm metadata in a very naive way: each feature is simply the bit address in the bitstream that should be set to 1, so reading the bits directly is good enough for now. In the long run I might have other configuration memory types, which may need more complex tooling like a visualizer.