verilog-to-routing/vtr-verilog-to-routing #1009: VTR_ENABLE_SANITIZE should support Clang

litghost, 2019-10-17:
#### Proposed Behaviour

VTR_ENABLE_SANITIZE = ON should enable sanitizers on Clang.

#### Current Behaviour

VTR_ENABLE_SANITIZE = ON results in errors around the linker flags.

#### Possible Solution

The CMake functions to enable sanitization on Clang are:
```
add_compile_flags("-fsanitize=address -fsanitize=leak -fsanitize=undefined")
add_link_flags("-fsanitize=address -fsanitize=leak -fsanitize=undefined")
```

#### Context

I was attempting to enable sanitization on VTR when using Clang as the compiler, and it failed to compile.

kmurray, 2019-10-25:
Yep makes sense that VTR_ENABLE_SANITIZE should work for clang too.

I recall there was an issue related to the linker (with some versions of GCC?) which required a GCC specific option to make work. I think backing that out to only be active if GCC was the compiler would fix this.

litghost, 2019-10-25:
> I recall there was an issue related to the linker (with some versions of GCC?) which required a GCC specific option to make work. I think backing that out to only be active if GCC was the compiler would fix this.

Clang requires that the sanitize flags be passed to both the compile and link steps, hence why I listed both `add_compile_flags` and `add_link_flags`.

closed 2025-05-16 by github-actions[bot]