verilog-to-routing/vtr-verilog-to-routing #1014: Placer awareness of "obstructions"

litghost, 2019-10-23:
#### Expected Behaviour

Placer should be aware of and avoid placing near obstructions.

#### Current Behaviour

Placer relies on place delay matrix, which does not account for obstructions.

#### Possible Solution

Unclear

#### Context

On the 7-series arch/routing graph, there are "obstructions" that are not captured with the place delay matrix (which works best in the unobstructed sections of the fabric).  In particular, on the 50T graph CMT's X0Y2, X1Y1 and X1Y0 are fully regular, but X0Y1, X0Y0 and X1Y2 especially have large obstructions.  X0Y1/X0Y0 have the control circuitry, and X1Y2 have the PCIE hardblocks.

The placer delay matrix as a delay abstraction works on the unobstructed portions of the fabric where the delay structure is very regular, however the portions of the fabric with obstructions have lower routablility, and increased delays as the router has to route around the obstructions.

I do **not** have concrete numbers on the affect this have on the placement solution, but I wanted to raise the idea and discuss how the place delay model and/or the place might need to be adapted to handle these features.  In general from first principles I'd expect the placer to avoid the fabric around obstructions if it can, and that does not appear to be the current behavior.

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

mithro, 2019-10-23:
Could you render some diagrams of this to make it easier to understand the obstructions? I believe vpr recently got the ability to dump a grid to an image file?

kmurray, 2019-10-25:
> Could you render some diagrams of this to make it easier to understand the obstructions?

Yes, these would be helpful

> I believe vpr recently got the ability to dump a grid to an image file?

Yes, via the [--save_graphics](https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-save-graphics) option.

kmurray, 2019-10-25:
> but I wanted to raise the idea and discuss how the place delay model and/or the placer might need to be adapted to handle these features

It makes sense that this should have an effect. 

A quick and dirty (i.e. probably not a good long term solution), would be to leverage [placer delay model overrides](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/place_delay_model.h#L82l). Currently they are used to override the delay estimates between particular pairs of pins. You could try leveraging them to override the delays for pairs of pins particularly effected by the routing blockage. However for significant blockages which effect a large number of pins this would likely face scalability/run-time issues (as the override list gets big).

Another potential idea (not sure if its a good one) would be to build a seperate delay matrix for each 'region' (e.g. quadrant, 16th) of the chip, and use the relevant matrix when the source is in that region. This might do a somewhat better job of capturing the regional delay variations these types of blockages introduce.

I don't think either of these is a perfect solution, so we'll need to think about how we can come up with a more general and efficient way to capture these types of behaviour.

closed 2025-05-16 by github-actions[bot]