verilog-to-routing/vtr-verilog-to-routing #1018: Thread safety in core router algorithms

litghost, 2019-10-28:
## Proposed Behaviour

All core router algorithms should have thread safety, by which I mean the following properties:
 - Constant read-only data may be shared between router data structures and invocations.  Things includes things like the rr node definition list and switch list, which is only mutated during rr graph generation.
  - Examples:
   - RR nodes
   - Switch list
   - Non configurable node sets
 - Writable data should either encapsulated in a non-global object and references, or passed as arguments to the relevant functions and or methods:
  - Examples:
   - Router heap 
   - Current best route/path information (e.g. https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/base/vpr_context.h#L272)

#### Current Behaviour

The router and relevant functions/methods use route_ctx to pass shared state.  Tracking mutation of route_ctx is adhoc, and calling functions can use global data unexpectedly.

#### Possible Solution

Starting with the leaf utilities (looking at route_tree_timing.h/.cpp now), use of mutable global data in the router should prohibited, and convert to passed arguments.  Core router algorithms should take pointers to the relevant mutable state they need, either via a `this` pointer via a router object, or via arguments.

#### Context

@HackerFoo FYI

Some router operations can be done in parallel, but identifying and implementing them is extremely difficult because of the use of mutable global state.

litghost, 2019-10-28:
Remaining callsites where mutable global data is use:
 - mutable_* : 
```
ack g_vpr_ctx.mutable_ | grep g_vpr_ctx | wc
     73     365    5469
```

Distribution of mutable data:
```
ack g_vpr_ctx.mutable_ | grep g_vpr_ctx | cut -d ':' -f 1 | sort | uniq -c
      1 check_route.cpp
      3 clock_connection_builders.cpp
      2 clock_network_builders.cpp
      4 route_breadth_first.cpp
     20 route_common.cpp
      1 router_delay_profiling.cpp
      2 router_lookahead.cpp
      8 route_timing.cpp
      2 route_tree_timing.cpp
      6 rr_graph_clock.cpp
      7 rr_graph.cpp
      5 rr_graph_indexed_data.cpp
      8 rr_graph_reader.cpp
      1 rr_graph_timing_params.cpp
      2 rr_metadata.cpp
      1 rr_node.cpp
```

litghost, 2019-10-28:
Use of routing_ctx data:
```
ack g_vpr_ctx.routing | grep g_vpr_ctx | cut -d ':' -f 1 | sort | uniq -c
      1 bucket.h
      3 check_route.cpp
      1 connection_box_lookahead_map.cpp
      1 connection_box_lookahead_map.cpp.orig
      2 route_breadth_first.cpp
     14 route_common.cpp
      1 router_delay_profiling.cpp
      1 router_lookahead.cpp
     10 route_timing.cpp
     11 route_tree_timing.cpp
      1 route_util.cpp
      1 segment_stats.cpp
      1 spatial_route_tree_lookup.cpp
```

litghost, 2019-10-29:
- [x] check_route.cpp
 - Used in `recompute_occupancy_from_scratch`, which is only done during `check_route` and `read_route`
- [x] clock_connection_builders.cpp
 - Okay, done prior to routing
- [ ] clock_network_builders.cpp
- [ ] route_breadth_first.cpp
- [ ] route_common.cpp
- [ ] router_delay_profiling.cpp
- [ ] router_lookahead.cpp
- [ ] route_timing.cpp
- [ ] route_tree_timing.cpp
- [ ] rr_graph_clock.cpp
- [ ] rr_graph.cpp
- [ ] rr_graph_indexed_data.cpp
- [ ] rr_graph_reader.cpp
- [ ] rr_graph_timing_params.cpp
- [ ] rr_metadata.cpp
- [ ] rr_node.cpp

vaughnbetz, 2019-10-30:
I generally agree it's best to minimize global state, and making a parallel router would be useful. Sameh Attia is interested in trying some new ideas we have for parallel routing for his PhD, so we may have a person to work on an algorithm that should scale better than existing approaches (but that is a ways off as he has other things to work on first).

Basic principles I'd like to follow:

1. Have an algorithm, or class of algorithms, in mind as you refactor router code, so that we don't make data structure changes that won't pay off, or are counter-productive. For example, the occupancy information per rr_node (and the associated acc_cost) makes the most sense to keep in one big data structure indexed by the rr_node id. Race conditions in updating it are (in the best known algorithms) avoided by scheduling nets with non-overlapping bounding boxes to be routed in parallel. While this has multiple threads reading & updating a single data structure, they are all working on different parts so there is no race. The parts they are working on change quickly, as a new set of nets is scheduled to be routed in parallel after the first set, still using the same big data structure -- this means we don't want to do a bunch of copies of parts of the rr_node_route_inf[] data to various threads. The approach Sameh and I are considering builds on these net partitioning approaches, but should go further to allow more parallelism.

2. Measure the QoR impact of each change. We don't want to bloat or slow down the current router with a data structure change that might lead to a parallel router in the future.  The router data structures are pretty tuned, so we could easily de-optimize them if each change isn't measured for QoR.

vaughnbetz, 2019-10-30:
Adding Sameh @samehattia

closed 2025-05-16 by github-actions[bot]

vaughnbetz, 2025-05-16:
This is done ... the router is parallel!