verilog-to-routing/vtr-verilog-to-routing #1022: Create VPR timing report on unconstrained paths

vaughnbetz, 2019-10-30:
See #1013 .
This is to track item 4 from that issue -- creating a report on unconstrained paths (clocks in particular, but also can include IO->reg paths and so on).

#### Proposed Behaviour
We should produce a report on unconstrained paths in final timing sign off.
Possibly we should also produce a big warning if we find nothing to optimize (but I'm not sure if that behaviour is easy to trigger given the other fixes in #1013, so perhaps a report is enough).

#### Current Behaviour
We create default constraints on all clocks if there is no sdc and obey the sdc if there is one. We don't produce any report on how much of the design the sdc constrains.

vaughnbetz, 2025-05-16:
This is a useful issue that we never resolved. Sending to @AlexandreSinger to see if he wants to take a crack at it, as he's messing about with SDC files right now.  Basically we could produce a report on which clock nets had no timing constraint at least. 
We might also do the same for IO paths, which might be a little more involved.
More involved still would be to do check the timing tags at the end of a timing analysis to see if we had found paths between registers where we don't know any timing constraint between them.

AlexandreSinger, 2025-05-17:
I'll reopen this issue since it is something I may want to look into if I find the time!