verilog-to-routing/vtr-verilog-to-routing #1029: Better documentation/description of lookahead definition of "congestion"

litghost, 2019-11-01:
The map lookahead precomputes two values:
 - Delay
 - Congestion

And the cost calculation is:
```
cost = criticality * delay + (1 - criticality) * congestion
```

I believe I understand the definition for delay, but I don't really grok what the "congestion" number is.  What is a good definition of the `congestion` value?

I assume there is a VTR paper (or potentially a code comment) explaining it, but I haven't located it just yet.

For context, I'm debugging a lookahead for the 7-series graph, and have begun to dial in the delay predication, but don't have a good enough understanding of congestion to understand how to fix lookahead miscosts.

@vaughnbetz / @kmurray

litghost, 2019-11-01:
An alternative name for this idea I believe is the "base cost".

One example of base cost is delay normalized, which seems to be closest to what the map lookahead implements

```
static float get_delay_normalization_fac(int nodes_per_chan,                   
                                         const t_rr_node_indices& L_rr_node_indices) {
    /* Returns the average delay to go 1 CLB distance along a wire.  */        
                                                                               
    const int clb_dist = 3; /* Number of CLBs I think the average conn. goes. */
                                                                               
    int inode, itrack, cost_index;                                             
    float Tdel, Tdel_sum, frac_num_seg;                                        
                                                                               
    auto& device_ctx = g_vpr_ctx.device();                                     
                                                                               
    Tdel_sum = 0.;                                                             
                                                                               
    for (itrack = 0; itrack < nodes_per_chan; itrack++) {                      
        inode = find_average_rr_node_index(device_ctx.grid.width(), device_ctx.grid.height(), CHANX, itrack,
                                           L_rr_node_indices);                 
        if (inode == -1)                                                       
            continue;                                                          
        cost_index = device_ctx.rr_nodes[inode].cost_index();                  
        frac_num_seg = clb_dist * device_ctx.rr_indexed_data[cost_index].inv_length;
        Tdel = frac_num_seg * device_ctx.rr_indexed_data[cost_index].T_linear  
               + frac_num_seg * frac_num_seg                                   
                     * device_ctx.rr_indexed_data[cost_index].T_quadratic;     
        Tdel_sum += Tdel / (float)clb_dist;                                    
    }                                                                          
                                                                               
    for (itrack = 0; itrack < nodes_per_chan; itrack++) {                      
        inode = find_average_rr_node_index(device_ctx.grid.width(), device_ctx.grid.height(), CHANY, itrack,
                                           L_rr_node_indices);                 
        if (inode == -1)                                                       
            continue;                                                          
        cost_index = device_ctx.rr_nodes[inode].cost_index();                  
        frac_num_seg = clb_dist * device_ctx.rr_indexed_data[cost_index].inv_length;
        Tdel = frac_num_seg * device_ctx.rr_indexed_data[cost_index].T_linear  
               + frac_num_seg * frac_num_seg                                   
                     * device_ctx.rr_indexed_data[cost_index].T_quadratic;     
        Tdel_sum += Tdel / (float)clb_dist;                                    
    }                                                                          
                                                                               
    return (Tdel_sum / (2. * nodes_per_chan));                                 
} 
```

litghost, 2019-11-01:
In particular the congestion value that is tripping up the lookahead at this point is for the "FAN_ALT" node type.  This node is fairly odd, I've made a little diagram:

[image]

mithro, 2019-11-01:
@litghost - That image appears to be broken?

litghost, 2019-11-01:
> @litghost - That image appears to be broken?

Fixed, looks like Google Drawing returns to wrong context type, as a result github's image caching breaks.

vaughnbetz, 2019-11-03:
The congestion cost is really base_cost * pres_cost * acc_cost.  pres_cost and acc_cost are automatically computed, so base_cost is the only thing to worry about setting.

Base costs are set at https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/6c8608c8afab583e52c690133d090db7ccc7e3cd/vpr/src/route/rr_graph_indexed_data.cpp#L145

The default base_cost_type is DELAY_NORMALIZED_LENGTH so that's the code to look at.  That makes the base cost of each wire equal to its length * the average delay of the wires in the architecture.

OPINs have a slightly lower base_cost (.95 of a length 1 wire) while IPINs have the same base cost as a length 1 wire.

All numbers are normalized to the average delay of a wire so we can add congestion costs and delay costs in the router heap sorting in a meaningful way (they have the same units).  Costing OPINs slightly lower makes the router explore them a bit earlier, which has been found to have a small CPU time benefit.

So you can adjust base costs to some other number times the delay_normalization_factor. The goal is to make the router explore the most promising paths first, if there is no congestion and no timing issues.  So if you need to go through several nodes and compare that to a direct wire->wire path, you might make the base cost of these unusual nodes (look like extra muxes for more flexibility in indirect routing paths) lower; perhaps 0.5 or 0.33 of the usual base cost.  But it depends if you think they should be explored more aggressively or not -- if you usually don't need these nodes to complete a route then maybe leaving their base cost at 1 * the delay normalization factor makes sense.

closed 2019-12-10 by litghost