verilog-to-routing/vtr-verilog-to-routing #1032: Unbounded memory usage in case of pathelogical heap behavior

litghost, 2019-11-04:
#### Expected Behaviour

Router should have bounded memory usage in all circumstances

#### Current Behaviour

A pathological graph can trivially cause a memory explosion.

#### Possible Solution

I haven't thought about solutions, I wanted to document the issue first. 

#### Context

The core router expansion is roughly:
 1. Expansion lowest code node
 2. Add heap entries for all nodes that have cheaper cost than the previously seen entry (see https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/route_timing.cpp#L1931)
 3. Repeat until target is found

Step 2 is where the memory explosion can occur, because the same node can be added to the heap infinitely many times, as long as each time the expansion sees a lower cost, even if the cost is only lower by a small amount.  Even worse, the best cost is not updated when a node is placed on the heap, but when it is popped!

See check for lower cost is here:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/route_timing.cpp#L1925-L1931

But costs are only updated once popped:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/route_timing.cpp#L1490-L1491

The heap will contain N-1 duplicate entries for a node, if the node is added N times prior to being popped.  This can add up pretty quickly, I've seen cases of memory balloning to ~100 GB.

The maximum number of heap entries should be bounded to the number of nodes, if only there was a way to prune duplicate nodes of higher cost.

litghost, 2020-07-27:
@vaughnbetz We are seeing failures of this class with semi-regular frequency.  In the case of non-RCV routers, we only need to keep the "best" copy of each rr node.  For RCV routers, what should we keep?  I could code a pruning routing to keep a copy of each rr node with a histogram of codes.

I'll work on a proposed fix for this pathological behavior.

vaughnbetz, 2020-07-27:
I think it's too soon to say what to do for RCV.  So long as we can turn the pruning on and off, we can experiment with it later when we have RCV, so your changes aren't gated by any future (impossible to predict) RCV requirements.

A quick thing to test: if you update the best_cost when you put something on the heap, and make sure the pruning test when you take things off the heap at
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/9d967a2b0c929f1c86d7b5dedfc566515ea9b52f/vpr/src/route/connection_router.cpp#L335
is changed to explore the node if the new_cost <= best_cost, is it faster and equivalent quality?

litghost, 2020-07-27:
My first PR is to add heap pruning in the first place, to prevent the unbounded memory usage in all cases.  A follow up PR can try to tune avoiding heap pushes that would never get seen.

litghost, 2020-07-27:
I guess what I'm saying is I have two goals:
 1. Prevent unbounded memory usage in the pathological case
 2. Avoid unneed heap pushes that result in work (allocation and / or pointer manipulation)

The first change is a robustness change, the second change is a performance fix

litghost, 2020-07-27:
> A quick thing to test: if you update the path_cost when you put something on the heap, and make sure the pruning test when you take things off the heap at
> https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/9d967a2b0c929f1c86d7b5dedfc566515ea9b52f/vpr/src/route/connection_router.cpp#L335

I almost want to add a route_inf var that is "cheapest pushed" and a "cheapest popped", because those are two different parameters.  The overhead of adding the second float is not clear compared with the performance gain by having a smaller heap.

vaughnbetz, 2020-07-27:
Yup, not sure if that would be a win or not.  Doing the first change (update best_cost early, and change to a >= in taking things off) also may not be a win (depends how many ties we have usually).  Easy to try though, so I think they're worth trying.  
Also, I understand this is separate from worst-case checks -- will look at that PR now.

closed 2025-05-16 by github-actions[bot]

vaughnbetz, 2025-05-16:
I think this has been superceded by various heap optimizations and pruning optimizations we've made in the last year that sped the router up significantly.