verilog-to-routing/vtr-verilog-to-routing #1038: Assertion failure with PLLs potentially due to propigated clocks

litghost, 2019-11-06:
#### Expected Behaviour

No assertion is fired on PLLs.

#### Current Behaviour

```
/usr/local/google/home/keithrothman/cat_x/vtr-verilog-to-routing/vpr/src/timing/timing_graph_builder.cpp:207 add_block_to_timing_graph: Assertion '!model_port->is_clock' failed (Primitive data output (CLKFBOUT) can not be a clock generator, attached to plle2_test.pll.CLKFBOUT.plle2_test.pll.CLKFBOUT[0]).
```

> "Primitive data output can not be a clock generator"
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blame/master/vpr/src/timing/timing_graph_builder.cpp#L200

Assertion was introduced in https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/53510d94baf7691e1f15b8e5a857edec140688ed#diff-fdc393e2374fe736fe08a9f1fc7fbdf9R194, however I believe the bug is related to https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/440e6b1baa04e889f11edd4259471a88291e1fa8#diff-cb6d86211fc138d5f2a87ba2ac745365 removing some clock sources from `find_netlist_clocks`.

#### Possible Solution

It seems like `find_netlist_clocks` clocks needs two outputs:
 - The list of `SOURCE` pins that are clocks
 - The map of `SOURCE` that are unconstrained and are implicitly defined as coming from a parent clock

#### Steps to Reproduce

Working on replication case now

1. 
2. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

litghost, 2019-11-06:
Sorry I think the bug was here: https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/5007035d3389c84b7506d050e7b2b01bd9017a26#diff-31ec6c191fb261c72ad6804a23bf1a26

litghost, 2019-11-06:
So as far as I can tell, because `find_netlist_logical_clock_drivers` now propagates clocks through black boxes (which correctly are marked ` <port is_clock="1" name="CLKFBOUT"/>` https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/primitives/plle2_adv/plle2_adv.model.xml#L21), but that results in the clock output not be marked as `is_netlist_clock_source`
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/4a16cd69263923109e321eb108b9f3b47508e637/vpr/src/timing/timing_graph_builder.cpp#L182

litghost, 2019-11-06:
I've created a test case demostrating this failure https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1041

litghost, 2019-11-21:
@kmurray Friendly bump

kmurray, 2020-04-02:
I've looked into this, and I believe (based on the test case from #1041) that this is primarily a modelling issue.

As noted in the [timing modelling tutorial](https://docs.verilogtorouting.org/en/latest/tutorials/arch/timing_modeling/#clock-generators), PLLs should specify their clock outputs with `is_clock="1"`.

However the example architecture specifies both `is_clock="1"` **and** combinationally connects the input and output clocks:
```xml
    <model name="pll">
      <input_ports>
        <port is_clock="1" name="dclk"/>
        <port clock="dclk" name="den"/>
        <port clock="dclk" name="dwe"/>
        <port clock="dclk" name="daddr"/>
        <port clock="dclk" name="di"/>

        <port is_clock="1" name="clkfbin"/>
        <port is_clock="1" name="clkin1" combinational_sink_ports="clkfbout clkout0 clkout1 clkout2 clkout3 clkout4 clkout5"/>
        <port is_clock="1" name="clkin2" combinational_sink_ports="clkfbout clkout0 clkout1 clkout2 clkout3 clkout4 clkout5"/>
      </input_ports>
      <output_ports>
        <port clock="dclk" name="do"/>
        <port clock="dclk" name="drdy"/>

        <port is_clock="1" name="clkfbout"/>
        <port is_clock="1" name="clkout0"/>
        <port is_clock="1" name="clkout1"/>
        <port is_clock="1" name="clkout2"/>
        <port is_clock="1" name="clkout3"/>
        <port is_clock="1" name="clkout4"/>
        <port is_clock="1" name="clkout5"/>
      </output_ports>
    </model>
```

This is telling VPR conflicting things: 
1. The combinational connections tells VPR the clkin->clockout connections are clock buffers, 
2. The `is_clock="1"` on the outputs tells VPR they should be clock sources 

Since the current code checks (1) first, it then fails the assertion that (2) shouldn't be true in case (1).

Removing the combinational connections:
```xml
    <model name="pll">
      <input_ports>
        <port is_clock="1" name="dclk"/>
        <port clock="dclk" name="den"/>
        <port clock="dclk" name="dwe"/>
        <port clock="dclk" name="daddr"/>
        <port clock="dclk" name="di"/>

        <port is_clock="1" name="clkfbin"/>
        <port is_clock="1" name="clkin1"/> <!-- no comb sink ports -->
        <port is_clock="1" name="clkin2"/> <!-- no  comb sink ports -->
      </input_ports>
      <output_ports>
        <port clock="dclk" name="do"/>
        <port clock="dclk" name="drdy"/>

        <port is_clock="1" name="clkfbout"/>
        <port is_clock="1" name="clkout0"/>
        <port is_clock="1" name="clkout1"/>
        <port is_clock="1" name="clkout2"/>
        <port is_clock="1" name="clkout3"/>
        <port is_clock="1" name="clkout4"/>
        <port is_clock="1" name="clkout5"/>
      </output_ports>
    </model>
```
ensures that VPR correctly treats all the clkout ports as unique clock sources.

Note that I think this is the correct way to model this. VPR has no way of knowing what the relationship is between the input and output clocks of a PLL. Modelling them as clock buffers doesn't make sense either, as that assumes they are the same logical clock. In contrast a PLLs input/ouput clocks, while 'related' in some way are logically distinct (e.g. may be different frequencies/phases).

What that relationship is has to be supplied to VPR via SDC constraints. VPR has no way to figure that out from a black box netlist primitive.

So for the example netlist from #1041:
```
subckt pll \
    clkin1=clk2 \
    clkout0=clk2_pll_out \
    clkfbin=clkfb \
    clkfbout=clkfb
```

The SDC could be something like:
```
create_clock clk2 -period 1
create_clock clk2_pll_out -period 2 #2x clk2
```
(As discussed in #1134 this is equivalent to creating clk2 and then using create_generated_clock for our purposes).

commit 96ae623e (2020-04-03): archfpga: Forbid specifying model outputs as both clock sources buffers

Previously such specifications would cause a hard to diagnose assertion
error in VPR. We now provide a clearer error message about the invalid
specification earlier when processing the architecture file.

Closes #1038
PR #1259 (2020-04-03): Improve pll error checking and docs
Code fixes, documentation updates and improved tests for PLL clock source specification/detection.

Closes #1038
commit 323ef1e1 (2020-04-03): archfpga: Forbid specifying model outputs as both clock sources buffers

Previously such specifications would cause a hard to diagnose assertion
error in VPR. We now provide a clearer error message about the invalid
specification earlier when processing the architecture file.

Closes #1038
closed 2020-04-03 by kmurray