verilog-to-routing/vtr-verilog-to-routing #1039: Better documentation on the libtatum, and its usage in VPR

litghost, 2019-11-06:
#### Context

I've been trying to debug #1038, but I don't understand how libtatum builds it timing graph, and there is insufficient comments in https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/timing/timing_graph_builder.cpp for me to understand what is being done.

There are two asks:
 - A short piece of documentation explaining libtatum from a higher level.  Example questions:
   - What is a tatum node?  What types of nodes are there?
   - What is a tag, and how are they used?
 - Better documentation in https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/timing/timing_graph_builder.cpp explaining however VPR is mapping the arch/rrgraph model into libtatum

I'll continue to stare at https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/timing/timing_graph_builder.cpp and see if I can grok what is going on.  However the amount of time it is taking me to understand the code (as someone who knows what an STA engine is and the relevant VPR data structures), tells me the documentation around libtatum is too lite.

litghost, 2019-11-06:
@kmurray

kmurray, 2019-11-06:
Yep, agreed this is an area in need of improvement!

closed 2025-05-16 by github-actions[bot]