verilog-to-routing/vtr-verilog-to-routing #1043: Placer congestion awareness

litghost, 2019-11-12:
#### Expected Behaviour

Placer avoids creating "hard to route" placement.

#### Current Behaviour

Placer may generate two placements with roughly equivilant CPD, but have dramatically different routability.

#### Possible Solution

A quick google search has lead me to believe that congestion aware placement exists (example: [GPlace 3](https://dl.acm.org/citation.cfm?id=3233244)).  Does VPR's placer account for congestion?  Is there a way to add a congestion metric to cost function without a wholescale re-write?

#### Steps to Reproduce

1. 
2. 

#### Context

In the 7-series graph, the picosoc place/route targets on the full 50T have shown strong sensitivity to placement.  With only changes to placement the routing has:
 - Completed in 1500 secs with a CPD of ~18 ns
 - Completed in 3700 secs with a CPD of ~18.5 ns
 - Completed in 5800 secs with a CPD of ~85 ns
 - Crashed with an OOM after taking 5+ hours and consuming ~90 GB of RAM

The circuit and packing were the same in each of the above cases.

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

litghost, 2019-11-12:
@vaughnbetz @kmurray

vaughnbetz, 2019-11-12:
Yes, you can make placers congestion aware. There is a significant literature on it, so probably a longer conversation about approaches, etc.
Before making the placer congestion aware you want to trust the router, however, as congestion terms will compete against wirelength minimization and timing optimization, so you don't want to do more congestion optimization than necessary.  If you don't trust the router and/or don't know what kind of congestion you're trying to reduce you are likely to de-optimize placement instead of optimize it more.
For designs the size of picosoc I would not expect global (inter-slice) congestion to be a signficant issue. 
- What resources are congested?  Are there some overused nodes for special signals like CE that are getting overused, and which are best attacked with clustering changes?
- Is the router going into some strange behaviour for other reasons?

mithro, 2019-11-12:
I agree with what @vaughnbetz says. Let's first figure out what is congested...

I did find http://www.ispd.cc/contests/16/ispd2016_contest.html - ISPD 2016 : Routability-Driven FPGA Placement Contest (targeting Xilinx Ultrascale rather than Xilinx Series 7 -- but still might have interesting information).

litghost, 2019-11-12:
> * What resources are congested?  Are there some overused nodes for special signals like CE that are getting overused, and which are best attacked with clustering changes?

I'm running the iteration now, but the congested nodes do not look super specialized.

litghost, 2019-11-12:
Preliminary results (iteration 24, over used nodes 397):
```
FAN | 96
BYP | 108
GFAN | 9
interconnect | 184
```
FAN, BYP and GFAN are specialized routing resources.  Overall about half of the congestion is on specialized resources, but the other half is on more generalized routing resources.

The router tends to start stalling hard around now (between 200 and 300 overused nodes).

Here an example where the router behaviors "cleanly":
```
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 109: No routing path for connection to sink_rr 2974479, retrying with full device bounding box
Warning 110: No routing path for connection to sink_rr 1913747, retrying with full device bounding box
Warning 111: 90 timing endpoints were not constrained during timing analysis
   1   38.7     0.0    0 1.4e+08    5608   17086   11932 ( 0.337%)  738409 ( 8.7%)   18.018 -1.975e+04    -18.018      0.000      0.000      N/A
   2  140.8     0.5  547 4.7e+08    4435   13801   10107 ( 0.286%)  477837 ( 5.6%)   17.952 -2.149e+04    -17.952      0.000      0.000      N/A
   3  244.6     0.6  105 7.5e+08    4202   13267    9230 ( 0.261%)  478998 ( 5.6%)   17.942 -2.202e+04    -17.942      0.000      0.000      N/A
   4  326.4     0.8   83 9.9e+08    3995   12661    8549 ( 0.242%)  482078 ( 5.7%)   18.067 -2.265e+04    -18.067      0.000      0.000      N/A
   5  358.3     1.1   71 1.1e+09    3740   12029    7460 ( 0.211%)  485826 ( 5.7%)   18.069 -2.298e+04    -18.069      0.000      0.000      N/A
   6  345.8     1.4   45 1.1e+09    3422   11066    6118 ( 0.173%)  491781 ( 5.8%)   18.068 -2.320e+04    -18.068      0.000      0.000      N/A
   7  334.5     1.9   57 1.0e+09    3091    9912    4860 ( 0.137%)  498352 ( 5.9%)   18.105 -2.338e+04    -18.105      0.000      0.000      N/A
   8  310.5     2.4   57 9.2e+08    2741    8777    3653 ( 0.103%)  507371 ( 6.0%)   18.204 -2.355e+04    -18.204      0.000      0.000      N/A
   9  269.3     3.1   42 8.2e+08    2383    7790    2706 ( 0.076%)  514139 ( 6.0%)   18.195 -2.380e+04    -18.195      0.000      0.000      N/A
  10  205.4     4.1   49 6.6e+08    1980    6481    1843 ( 0.052%)  521312 ( 6.1%)   18.140 -2.398e+04    -18.140      0.000      0.000       40
  11  174.8     5.3   39 5.1e+08    1641    5287    1395 ( 0.039%)  526166 ( 6.2%)   18.161 -2.404e+04    -18.161      0.000      0.000       35
  12  139.4     6.9   34 3.6e+08    1317    4157    1000 ( 0.028%)  529907 ( 6.2%)   18.146 -2.416e+04    -18.146      0.000      0.000       35
  13  100.0     9.0   31 2.7e+08    1050    3314     724 ( 0.020%)  532082 ( 6.3%)   18.164 -2.428e+04    -18.164      0.000      0.000       34
  14   74.6    11.6   25 2.1e+08     844    2687     502 ( 0.014%)  535855 ( 6.3%)   18.190 -2.426e+04    -18.190      0.000      0.000       34
  15   55.0    15.1   24 1.5e+08     638    2107     344 ( 0.010%)  538537 ( 6.3%)   18.190 -2.436e+04    -18.190      0.000      0.000       33
  16   51.8    19.7   22 1.3e+08     545    1822     240 ( 0.007%)  540369 ( 6.4%)   18.250 -2.434e+04    -18.250      0.000      0.000       33
  17   47.3    25.6   21 1.1e+08     444    1562     193 ( 0.005%)  541769 ( 6.4%)   18.767 -2.436e+04    -18.767      0.000      0.000       32
  18   48.1    33.3   13 9.3e+07     352    1193     110 ( 0.003%)  542982 ( 6.4%)   18.666 -2.437e+04    -18.666      0.000      0.000       32
  19   36.8    43.3   12 7.0e+07     291     995      88 ( 0.002%)  543819 ( 6.4%)   18.377 -2.440e+04    -18.377      0.000      0.000       32
  20   35.9    56.2    5 6.4e+07     289    1029      70 ( 0.002%)  544633 ( 6.4%)   18.364 -2.440e+04    -18.364      0.000      0.000       32
  21   33.5    73.1    5 6.1e+07     269     969      49 ( 0.001%)  545616 ( 6.4%)   18.569 -2.441e+04    -18.569      0.000      0.000       32
  22   24.8    95.0    8 4.7e+07     244     873      49 ( 0.001%)  546072 ( 6.4%)   18.364 -2.443e+04    -18.364      0.000      0.000       32
  23   22.3   123.5    4 4.5e+07     259     954      36 ( 0.001%)  546407 ( 6.4%)   18.569 -2.441e+04    -18.569      0.000      0.000       33
  24   19.6   160.6    6 3.6e+07     222     826      28 ( 0.001%)  546511 ( 6.4%)   18.568 -2.443e+04    -18.568      0.000      0.000       34
  25   16.0   208.8    4 3.2e+07     224     827      23 ( 0.001%)  547090 ( 6.4%)   18.568 -2.443e+04    -18.568      0.000      0.000       35
  26   15.4   271.4    4 2.8e+07     214     809      21 ( 0.001%)  547152 ( 6.4%)   18.572 -2.443e+04    -18.572      0.000      0.000       35
  27   15.7   352.8    3 3.1e+07     236     866      20 ( 0.001%)  547569 ( 6.4%)   18.554 -2.443e+04    -18.554      0.000      0.000       37
  28   13.4   458.7    3 2.6e+07     219     821      16 ( 0.000%)  547599 ( 6.4%)   18.572 -2.443e+04    -18.572      0.000      0.000       37
[... cut ...]
  51  160.8 1.9e+05   16 3.3e+08     228     737      55 ( 0.002%)  653680 ( 7.7%)   21.069 -2.812e+04    -21.069   -0.03016     -0.030      171
  52  726.0 2.5e+05   13 8.5e+08     212     702      56 ( 0.002%)  684072 ( 8.0%)   53.537 -3.454e+04    -53.537   -0.03016     -0.030      168
  53 22436.9 3.2e+05   53 2.2e+10     110     317      60 ( 0.002%)  721984 ( 8.5%)   90.519 -4.628e+04    -90.519   -0.03016     -0.030      172
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

The only difference (to my knowledge) between these runs is the output of the placer.

litghost, 2019-11-12:
Overuse counts at iteration 29 (overuse count 346):

Segment | Overuse count
-- | --
FAN | 97
BYP | 97
GFAN | 13
interconnect | 162

litghost, 2019-11-12:
Here is a visual display of the overuse:
[image]

litghost, 2019-11-12:
Overuse counts at iter 32 (overused nodes 305):

Segment | Overuse count
-- | --
FAN | 87
BYP | 72
GFAN | 7
BRAM Address | 2
interconnect | 148

litghost, 2019-11-12:
Router iterations for above data:
```
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 110: No routing path for connection to sink_rr 1169968, retrying with full device bounding box
Warning 111: No routing path for connection to sink_rr 1725587, retrying with full device bounding box
Warning 112: No routing path for connection to sink_rr 1001773, retrying with full device bounding box
Warning 113: 90 timing endpoints were not constrained during timing analysis
   1   17.6     0.0    0 1.3e+08    5444   16741   11717 ( 0.331%)  716401 ( 8.4%)   19.039 -1.936e+04    -19.039  -0.008172     -0.008      N/A
Warning 114: No routing path for connection to sink_rr 1001773, retrying with full device bounding box
   2   90.0     0.5  533 5.4e+08    4363   13603   10004 ( 0.283%)  470370 ( 5.5%)   19.079 -2.121e+04    -19.079      0.000      0.000      N/A
   3  142.7     0.6   88 8.3e+08    4128   13103    9990 ( 0.282%)  475735 ( 5.6%)   19.073 -2.250e+04    -19.073      0.000      0.000      N/A
   4  178.1     0.8   80 1.0e+09    3969   12800    9214 ( 0.260%)  482555 ( 5.7%)   19.038 -2.326e+04    -19.038      0.000      0.000      N/A
   5  194.5     1.1   68 1.1e+09    3766   11989    8384 ( 0.237%)  489483 ( 5.8%)   19.044 -2.356e+04    -19.044      0.000      0.000      N/A
   6  190.2     1.4   47 1.1e+09    3534   11208    6782 ( 0.192%)  499129 ( 5.9%)   19.019 -2.397e+04    -19.019      0.000      0.000      N/A
   7  184.3     1.9   58 1.1e+09    3282   10410    5681 ( 0.161%)  509151 ( 6.0%)   19.041 -2.419e+04    -19.041      0.000      0.000      N/A
   8  167.8     2.4   81 9.6e+08    3045    9382    4514 ( 0.128%)  519817 ( 6.1%)   19.033 -2.444e+04    -19.033      0.000      0.000      N/A
   9  152.5     3.1   73 8.7e+08    2605    8137    3400 ( 0.096%)  529335 ( 6.2%)   19.338 -2.493e+04    -19.338      0.000      0.000      N/A
  10  127.3     4.1   74 7.2e+08    2272    6989    2649 ( 0.075%)  537591 ( 6.3%)   19.051 -2.494e+04    -19.051      0.000      0.000       46
  11  105.3     5.3   74 6.0e+08    1920    5964    1985 ( 0.056%)  545591 ( 6.4%)   19.051 -2.511e+04    -19.051      0.000      0.000       43
  12   78.1     6.9   67 4.4e+08    1544    4698    1541 ( 0.044%)  550019 ( 6.5%)   19.046 -2.532e+04    -19.046      0.000      0.000       42
  13   71.0     9.0   63 3.9e+08    1325    3957    1252 ( 0.035%)  555231 ( 6.5%)   19.046 -2.534e+04    -19.046      0.000      0.000       40
  14   71.7    11.6   48 3.6e+08    1133    3371    1065 ( 0.030%)  559081 ( 6.6%)   19.046 -2.538e+04    -19.046      0.000      0.000       41
  15   75.4    15.1   43 3.5e+08    1010    2954     933 ( 0.026%)  562199 ( 6.6%)   19.047 -2.533e+04    -19.047      0.000      0.000       42
  16   75.8    19.7   34 3.3e+08     870    2527     787 ( 0.022%)  564826 ( 6.6%)   19.042 -2.534e+04    -19.042      0.000      0.000       44
  17   78.6    25.6   26 3.2e+08     763    2180     679 ( 0.019%)  568541 ( 6.7%)   19.032 -2.534e+04    -19.032      0.000      0.000       48
  18   84.8    33.3   29 3.2e+08     684    2116     612 ( 0.017%)  571692 ( 6.7%)   19.012 -2.538e+04    -19.012      0.000      0.000       49
  19   96.5    43.3   30 3.5e+08     627    1859     552 ( 0.016%)  574446 ( 6.8%)   19.012 -2.540e+04    -19.012      0.000      0.000       53
  20   92.9    56.2   27 3.4e+08     569    1651     498 ( 0.014%)  577781 ( 6.8%)   19.012 -2.541e+04    -19.012      0.000      0.000       56
  21   86.7    73.1   32 3.1e+08     544    1571     467 ( 0.013%)  580134 ( 6.8%)   19.012 -2.545e+04    -19.012      0.000      0.000       61
  22   90.7    95.0   30 3.1e+08     500    1391     414 ( 0.012%)  584085 ( 6.9%)   19.017 -2.547e+04    -19.017      0.000      0.000       63
  23   99.7   123.5   26 3.4e+08     478    1397     416 ( 0.012%)  587534 ( 6.9%)   19.018 -2.551e+04    -19.018      0.000      0.000       68
  24  110.8   160.6   35 3.6e+08     449    1364     397 ( 0.011%)  590867 ( 7.0%)   19.017 -2.548e+04    -19.017      0.000      0.000       72
  25  107.5   208.8   35 3.4e+08     428    1344     388 ( 0.011%)  593847 ( 7.0%)   19.017 -2.555e+04    -19.017      0.000      0.000       79
  26  121.3   271.4   38 3.8e+08     432    1357     402 ( 0.011%)  596775 ( 7.0%)   19.003 -2.559e+04    -19.003      0.000      0.000       83
  27  120.7   352.8   30 3.8e+08     431    1310     368 ( 0.010%)  599023 ( 7.0%)   19.003 -2.558e+04    -19.003      0.000      0.000       95
  28   97.2   458.7   38 3.0e+08     400    1206     367 ( 0.010%)  601256 ( 7.1%)   19.064 -2.564e+04    -19.064      0.000      0.000      100
  29  103.4   596.3   36 3.3e+08     566    1791     346 ( 0.010%)  603618 ( 7.1%)   21.309 -2.564e+04    -21.309      0.000      0.000      114
[... cut ...]
  32  117.3  1310.0   37 3.4e+08     407    1318     305 ( 0.009%)  608532 ( 7.2%)   20.516 -2.574e+04    -20.516      0.000      0.000      136
  33  120.6  1703.0   31 3.5e+08     438    1410     331 ( 0.009%)  610288 ( 7.2%)   21.042 -2.578e+04    -21.042      0.000      0.000      151
  34  124.9  2213.9   34 3.7e+08     447    1397     322 ( 0.009%)  613277 ( 7.2%)   21.831 -2.586e+04    -21.831      0.000      0.000      161
  35  126.2  2878.1   36 3.7e+08     412    1261     314 ( 0.009%)  614341 ( 7.2%)   21.669 -2.589e+04    -21.669      0.000      0.000      184
```

litghost, 2019-11-12:
Overuse logs for iteration 1-40:
[overuse_logs.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3838334/overuse_logs.tar.gz)

litghost, 2019-11-12:
@vaughnbetz @mithro So I think your questions are valid, but from my untrained eye, I believe that the hypothesis of router misbehavior matches the data.  What are some concrete things to check and/or test for to estimate router "trust"?

litghost, 2019-11-12:
Overuse counts for iteration 41 (overused nodes 227):

Segment | Overuse count
-- | --
FAN | 56
BYP | 73
GFAN | 8
BRAM Address | 4
interconnect | 98

mithro, 2019-11-12:
It might be worth checking what the nets are that have not been successful routed? I could imagine an issue where a clock net or high fan out net gets routed really late and that causes issues.

vaughnbetz, 2019-11-12:
The router often takes more than 50 iterations to converge.  So the fact that it runs out of memory at iteration 53 is odd.  Is this with the re-tuned / improved lookahead?  
If you set the maximum criticality in the router to 0 does it converge?  That would test what happens when the router is focusing less on timing. Could do the same experiment with maximum criticality of 0.1.

litghost, 2019-11-13:
> Is this with the re-tuned / improved lookahead?

Ya, this is with an expansion on summed base cost, and base cost is correctly used.

litghost, 2019-11-13:
Criticality 0:
```
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 110: No routing path for connection to sink_rr 1169968, retrying with full device bounding box
Warning 111: No routing path for connection to sink_rr 1725587, retrying with full device bounding box
Warning 112: No routing path for connection to sink_rr 1001773, retrying with full device bounding box
Warning 113: 90 timing endpoints were not constrained during timing analysis
   1   23.5     0.0    0 1.8e+08    5444   16741   15947 ( 0.451%)  414632 ( 4.9%)   37.949 -3.322e+04    -37.949      0.000      0.000      N/A
Warning 114: No routing path for connection to sink_rr 1001773, retrying with full device bounding box
   2   97.3     0.5  105 6.0e+08    4799   14683   17598 ( 0.497%)  427775 ( 5.0%)   35.666 -3.562e+04    -35.666      0.000      0.000      N/A
   3  131.5     0.6   82 7.5e+08    4535   14169   14691 ( 0.415%)  441308 ( 5.2%)   37.335 -3.707e+04    -37.335      0.000      0.000      N/A
   4  157.1     0.8  115 8.5e+08    4348   14075   11657 ( 0.329%)  449698 ( 5.3%)   37.134 -3.754e+04    -37.134      0.000      0.000      N/A
   5  163.6     1.1   75 8.6e+08    4058   13518    8860 ( 0.250%)  462611 ( 5.4%)   38.664 -3.754e+04    -38.664      0.000      0.000      N/A
   6  162.4     1.4   59 8.4e+08    3840   12874    6777 ( 0.191%)  475055 ( 5.6%)   37.403 -3.713e+04    -37.403      0.000      0.000      N/A
   7  154.3     1.9   52 7.8e+08    3656   12405    4850 ( 0.137%)  488825 ( 5.8%)   40.636 -3.921e+04    -40.636      0.000      0.000      N/A
   8  141.5     2.4   67 7.2e+08    3447   11863    3436 ( 0.097%)  500284 ( 5.9%)   37.473 -3.749e+04    -37.473      0.000      0.000      N/A
   9  131.8     3.1   78 6.7e+08    3285   11500    2368 ( 0.067%)  508424 ( 6.0%)   38.345 -3.751e+04    -38.345      0.000      0.000      N/A
  10  122.5     4.1   62 6.3e+08    3160   11207    1682 ( 0.048%)  515499 ( 6.1%)   38.003 -3.818e+04    -38.003      0.000      0.000       33
  11  119.7     5.3   69 6.0e+08    3081   10942    1163 ( 0.033%)  522368 ( 6.1%)   37.835 -3.820e+04    -37.835      0.000      0.000       31
  12  124.3     6.9   69 5.9e+08    3041   10866     854 ( 0.024%)  525719 ( 6.2%)   37.991 -3.800e+04    -37.991      0.000      0.000       31
  13  122.6     9.0   50 5.7e+08    2980   10770     633 ( 0.018%)  530512 ( 6.2%)   38.663 -3.812e+04    -38.663      0.000      0.000       31
  14  129.0    11.6   49 5.6e+08    2961   10681     470 ( 0.013%)  534158 ( 6.3%)   38.002 -3.806e+04    -38.002      0.000      0.000       32
  15  122.7    15.1   54 5.2e+08    2915   10595     354 ( 0.010%)  538727 ( 6.3%)   37.709 -3.819e+04    -37.709      0.000      0.000       32
  16  112.7    19.7   53 4.7e+08    2896   10555     284 ( 0.008%)  541531 ( 6.4%)   39.685 -3.859e+04    -39.685      0.000      0.000       33
  17  121.8    25.6   56 4.8e+08    2885   10527     228 ( 0.006%)  544278 ( 6.4%)   39.549 -3.870e+04    -39.549      0.000      0.000       34
  18  123.0    33.3   46 4.7e+08    2881   10503     206 ( 0.006%)  546888 ( 6.4%)   40.302 -3.921e+04    -40.302      0.000      0.000       35
  19  109.0    43.3   35 4.4e+08    2867   10478     178 ( 0.005%)  549650 ( 6.5%)   39.765 -3.918e+04    -39.765      0.000      0.000       37
  20  103.6    56.2   45 4.2e+08    2868   10489     179 ( 0.005%)  552243 ( 6.5%)   39.704 -3.919e+04    -39.704      0.000      0.000       39
  21  110.7    73.1   45 4.3e+08    2862   10477     174 ( 0.005%)  553601 ( 6.5%)   39.784 -3.958e+04    -39.784      0.000      0.000       43
  22  115.4    95.0   50 4.4e+08    2849   10457     142 ( 0.004%)  556504 ( 6.5%)   40.416 -3.964e+04    -40.416      0.000      0.000       46
  23  109.3   123.5   48 4.1e+08    2846   10455     136 ( 0.004%)  558142 ( 6.6%)   38.788 -3.912e+04    -38.788      0.000      0.000       50
```

Router is still having trouble resolving the same type of nodes, but it is making more progress with `--max_criticality 0`.  

Given how slower the router is resolving the congestion on a hand full of nodes, I've taken a look at congestion lookahead for the relevant bits, and they look pretty reasonable.  One thing I've thought about (but haven't had an answer too) is the fact that the congestion lookahead assume that `pres_fac * acc_fac` ~= 1.  I totally expect the map lookahead to do poorly once `pres_fac` and `acc_fac` grow.  Even more so, as `pres_fac` and `acc_fac` grow, it will effectively disable the lookahead by virtue of weighting the backward path so much higher than the forward estimate.  Should there by a "dynamic" or adaptive part of the lookahead based on `acc_fac`?  If so, what would it look like?  I haven't come up with any bright ideas yet.

vaughnbetz, 2019-11-13:
The lookahead assumes no congestion, so pres_cost and acc_cost = 1.  Optimistic, but hard to predict much better. Note that if a node is not congested, and has never been, acc_cost and pres_cost are both 1, no matter how big pres_fac and acc_fac become.

litghost, 2019-11-13:
@vaughnbetz Sorry for the lengthy discussion, I've isolated the problem to the initial bound box sizes.  By relaxing the initial bb factor (from 6 to 10), the router behaves well.  There is an open question why the router didn't open the bounding boxes on its own, but that probably should be examined in another issue.

closed 2019-11-13 by litghost

vaughnbetz, 2019-11-13:
Ah, that is good.  We should still figure that out though!

litghost, 2019-11-13:
```
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 110: No routing path for connection to sink_rr 1725587, retrying with full device bounding box
Warning 111: 90 timing endpoints were not constrained during timing analysis
   1    9.7     0.0    0 6.8e+07    5444   16741   11687 ( 0.330%)  732255 ( 8.6%)   19.117 -1.928e+04    -19.117 -0.0008915     -0.001      N/A
   2   92.1     0.5  112 5.3e+08    4319   13576    9905 ( 0.280%)  484620 ( 5.7%)   18.991 -2.114e+04    -18.991      0.000      0.000      N/A
   3  193.2     0.6   35 1.1e+09    4086   13067    9178 ( 0.259%)  492509 ( 5.8%)   18.963 -2.206e+04    -18.963      0.000      0.000      N/A
   4  249.0     0.8   59 1.4e+09    3871   12680    8166 ( 0.231%)  501236 ( 5.9%)   19.007 -2.262e+04    -19.007      0.000      0.000      N/A
   5  269.4     1.1   35 1.5e+09    3585   11961    6634 ( 0.187%)  508381 ( 6.0%)   19.001 -2.291e+04    -19.001      0.000      0.000      N/A
   6  288.4     1.4   38 1.5e+09    3288   10876    5258 ( 0.149%)  515671 ( 6.1%)   18.991 -2.328e+04    -18.991      0.000      0.000      N/A
   7  244.9     1.9   25 1.3e+09    2849    9475    3843 ( 0.109%)  523968 ( 6.2%)   18.994 -2.376e+04    -18.994      0.000      0.000      N/A
   8  205.7     2.4   24 1.1e+09    2424    7955    2659 ( 0.075%)  531127 ( 6.2%)   18.968 -2.369e+04    -18.968      0.000      0.000      N/A
   9  159.4     3.1   20 8.5e+08    1927    6261    1806 ( 0.051%)  537207 ( 6.3%)   18.960 -2.393e+04    -18.960      0.000      0.000      N/A
  10  107.2     4.1   14 5.7e+08    1444    4702    1154 ( 0.033%)  541073 ( 6.4%)   18.975 -2.397e+04    -18.975      0.000      0.000       32
  11   72.8     5.3   12 3.9e+08    1049    3528     725 ( 0.020%)  543196 ( 6.4%)   18.970 -2.408e+04    -18.970      0.000      0.000       29
  12   48.3     6.9    7 2.5e+08     709    2432     478 ( 0.014%)  545296 ( 6.4%)   18.980 -2.411e+04    -18.980      0.000      0.000       28
  13   29.6     9.0    4 1.5e+08     504    1691     302 ( 0.009%)  545963 ( 6.4%)   18.986 -2.414e+04    -18.986      0.000      0.000       27
  14   19.7    11.6    3 9.8e+07     343    1118     223 ( 0.006%)  546861 ( 6.4%)   19.000 -2.415e+04    -19.000      0.000      0.000       26
  15   14.4    15.1    1 7.4e+07     264     854     153 ( 0.004%)  547634 ( 6.4%)   19.002 -2.417e+04    -19.002      0.000      0.000       27
  16   11.4    19.7    1 5.5e+07     190     601     106 ( 0.003%)  548242 ( 6.5%)   18.986 -2.417e+04    -18.986      0.000      0.000       27
  17   10.7    25.6    0 4.3e+07     142     457      83 ( 0.002%)  548587 ( 6.5%)   18.986 -2.416e+04    -18.986      0.000      0.000       27
  18    8.7    33.3    0 3.6e+07     115     344      54 ( 0.002%)  549381 ( 6.5%)   18.980 -2.416e+04    -18.980      0.000      0.000       28
  19    8.6    43.3    1 3.3e+07      82     251      37 ( 0.001%)  549656 ( 6.5%)   18.980 -2.416e+04    -18.980      0.000      0.000       29
  20    5.8    56.2    0 2.4e+07      67     190      20 ( 0.001%)  550042 ( 6.5%)   18.980 -2.416e+04    -18.980      0.000      0.000       29
  21    7.1    73.1    2 2.4e+07      42     103      20 ( 0.001%)  550201 ( 6.5%)   18.980 -2.416e+04    -18.980      0.000      0.000       28
  22    4.0    95.0    0 1.4e+07      42     116      13 ( 0.000%)  550192 ( 6.5%)   19.121 -2.419e+04    -19.121      0.000      0.000       29
  23    6.4   123.5    2 2.4e+07     140     568      15 ( 0.000%)  550325 ( 6.5%)   19.103 -2.422e+04    -19.103      0.000      0.000       29
  24    4.6   160.6    0 1.8e+07     144     581      11 ( 0.000%)  550385 ( 6.5%)   19.097 -2.422e+04    -19.097      0.000      0.000       30
  25    6.1   208.8    0 2.1e+07     141     585       9 ( 0.000%)  550571 ( 6.5%)   19.086 -2.426e+04    -19.086      0.000      0.000       31
  26    3.5   271.4    1 1.5e+07     141     585       5 ( 0.000%)  551046 ( 6.5%)   19.409 -2.422e+04    -19.409      0.000      0.000       31
  27    2.3   352.8    0 1.0e+07     117     480       2 ( 0.000%)  551089 ( 6.5%)   19.133 -2.426e+04    -19.133      0.000      0.000       31
  28    1.3   458.7    0 6536958     130     554       2 ( 0.000%)  551255 ( 6.5%)   19.409 -2.426e+04    -19.409      0.000      0.000       31
  29    1.1   596.3    0 5200971     114     474       0 ( 0.000%)  551194 ( 6.5%)   19.409 -2.426e+04    -19.409      0.000      0.000       31
```