verilog-to-routing/vtr-verilog-to-routing #1047: Debugging router congestion speed

litghost, 2019-11-15:
#### Context

On the 7-series graph, congestion resolution for the picosoc on the 50T graph is quiet slow.  This issue documents the data I've gather attempting to diagnose the "failure".  

@vaughnbetz @kmurray If you have specific data to gather to debug this issue, that would be very value.

litghost, 2019-11-15:
The baseline router convergence I'm debugging is the following one:
```
Confirming router algorithm: TIMING_DRIVEN.        
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 109: No routing path for connection to sink_rr 1725587, retrying with full device bounding box   
Warning 110: 90 timing endpoints were not constrained during timing analysis                             
   1   11.0     0.0    0 6.8e+07    5458   16533   11679 ( 0.330%)  741454 ( 8.7%)   20.014 -1.983e+04    -20.014      0.000      0.000      N/A
   2   85.8     0.5  118 4.3e+08    4286   13425    9804 ( 0.277%)  466857 ( 5.5%)   20.033 -2.157e+04    -20.033      0.000      0.000      N/A
   3  208.3     1.0   22 9.2e+08    3948   12725    8110 ( 0.229%)  477884 ( 5.6%)   19.891 -2.243e+04    -19.891      0.000      0.000      N/A
   4  278.7     2.0   57 1.3e+09    3493   11789    5620 ( 0.159%)  493624 ( 5.8%)   19.960 -2.313e+04    -19.960      0.000      0.000      N/A
   5  254.7     4.0   40 1.1e+09    2725    9644    2995 ( 0.085%)  511249 ( 6.0%)   19.927 -2.358e+04    -19.927      0.000      0.000      N/A                                                                   
   6  179.4     8.0   41 8.0e+08    1787    6521    1132 ( 0.032%)  524349 ( 6.2%)   19.966 -2.380e+04    -19.966      0.000      0.000      N/A
   7   97.0    16.0   24 4.1e+08     936    3339     424 ( 0.012%)  530892 ( 6.2%)   19.946 -2.386e+04    -19.946      0.000      0.000      N/A
   8   45.5    32.0   13 1.8e+08     491    1591     169 ( 0.005%)  534270 ( 6.3%)   19.943 -2.389e+04    -19.943      0.000      0.000      N/A
   9   40.0    64.0   12 1.4e+08     286     998      72 ( 0.002%)  535903 ( 6.3%)   19.928 -2.389e+04    -19.928      0.000      0.000      N/A
  10   27.6   128.0    8 8.9e+07     195     734      36 ( 0.001%)  536822 ( 6.3%)   19.928 -2.395e+04    -19.928      0.000      0.000       14
  11   23.3   256.0    2 7.5e+07     176     713      19 ( 0.001%)  537304 ( 6.3%)   20.116 -2.408e+04    -20.116      0.000      0.000       14
  12   24.0   512.0    0 7.4e+07     181     716      22 ( 0.001%)  537394 ( 6.3%)   19.928 -2.397e+04    -19.928      0.000      0.000       14
  13   19.8  1024.0    0 6.2e+07     171     688       8 ( 0.000%)  537525 ( 6.3%)   20.232 -2.406e+04    -20.232      0.000      0.000       16
  14   14.4  2048.0    1 5.0e+07     161     693       5 ( 0.000%)  538011 ( 6.3%)   22.397 -2.428e+04    -22.397      0.000      0.000       16
  15    3.1  4096.0    0 1.2e+07      56     197       2 ( 0.000%)  538117 ( 6.3%)   22.397 -2.433e+04    -22.397      0.000      0.000       17
  16    1.8  8192.0    1 8053712      53     189       0 ( 0.000%)  538153 ( 6.3%)   22.397 -2.433e+04    -22.397      0.000      0.000       17
Restoring best routing
Critical path: 22.397 ns
Successfully routed after 16 routing iterations.
```

Parameters for this routing were as follows:

LUT equivilance | pres_fac_mult | acc_fac | max_criticality | A* factor | BB factor | first_iter_pres_fac | initial_pres_fac | Reconvergence count
-- | -- | -- | -- | -- | -- | -- | -- | --
Off | 2 | 1 | 0.99 | 1.2 | 10 | 0 | 0.5 | 1

litghost, 2019-11-15:
I've adding debugging display the overused nodes at each iteration, and the over used node log is attached: 
[overused_nodes.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3852443/overused_nodes.tar.gz)

litghost, 2019-11-15:
I've added debugging to display time take for each iteration per connection, so that slow connections can be isolated and debugged.  I've gather the data for the first two iterations, and pasted them here:
```
Iter 1, took 10.7, re-routed 16533 connections
Longest net, from rr node 888, took 0.8406039999999999
Longest connection, from rr node 888 to 1725587, took 0.038923
Iter 2, took 65.0, re-routed 13425 connections
Longest net, from rr node 799450, took 0.482804
Longest connection, from rr node 799450 to 3081251, took 0.482804
Saw 2 iterations
```

I enabled sink debugging on 3081251 to determine the behavioral change from iteration 1 (pres_fac = 0) and iteration 2 (pres_fac = 0.5).  Here is the log for iteration 1:
```
1. 799538 "CLBLM_R_X27Y132/CLBLM_M_AQ", cost = 4.93243e-14, first explored at pop 2 (5.405405405405405 %) @ 20
2. 794810 "CLBLM_R_X27Y132/CLBLM_M_AQ", cost = 1.08912e-09, first explored at pop 3 (8.108108108108109 %) @ 35
3. 794654 "CLBLM_R_X27Y132/CLBLM_LOGIC_OUTS4", cost = 1.09382e-09, first explored at pop 4 (10.81081081081081 %) @ 50
4. 795010 "INT_R_X27Y132/NR1BEG0", cost = 1.08307e-09, first explored at pop 8 (21.62162162162162 %) @ 1290
5. 794981 "HCLK_R_X69Y130/HCLK_LV9", cost = 1.07995e-09, first explored at pop 9 (24.324324324324323 %) @ 1555
6. 1224238 "INT_R_X27Y103/LVB0", cost = 1.07386e-09, first explored at pop 10 (27.027027027027028 %) @ 1920
7. 1196361 "BRKH_INT_X27Y99/BRKH_INT_LVB9", cost = 1.07391e-09, first explored at pop 11 (29.72972972972973 %) @ 2105
8. 1174506 "INT_R_X27Y79/LVB0", cost = 1.07287e-09, first explored at pop 12 (32.432432432432435 %) @ 2289
9. 1490397 "HCLK_R_X69Y78/HCLK_LVB8", cost = 1.0721e-09, first explored at pop 13 (35.13513513513514 %) @ 2473
10. 1456616 "INT_R_X27Y55/LVB0", cost = 1.0713e-09, first explored at pop 14 (37.83783783783784 %) @ 2657
11. 1993538 "BRKH_INT_X27Y49/BRKH_INT_LVB7", cost = 1.07053e-09, first explored at pop 15 (40.54054054054054 %) @ 2841
12. 1957383 "INT_R_X27Y31/LVB0", cost = 1.06973e-09, first explored at pop 16 (43.24324324324324 %) @ 3025
13. 1931621 "HCLK_R_X69Y26/HCLK_LVB6", cost = 1.0689e-09, first explored at pop 17 (45.945945945945944 %) @ 3209
14. 2986598 "INT_R_X27Y10/LVB3", cost = 1.06822e-09, first explored at pop 18 (48.648648648648646 %) @ 3393
15. 2849051 "CLBLL_L_X28Y7/CLBLL_SE4BEG2", cost = 1.03307e-09, first explored at pop 23 (62.16216216216216 %) @ 4157
16. 3079378 "CLBLL_L_X28Y7/CLBLL_SE4BEG2", cost = 1.03426e-09, first explored at pop 25 (67.56756756756756 %) @ 4366
17. 3082399 "INT_R_X29Y3/NR1BEG2", cost = 1.03138e-09, first explored at pop 26 (70.27027027027027 %) @ 4560
18. 3082327 "CLBLM_R_X29Y4/CLBLM_IMUX28", cost = 1.01291e-09, first explored at pop 33 (89.1891891891892 %) @ 4897
19. 3082489 "CLBLM_R_X29Y4/CLBLM_M_C4", cost = 1.01301e-09, first explored at pop 36 (97.29729729729729 %) @ 4935
20. 3081321 "CLBLM_R_X29Y4/CLBLM_M_C4", cost = 9.5354e-10, first explored at pop 37 (100.0 %) @ 4943
```

And iteration 2:
```
1. 799538 "CLBLM_R_X27Y132/CLBLM_M_AQ", cost = 4.15763e-12, first explored at pop 2 (0.00029450309964512375 %) @ 20
2. 794810 "CLBLM_R_X27Y132/CLBLM_M_AQ", cost = 1.00194e-09, first explored at pop 3 (0.00044175464946768567 %) @ 35
3. 794654 "CLBLM_R_X27Y132/CLBLM_LOGIC_OUTS4", cost = 1.01099e-09, first explored at pop 4 (0.0005890061992902475 %) @ 50
4. 798140 "INT_R_X27Y132/NN2BEG0", cost = 1.14636e-09, first explored at pop 992 (0.14607353742398138 %) @ 238060
5. 759168 "CLBLL_L_X26Y133/CLBLL_WW2END3", cost = 1.17182e-09, first explored at pop 1112 (0.1637437234026888 %) @ 269683
6. 757874 "INT_L_X24Y132/SW2END3", cost = 1.11555e-09, first explored at pop 1113 (0.16389097495251137 %) @ 270008
7. 757637 "HCLK_L_X63Y130/HCLK_LVB5", cost = 1.12995e-09, first explored at pop 1116 (0.16433272960197906 %) @ 271018
8. 1232710 "INT_L_X24Y108/LVB_L0", cost = 1.14108e-09, first explored at pop 1119 (0.16477448425144675 %) @ 271812
9. 1246310 "BRKH_INT_X24Y99/BRKH_INT_LVB_L4", cost = 1.18421e-09, first explored at pop 2389 (0.35178395252610034 %) @ 527945
10. 1181558 "INT_L_X24Y84/LVB_L0", cost = 1.22735e-09, first explored at pop 6937 (1.0214840011191118 %) @ 1237971
11. 1512150 "HCLK_L_X63Y78/HCLK_LVB3", cost = 1.2975e-09, first explored at pop 21720 (3.198303662146044 %) @ 2999704
12. 1436244 "INT_L_X24Y60/LVB_L0", cost = 1.3947e-09, first explored at pop 75909 (11.17771789548085 %) @ 8752514
13. 1448885 "INT_L_X24Y54/SS6END2", cost = 1.49519e-09, first explored at pop 222920 (32.8253154864455 %) @ 19573777
14. 2005451 "INT_L_X24Y52/SS2END2", cost = 1.519e-09, first explored at pop 240375 (35.39559128859831 %) @ 20119660
15. 2006281 "BRKH_INT_X24Y49/BRKH_INT_SS6C2", cost = 1.57945e-09, first explored at pop 354073 (52.137798000323954 %) @ 25392136
16. 2012692 "INT_L_X24Y44/SS2END2", cost = 1.58972e-09, first explored at pop 354075 (52.1380925034236 %) @ 25392646
17. 1952108 "INT_L_X24Y38/SS6END2", cost = 1.64466e-09, first explored at pop 450687 (66.36435923488095 %) @ 28939400
18. 1946812 "INT_L_X24Y32/SS6END2", cost = 1.70612e-09, first explored at pop 536061 (78.93581304943234 %) @ 31775698
19. 1938497 "INT_L_X24Y31/SL1END2", cost = 1.72205e-09, first explored at pop 560004 (82.46145690683394 %) @ 32436867
20. 1936214 "INT_L_X24Y30/SE2A2", cost = 1.70512e-09, first explored at pop 560005 (82.46160415838376 %) @ 32437112
21. 1936042 "INT_L_X24Y29/SW2END2", cost = 1.79432e-09, first explored at pop 634476 (93.42757432521978 %) @ 35110609
[... cut ...]
    New back cost: 1.29328e-09

  Popping node INT_L_X24Y30/SE2A2 (1936214) (cost: 1.70512e-09)
    Better cost to 1936214
    New total cost: 1.70512e-09
    New back cost: 1.34561e-09

  Popping node INT_L_X24Y29/SW2END2 (1936042) (cost: 1.79432e-09)
    Better cost to 1936042
    New total cost: 1.79432e-09
    New back cost: 1.42483e-09

  Popping node HCLK_L_X63Y26/HCLK_LVB8 (2966236) (cost: 1.83635e-09)
    Better cost to 2966236
    New total cost: 1.83635e-09
    New back cost: 1.53349e-09
```

litghost, 2019-11-15:
Fun facts about iteration 2:
```
grep "Worse cost to" route_debug_iter_2_net_5395_sink_3081251_annotate.log | wc
 512217 2048868 13198963
```

So of the 679110 pops, 75% were not useful!

litghost, 2019-11-15:
Completed iteration with connection timing is below, worst conn and net is displayed, along with all connections that took long than 1 second.
```
Iter 1, took 10.7, re-routed 16533 connections                                                           
Longest net, from rr node 888, took 0.8406039999999999            
Longest connection, from rr node 888 to 1725587, took 0.038923
Iter 2, took 65.0, re-routed 13425 connections
Longest net, from rr node 799450, took 0.482804
Longest connection, from rr node 799450 to 3081251, took 0.482804 
Iter 3, took 153.6, re-routed 12725 connections
Longest net, from rr node 786464, took 0.7651929999999998
Longest connection, from rr node 786464 to 2997527, took 0.586622
Iter 4, took 213.0, re-routed 11789 connections
Longest net, from rr node 2841562, took 3.358916999999999
Longest connection, from rr node 786458 to 3077321, took 0.730649 
Iter 5, took 193.5, re-routed 9644 connections
Longest net, from rr node 2841562, took 4.262686
Longest connection, from rr node 786464 to 2997527, took 0.954452
Iter 6, took 132.3, re-routed 6521 connections
Longest net, from rr node 2841562, took 2.9753280000000006
Longest connection, from rr node 1695451 to 1920516, took 0.687037
Iter 7, took 71.6, re-routed 3339 connections
Longest net, from rr node 788343, took 2.5174080000000005
Longest connection, from rr node 1943199 to 3379222, took 0.804792
Iter 8, took 33.7, re-routed 1591 connections
Longest net, from rr node 788343, took 3.1899210000000005
Longest connection, from rr node 2623366 to 1557506, took 1.242026
 Conn #1, from 2623366 to 1557506 took 1.242026
Iter 9, took 29.3, re-routed 998 connections
Longest net, from rr node 788343, took 3.8052030000000006
Longest connection, from rr node 2623366 to 1557506, took 1.780764
 Conn #1, from 2623366 to 1557506 took 1.780764
 Conn #2, from 1882983 to 1979698 took 1.58956
 Conn #3, from 1570218 to 1982463 took 1.13363
 Conn #4, from 1570217 to 3009679 took 1.085349
 Conn #5, from 2030202 to 1895567 took 1.054833
Iter 10, took 19.5, re-routed 734 connections
Longest net, from rr node 788343, took 3.6643129999999995
Longest connection, from rr node 2623366 to 1557506, took 1.923264
 Conn #1, from 2623366 to 1557506 took 1.923264
 Conn #2, from 1882983 to 1979698 took 1.718979
Iter 11, took 16.6, re-routed 713 connections
Longest net, from rr node 788343, took 3.2276120000000006
Longest connection, from rr node 1943199 to 1983437, took 2.505106
 Conn #1, from 1943199 to 1983437 took 2.505106
 Conn #2, from 3328601 to 1895733 took 1.094892
Iter 12, took 17.1, re-routed 716 connections
Longest net, from rr node 788343, took 3.241411
Longest connection, from rr node 1943199 to 1983437, took 2.641494
 Conn #1, from 1943199 to 1983437 took 2.641494
 Conn #2, from 3328601 to 1895733 took 1.175568
Iter 13, took 13.8, re-routed 688 connections
Longest net, from rr node 788343, took 2.8813340000000003
Longest connection, from rr node 1943199 to 1983437, took 2.725193
 Conn #1, from 1943199 to 1983437 took 2.725193
Iter 14, took 10.3, re-routed 693 connections
Longest net, from rr node 788343, took 3.005588
Longest connection, from rr node 3037018 to 1987699, took 0.969057
Iter 15, took 2.3, re-routed 197 connections
Longest net, from rr node 1571284, took 0.500571
Longest connection, from rr node 878 to 1972628, took 0.426693
Iter 16, took 1.4, re-routed 189 connections
Longest net, from rr node 3037016, took 0.280573
Longest connection, from rr node 1988068 to 3008742, took 0.233737
Saw 16 iterations
```

vaughnbetz, 2019-11-15:
Thanks @litghost 
I see some strange data in these logs that I think should be checked out (or explained, if you know why they're happening).  

1.
799538 "CLBLM_R_X27Y132/CLBLM_M_AQ", cost = **4.93243e-14**, first explored at pop 2 (5.405405405405405 %) @ 20
2. 794810 "CLBLM_R_X27Y132/CLBLM_M_AQ", cost = **1.08912e-09**, first explored at pop 3 (8.108108108108109 %) @ 35

How do we get a cost that is 10-14?  Is this a time critical connection (criticality = 0.99) and the node is modeled as having 0 delay? Even then I'd expect cost = 0.01 * base_cost, which would still be 10-13 if the average delay (normalization factor for base cost) is 100 ps.

2. 
It looks like the HCLK lines are being explored, e.g.:
794981 "HCLK_R_X69Y130/HCLK_LV9", cost = 1.07995e-09, first explored at pop 9 (24.324324324324323 %) @ 1555

Do the captured traces include any clock nets?  Do you have reasonable delay numbers on the clock nodes, and a reasonable base cost?
The clock nodes may look like very fast wires (especially if their timing is not accurate) but may not lead to many input pins, and hence lead to congestion.
If the clock nodes are getting expanded and causing the problems I listed above, you probably want to stop expansion of the clock nodes when routing regular connections.  This could be done by adding an expand flag to the rr_indexed_data, and updating it for clock vs. non-clock nets before starting the routing of a net (that will be fast, and can be made pretty architecture-independent/data-driven).

litghost, 2019-11-15:
> It looks like the HCLK lines are being explored, e.g.:
> 794981 "HCLK_R_X69Y130/HCLK_LV9", cost = 1.07995e-09, first explored at pop 9 (24.324324324324323 %) @ 1555

That isn't a clock wire, it is a portion of a long vertical length 9 wire that is traversing the HCLK tile.  Clock wires are named like `HCLK_CK_` or `GCLK_L_B0` or `CLK_L0`.

litghost, 2019-11-15:
> 799538 "CLBLM_R_X27Y132/CLBLM_M_AQ", cost = **4.93243e-14**, first explored at pop 2 (5.405405405405405 %) @ 20
> 2. 794810 "CLBLM_R_X27Y132/CLBLM_M_AQ", cost = **1.08912e-09**, first explored at pop 3 (8.108108108108109 %) @ 35

Those early nodes are the nodes modelling the output site pin delays.  The lookahead for those nodes are very bad (and very simple), but I expect they shouldn't matter because there is no choice, which is why the router very quickly gets out of that space.

litghost, 2019-11-15:
Profiling data confirms some things:
 - Dead pops are relatively cheap, about 8.5% of the connection runtime total.  So even if the number of dead pops were lowered.
 - Dead pushes are problematic, because an unused push still requires the computation of the lookahead and congestion values. Given that only 28% of pushed nodes were even popped, that means roughly 60% of the connection runtime was wasted.  It is unclear if it is possible to avoid unhelpful pushes before spending the runtime cost though :/

litghost, 2019-11-15:
Flame graph for profile of relevant run:

[image]

litghost, 2019-11-15:
Breakdown of time:
- 91% of the time is spent expanding nodes
- 9% is spent popping nodes

litghost, 2019-11-15:
Important observation about the lookahead behavior:

```
grep "Lookahead returned" route_debug_iter_2_net_5395_sink_3081251.log | wc
3332957 36662527 227162794
grep "Lookahead returned" route_debug_iter_2_net_5395_sink_3081251.log | sort | uniq -c | wc
  39006  468072 2931648
```

Lookahead queries account ~21% of the runtime, because there were 33 million lookups (total).  However only 39k unique queries were made, so it is very likely a lookup LRU cache would do very well.

vaughnbetz, 2019-11-15:
So the first opin pop is just the lookahead cost of that opin, and the second has the real base cost of that opin added?  If so, that should be OK (i.e. the base cost is reasonable / not near 0).

litghost, 2019-11-15:
> So the first opin pop is just the lookahead cost of that opin, and the second has the real base cost of that opin added? If so, that should be OK (i.e. the base cost is reasonable / not near 0).

Yep!

vaughnbetz, 2019-11-15:
I believe we're doing too much expansion for a design of this size; not sure why (maybe try higher astar_fac or ensuring the lookahead understands how many nodes are necessary to reach the destination once you are in the various final muxes?).  I think those will be higher impact than low level code optimization at this point.

To test that theory, you could run the same design on the stratixiv-like architecture and compare the router runtime and heap ops.  I suspect it has significantly lower heap ops and run time, but that would be good to quantify.

litghost, 2019-11-15:
> I believe we're doing too much expansion for a design of this size

I agree, but as far as I can tell, the reason for the excessive expansion is that the cost map is flatish.  So once congestion is introduced, the router spills into parallel reasonable paths.

vaughnbetz, 2019-11-15:
That should be addressable with good base_cost prediction, possibly in combination with a higher astar_fac though.  If you turn off all timing costs (set max_crit to 0), but predict the base_costs well, the router should find it easier to converge, but it seems the opposite is happening.

litghost, 2019-11-15:
> That should be addressable with good base_cost prediction, possibly in combination with a higher astar_fac though. If you turn off all timing costs (set max_crit to 0), but predict the base_costs well, the router should find it easier to converge, but it seems the opposite is happening.

I'll start look at the router behavior with only base cost lookahead (e.g. max_crit = 0).

litghost, 2019-11-15:
> That should be addressable with good base_cost prediction

So I believe that we have good base cost prediction, based on the following evidence:

Max_criticality = 0.99 and pres_fac = 0:
```
   1   11.0     0.0    0 6.8e+07    5458   16533   11679 ( 0.330%)  741454 ( 8.7%)   20.014 -1.983e+04    -20.014      0.000      0.000      N/A
```

Max_criticality = 0.0 and pres_fac = 0:
```
   1   12.6     0.0    0 9.3e+07    5458   16533   16540 ( 0.467%)  405368 ( 4.8%)   39.837 -3.456e+04    -39.837     -25.61     -2.506      N/A
```

Should that first iteration at max_crit = 0 be even faster?

litghost, 2019-11-15:
I have a gut feeling there is a problem around how congestion through non-configurable node sets are computed.  @kmurray can you explain something about the expansion step that confuses me.

1. When a non-configurable node set is encountered, the congestion code for that entry includes all nodes in that set, see https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/30fe417dd191b2233b534cc42ee3a4d58dcf5bfa/vpr/src/route/route_common.cpp#L721-L737

Therefore the backwards congestion cost of the entire node set is seen.  This is not a bug in of itself, however!

2. I don't believe the non-configurable node set is fully expanded at this time, instead the other nodes of the non-configurable node set are placed in the heap (which is fine).  When they are expanded, their congestion is added again! See https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/30fe417dd191b2233b534cc42ee3a4d58dcf5bfa/vpr/src/route/route_timing.cpp#L2020 

As a result the congestion cost of non-configurable nodes will be counted multiple times, as a function of the non-configurable node set size!

I believe the solution is to only add backwards congestion cost of the non-configurable node when entering the non-configurable node set.  The expansion can continue as normal.

Am I missing something?

litghost, 2019-11-15:
I tried changing the behavior around https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1047#issuecomment-554536454 and it didn't have a large impact, so that isn't the whole picture.  Still digging.

kmurray, 2019-11-15:
@litghost I believe the logic was to push the nodes individually into the heap in order to account for timing differences between the different nodes. I think you have a draft of the VTR 8 paper, Section 9.4 describes the method used for non-configurable edges.

litghost, 2019-11-15:
> @litghost I believe the logic was to push the nodes individually into the heap in order to account for timing differences between the different nodes.

Individually pushing makes sense, I'm wondering if the backwards congestion cost computation is now wrong.

kmurray, 2019-11-15:
Actually, looking at 

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/30fe417dd191b2233b534cc42ee3a4d58dcf5bfa/vpr/src/route/route_common.cpp#L724-L737

Is the continue condition wrong? As written it will only add the congestion cost of other nodes in the set when node == inode (which was already done outside). That would mean we aren't counting the full congestion cost of the other nodes in the non-configurable node set.

litghost, 2019-11-15:
> Section 9.4 describes the method used for non-configurable edges.

Ya, I think the algorithm in Section 9.4 has the same issue.  Let me layout an example for clarity.

Let's imagine a U shaped wire, so it will have two CHANY and one CHANX, yes?  They will be connected via shorts, so they will be a non-configurable set.  Lets assume the router enters on one CHANY, and exits the other CHANY.  Algorithm 4 and 5 will behave as follows:

Diagram of u-shaped wire:
```
|                      |
|                      |
1                      3
|                      |
|                      |
 --------- 2 ----------
```
1. Router pops node 1.  `sum congestion(nodes)` will return the sum of the single congestion cost of node 1 and 2 and 3.  This is fine, as if only node 3 is overused, you want the router to avoid the entire set.  Backwards cost will node be `cost prior to node 1 + congestion cost of node 1 + congestion cost of node 2 + congestion cost of node 3`.  Router will push node 2 into the heap, with the cost of forward + backwards cost
2. Router pops node 2.  `sum congestion(nodes)` will return the same sum!  Now the backwards cost will be `cost prior to node 1 + 2 *(congestion cost of node 1 + congestion cost of node 2 + congestion cost of node 3)`.  Router will push node 3 into heap, etc.

Does that make sense?

litghost, 2019-11-15:
> Is the continue condition wrong? As written it will only add the congestion cost of other nodes in the set when node == inode (which was already done outside). That would mean we aren't counting the full congestion cost of the other nodes in the non-configurable node set.

No that particular code is actually correct.  The node == inode condition is to prevent a double counting of the congestion cost of inode, which was already accounted for here: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/30fe417dd191b2233b534cc42ee3a4d58dcf5bfa/vpr/src/route/route_common.cpp#L724

litghost, 2019-11-15:
Here you can see a routing trace for the criticality = 0:
```
1. 1001908 "CLBLM_R_X29Y129/CLBLM_M_CQ", cost = 0.0, first explored at pop 1 (0.00020202591588448966 %) @ 88
2. 1003058 "CLBLM_R_X29Y129/CLBLM_M_CQ", cost = 8.93758e-10, first explored at pop 147 (0.02969780963501998 %) @ 31830
3. 1002902 "CLBLM_R_X29Y129/CLBLM_LOGIC_OUTS6", cost = 8.58244e-10, first explored at pop 4 (0.0008081036635379586 %) @ 114
4. 1002920 "INT_R_X29Y127/SS2END2", cost = 8.55285e-10, first explored at pop 9 (0.001818233242960407 %) @ 1298
5. 1003747 "INT_R_X29Y125/SS2END2", cost = 8.52325e-10, first explored at pop 13 (0.0026263369064983655 %) @ 2298
6. 1004510 "INT_L_X28Y125/WL1END1", cost = 8.49366e-10, first explored at pop 19 (0.0038384924018053036 %) @ 3888
7. 778465 "CLBLL_L_X28Y125/CLBLL_WW2A1", cost = 9.9142e-10, first explored at pop 26963 (5.447224769993495 %) @ 3995880
8. 778512 "HCLK_L_X68Y130/HCLK_SS2A1", cost = 8.64163e-10, first explored at pop 420 (0.08485088467148566 %) @ 97673
9. 782718 "INT_L_X26Y121/SS2END1", cost = 8.96717e-10, first explored at pop 2967 (0.5994108924292808 %) @ 636903
10. 781835 "INT_L_X26Y119/SS2END1", cost = 8.8192e-10, first explored at pop 270 (0.05454699728881221 %) @ 59810
11. 1220489 "INT_L_X26Y117/SS2END1", cost = 8.7896e-10, first explored at pop 272 (0.054951049120581186 %) @ 60460
12. 1219264 "INT_L_X26Y115/SS2END1", cost = 8.76001e-10, first explored at pop 274 (0.05535510095235017 %) @ 61110
13. 1224435 "INT_L_X26Y113/SS2END1", cost = 8.73042e-10, first explored at pop 276 (0.055759152784119144 %) @ 61760
14. 1225230 "INT_L_X26Y111/SS2END1", cost = 8.70082e-10, first explored at pop 278 (0.056163204615888127 %) @ 62410
15. 1210141 "INT_L_X26Y109/SS2END1", cost = 8.84879e-10, first explored at pop 334 (0.06747665590541954 %) @ 76571
16. 1207700 "INT_L_X26Y107/SS2END1", cost = 8.99677e-10, first explored at pop 1004 (0.20283401954802763 %) @ 230960
17. 1206223 "INT_L_X26Y105/SS2END1", cost = 8.90798e-10, first explored at pop 1005 (0.2030360454639121 %) @ 231285
18. 1203793 "INT_L_X26Y103/SS2END1", cost = 8.87839e-10, first explored at pop 1006 (0.2032380713797966 %) @ 231610
19. 1196629 "INT_L_X26Y101/SS2END1", cost = 8.8488e-10, first explored at pop 1007 (0.20344009729568108 %) @ 231935
20. 1197349 "BRKH_INT_X26Y99/BRKH_INT_SS2END1", cost = 8.87839e-10, first explored at pop 1012 (0.20445022687510353 %) @ 233480
21. 1199721 "INT_L_X26Y97/SS2END1", cost = 8.96717e-10, first explored at pop 1020 (0.20606643420217946 %) @ 235607
22. 1169590 "INT_L_X26Y95/SS2END1", cost = 9.05595e-10, first explored at pop 1449 (0.2927355521166255 %) @ 328994
23. 1168101 "INT_L_X26Y94/SL1END1", cost = 9.03623e-10, first explored at pop 1451 (0.2931396039483945 %) @ 329644
24. 1181185 "CLBLL_L_X26Y94/CLBLL_WW2A1", cost = 9.7465e-10, first explored at pop 29774 (6.015119619544795 %) @ 4545698
25. 1181206 "INT_L_X24Y92/SS2END1", cost = 9.30258e-10, first explored at pop 18238 (3.6845486539013224 %) @ 2932372
26. 1178775 "INT_L_X24Y91/SL1END1", cost = 9.34204e-10, first explored at pop 18876 (3.8134411882356267 %) @ 3001103
27. 1178592 "INT_L_X24Y90/SL1END1", cost = 9.32231e-10, first explored at pop 18878 (3.813845240067396 %) @ 3001670
28. 1517462 "INT_L_X24Y88/SS2END1", cost = 9.29271e-10, first explored at pop 18880 (3.8142492918991646 %) @ 3002237
29. 1515024 "INT_L_X24Y87/SL1END1", cost = 9.33217e-10, first explored at pop 18123 (3.661315673574606 %) @ 2920327
30. 1514810 "INT_L_X24Y85/SS2END1", cost = 9.48015e-10, first explored at pop 21140 (4.270827861798112 %) @ 3278565
31. 1514443 "INT_L_X24Y83/SS2END1", cost = 9.56893e-10, first explored at pop 26169 (5.28681619278121 %) @ 3916729
32. 1511981 "INT_L_X24Y81/SS2END1", cost = 9.53933e-10, first explored at pop 26171 (5.287220244612979 %) @ 3917379
33. 1506455 "INT_L_X24Y79/SS2END1", cost = 9.50974e-10, first explored at pop 26173 (5.287624296444748 %) @ 3918029
34. 1503993 "INT_L_X24Y77/SS2END1", cost = 9.48015e-10, first explored at pop 26175 (5.288028348276517 %) @ 3918679
35. 1503626 "INT_L_X24Y75/SS2END1", cost = 9.45055e-10, first explored at pop 26177 (5.288432400108286 %) @ 3919329
36. 1501166 "HCLK_L_X63Y78/HCLK_SS2A1", cost = 9.7169e-10, first explored at pop 29842 (6.028857381824941 %) @ 4564162
37. 1436464 "INT_L_X24Y72/SL1END1", cost = 9.69717e-10, first explored at pop 29846 (6.029665485488478 %) @ 4565462
38. 1436289 "INT_L_X24Y71/SL1END1", cost = 9.73663e-10, first explored at pop 29858 (6.0320897964790925 %) @ 4569045
39. 1442762 "INT_L_X24Y70/SL1END1", cost = 9.7169e-10, first explored at pop 29861 (6.032695874226746 %) @ 4569940
40. 1442556 "INT_L_X24Y68/SS2END1", cost = 9.87967e-10, first explored at pop 30987 (6.260177055512681 %) @ 4759142
41. 1441658 "INT_L_X24Y62/SS6END1", cost = 1.02249e-09, first explored at pop 67581 (13.653113421389696 %) @ 7858606
42. 1449250 "INT_L_X24Y56/SS6END1", cost = 1.06294e-09, first explored at pop 134062 (27.083998335306454 %) @ 12345227
43. 2001645 "INT_L_X24Y50/SS6END1", cost = 1.07971e-09, first explored at pop 163195 (32.96961934276929 %) @ 13407573
44. 2008109 "BRKH_INT_X24Y49/BRKH_INT_SS2A1", cost = 1.07971e-09, first explored at pop 164903 (33.3146796071 %) @ 13452805
45. 2013581 "INT_L_X24Y46/SS2END1", cost = 1.08563e-09, first explored at pop 175095 (35.37372774179472 %) @ 13868748
46. 2012691 "INT_L_X24Y44/SS2END1", cost = 1.10339e-09, first explored at pop 218100 (44.06185225440719 %) @ 15118997
[... cut ...]
61. 2986729 "CLBLM_R_X27Y19/CLBLM_L_D3", cost = 1.19809e-09, first explored at pop 494985 (99.99979797408412 %) @ 23751959
62. 2996669 "CLBLM_R_X27Y19/CLBLM_L_D3", cost = 1.19094e-09, first explored at pop 494986 (100.0 %) @ 23751967
```

This is with A* factor = 1.2, and you can see that the cost does increase the whole way to the target.  So that does likely indicate an under-predict of the base cost.   However that under predication is somewhat expected, given that pres fac will cause the cost to increase in a way that is hard for a static lookahead to predict.

litghost, 2019-11-15:
Effectively the original lookahead for that connection was `8.58244e-10`, and the final congestion cost was `1.19094e-09`.  That is ~1.39 times, which is about what we would expect given that pres fac was 0.5, yes?

litghost, 2019-11-15:
I've isolated at least one gross mispredict, I'll keep digging.  Relevant data:

```

40. 1442556 "INT_L_X24Y68/SS2END1", cost = 9.87967e-10, first explored at pop 30987 (6.260177055512681 %) @ 4759142

      Expanding to node INT_L_X24Y68/SS2END1 (1442556) (RR node: 1442556 type: CHANY location: (83,85) <-> (83,87) track: 267 len: 2 longline: 0 seg_type: SS2BEG dir: BI_DIR capacity: 1)
Requested lookahead from node INT_L_X24Y68/SS2END1 (1442556) to 2996669
Lookahead returned SS2BEG (37) to IMUX (0) with distance (-6, -53)
Lookahead delay: 9.36697e-10
Lookahead congestion: 3.65e-10
Criticality: 0
Lookahead cost: 3.65e-10
Site pin delay: 3.46e-10
  Adding node  1442556 to heap from init route tree with cost 9.87967e-10 (RR node: 1442556 type: CHANY location: (83,85) <-> (83,87) track: 267 len: 2 longline: 0 seg_type: SS2BEG dir: BI_DIR capacity: 1)

  Popping node INT_L_X24Y68/SS2END1 (1442556) (cost: 9.87967e-10)
    Better cost to 1442556
    New total cost: 9.87967e-10
    New back cost: 5.49967e-10

41. 1441658 "INT_L_X24Y62/SS6END1", cost = 1.02249e-09, first explored at pop 67581 (13.653113421389696 %) @ 7858606  Popping node INT_L_X24Y62/SS6END1 (1441658) (cost: 1.02249e-09)
    Better cost to 1441658
    New total cost: 1.02249e-09
    New back cost: 6.14089e-10

      Expanding to node INT_L_X24Y62/SS6END1 (1441658) (RR node: 1441658 type: CHANY location: (83,87) <-> (83,93) track: 96 len: 6 longline: 0 seg_type: SS6BEG dir: BI_DIR capacity: 1)
Requested lookahead from node INT_L_X24Y62/SS6END1 (1441658) to 2996669
Lookahead returned SS6BEG (22) to IMUX (0) with distance (-6, -51)
Lookahead delay: 9.31863e-10
Lookahead congestion: 3.40338e-10
Criticality: 0
Lookahead cost: 3.40338e-10
Site pin delay: 3.46e-10
  Adding node  1441658 to heap from init route tree with cost 1.02249e-09 (RR node: 1441658 type: CHANY location: (83,87) <-> (83,93) track: 96 len: 6 longline: 0 seg_type: SS6BEG dir: BI_DIR capacity: 1)
```

The backwards cost from 1442556 -> 1441658 went from 5.49967e-10 to 6.14089e-10 (6.4122e-11), however the lookahead expected the forward cost to drop from 3.65e-10 to 3.40338e-10 (2.4662e-11).  So the lookahead under predicted the base cost change by a factor 3!  That's pretty bad.

litghost, 2019-11-16:
I've created a full router trace dump here: 
[route_debug_iter_2_net_5375_sink_2996576.json.tar.lzma.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3853662/route_debug_iter_2_net_5375_sink_2996576.json.tar.lzma.zip)

It is a in a JSON format, with a handful of keys.  The important ones are `final_route`, which has a list of the traceback rr nodes in the final route, and `node_costs` which contains data about the heap pushs and pops for each node.  Next week I'll try to through up some analyses of this data to try to isolate "weird" behavior from the lookahead and the router.  After that, hopefully that will narrow down where the problems are.

vaughnbetz, 2019-11-16:
For the first routing iteration, we shouldn't see any lookahead mispredicts due to congestion as congestion isn't being costed (pres_fac = 0).  Later iterations will get some mispredict *when there is congestion*, as you noted. If there is no congestion, there still should not be any mispredict, as pres_cost is 1 if there is no congestion, no matter what pres_fac is.  So we want to get rid of mispredicts when there isn't congestion.

The results with max_crit = 0 suggest the lookahead isn't completely broken when you are predicting base costs, but do suggest that they're predicted a bit worse than timing. I'm also curious of what happens in later iterations with max_crit = 0 or 0.1, as the router should converge faster (pays more attention to congestion). 

I think a comparison to the Stratix IV architecture on the same design would also be useful, to get some very rough targets for heap operations and route time on that architecture by iteration, as we know the router is working well on it.

kmurray, 2019-11-18:
> I think a comparison to the Stratix IV architecture on the same design would also be useful, to get some very rough targets for heap operations and route time on that architecture by iteration, as we know the router is working well on it.

Yep agreed, this would give a good ball park numbers which may point us in the correct direction.

litghost, 2019-11-18:
Here is my first shot at an analysis of the lookahead and router behavior:
```
1. 1001908 "CLBLM_R_X29Y129/CLBLM_M_CQ", cost = 0.0, first explored at pop 1 (0.00 %)
2. 1003058 "CLBLM_R_X29Y129/CLBLM_M_CQ", cost = 8.93758e-10, first explored at pop 145 (0.03 %)
3. 1002902 "CLBLM_R_X29Y129/CLBLM_LOGIC_OUTS6", cost = 8.58244e-10, first explored at pop 4 (0.00 %)
4. 1002920 "INT_R_X29Y127/SS2END2", cost = 8.55285e-10, first explored at pop 9 (0.00 %)
5. 1003577 "INT_L_X28Y127/WL1END1", cost = 8.52325e-10, first explored at pop 20 (0.00 %)
6. 777465 "CLBLL_L_X28Y127/CLBLL_WW2A1", cost = 1.14827e-09, first explored at pop 353555 (69.72 %)
7. 777490 "INT_L_X26Y125/SS2END1", cost = 8.67123e-10, first explored at pop 708 (0.14 %)
8. 778512 "HCLK_L_X68Y130/HCLK_SS2A1", cost = 8.64163e-10, first explored at pop 404 (0.08 %)
9. 782718 "INT_L_X26Y121/SS2END1", cost = 8.96717e-10, first explored at pop 2606 (0.51 %)
10. 781835 "INT_L_X26Y119/SS2END1", cost = 8.8192e-10, first explored at pop 362 (0.07 %)
11. 1220489 "INT_L_X26Y117/SS2END1", cost = 8.7896e-10, first explored at pop 364 (0.07 %)
12. 1219264 "INT_L_X26Y115/SS2END1", cost = 8.76001e-10, first explored at pop 366 (0.07 %)
13. 1224435 "INT_L_X26Y113/SS2END1", cost = 8.73042e-10, first explored at pop 368 (0.07 %)
14. 1225230 "INT_L_X26Y111/SS2END1", cost = 8.70082e-10, first explored at pop 370 (0.07 %)
15. 1210178 "INT_L_X26Y110/SL1END1", cost = 8.79947e-10, first explored at pop 386 (0.08 %)
16. 1207506 "INT_L_X26Y108/SS2END1", cost = 8.94744e-10, first explored at pop 887 (0.17 %)
17. 1206056 "INT_L_X26Y106/SS2END1", cost = 8.91785e-10, first explored at pop 889 (0.18 %)
18. 1203626 "INT_L_X26Y104/SS2END1", cost = 8.88825e-10, first explored at pop 890 (0.18 %)
19. 1196027 "INT_L_X26Y102/SS2END1", cost = 8.97704e-10, first explored at pop 1056 (0.21 %)
20. 1196804 "INT_L_X26Y100/SS2END1", cost = 8.94744e-10, first explored at pop 1058 (0.21 %)
21. 1200455 "BRKH_INT_X26Y99/BRKH_INT_SS2A1", cost = 8.91785e-10, first explored at pop 1060 (0.21 %)
22. 1169407 "INT_L_X26Y96/SS2END1", cost = 9.06582e-10, first explored at pop 1765 (0.35 %)
23. 1168253 "INT_L_X26Y94/SS2END1", cost = 9.03623e-10, first explored at pop 1767 (0.35 %)
24. 1181185 "CLBLL_L_X26Y94/CLBLL_WW2A1", cost = 9.7465e-10, first explored at pop 34651 (6.83 %)
25. 1181206 "INT_L_X24Y92/SS2END1", cost = 9.30258e-10, first explored at pop 16172 (3.19 %)
26. 1178775 "INT_L_X24Y91/SL1END1", cost = 9.34204e-10, first explored at pop 16760 (3.30 %)
27. 1178592 "INT_L_X24Y90/SL1END1", cost = 9.32231e-10, first explored at pop 16762 (3.31 %)
28. 1517462 "INT_L_X24Y88/SS2END1", cost = 9.29271e-10, first explored at pop 16764 (3.31 %)
29. 1514993 "INT_L_X24Y86/SS2END1", cost = 9.3815e-10, first explored at pop 16148 (3.18 %)
30. 1514657 "INT_L_X24Y85/SL1END1", cost = 9.7169e-10, first explored at pop 34482 (6.80 %)
31. 1514474 "INT_L_X24Y84/SL1END1", cost = 9.69717e-10, first explored at pop 34488 (6.80 %)
32. 1512195 "INT_L_X24Y83/SL1END1", cost = 9.97339e-10, first explored at pop 41361 (8.16 %)
33. 1512012 "INT_L_X24Y82/SL1END1", cost = 1.00128e-09, first explored at pop 45599 (8.99 %)
34. 1506669 "INT_L_X24Y81/SL1END1", cost = 9.69717e-10, first explored at pop 34592 (6.82 %)
35. 1506486 "INT_L_X24Y80/SL1END1", cost = 9.73663e-10, first explored at pop 34800 (6.86 %)
36. 1504207 "INT_L_X24Y79/SL1END1", cost = 9.7169e-10, first explored at pop 34805 (6.86 %)
37. 1504024 "INT_L_X24Y78/SL1END1", cost = 9.75636e-10, first explored at pop 34918 (6.89 %)
38. 1503841 "INT_L_X24Y77/SL1END1", cost = 9.73663e-10, first explored at pop 34921 (6.89 %)
39. 1503657 "INT_L_X24Y76/SL1END1", cost = 9.65771e-10, first explored at pop 34714 (6.85 %)
40. 1501352 "INT_L_X24Y75/SL1END1", cost = 9.69717e-10, first explored at pop 34811 (6.86 %)
41. 1501159 "HCLK_L_X63Y78/HCLK_SL1END1", cost = 9.79582e-10, first explored at pop 35311 (6.96 %)
42. 1436608 "INT_L_X24Y72/SS2END1", cost = 9.95859e-10, first explored at pop 42413 (8.36 %)
43. 1436250 "INT_L_X24Y66/SS6END1", cost = 1.04814e-09, first explored at pop 114796 (22.64 %)
44. 1445447 "INT_L_X24Y60/SS6END1", cost = 1.08859e-09, first explored at pop 183345 (36.15 %)
45. 1448884 "INT_L_X24Y54/SS6END1", cost = 1.09944e-09, first explored at pop 206398 (40.70 %)
46. 2005378 "BRKH_INT_X24Y49/BRKH_INT_SS6E1", cost = 1.12213e-09, first explored at pop 287714 (56.74 %)
47. 2013573 "INT_L_X24Y42/SS6END1", cost = 1.12706e-09, first explored at pop 304281 (60.00 %)
48. 1949645 "INT_L_X24Y36/SS6END1", cost = 1.14383e-09, first explored at pop 377777 (74.50 %)
49. 1941293 "INT_L_X24Y34/SS2END1", cost = 1.14235e-09, first explored at pop 377781 (74.50 %)
50. 1938831 "INT_L_X24Y32/SS2END1", cost = 1.14679e-09, first explored at pop 378226 (74.58 %)
51. 1938465 "INT_L_X24Y30/SS2END1", cost = 1.15123e-09, first explored at pop 384890 (75.90 %)
52. 1936003 "INT_L_X24Y28/SS2END1", cost = 1.14975e-09, first explored at pop 384898 (75.90 %)
53. 2965543 "INT_L_X24Y26/SS2END1", cost = 1.15419e-09, first explored at pop 395057 (77.90 %)
54. 2965177 "HCLK_L_X63Y26/HCLK_SS6B1", cost = 1.16504e-09, first explored at pop 430934 (84.98 %)
55. 2974770 "CLBLL_L_X26Y20/CLBLL_EE2A1", cost = 1.16504e-09, first explored at pop 431028 (85.00 %)
56. 2986069 "INT_L_X26Y20/EL1BEG0", cost = 1.1828e-09, first explored at pop 485322 (95.70 %)
57. 2986558 "CLBLM_R_X27Y19/CLBLM_IMUX39", cost = 1.18083e-09, first explored at pop 485323 (95.70 %)
[... cut ...]
Forward cost decrement: 1.184e-11
Forward lookahead cost decrement: 9.86482e-12

First pop 2986558 (cost: 1.18083e-09, pop: 485323 (95.70 %)) -> 2986729 (cost: 1.19069e-09, pop: 507110 (100.00 %))
Backward cost increment: 9.86e-12
Forward cost decrement: 0
Forward lookahead cost decrement: 0

Best pop 2986558 (cost: 1.18083e-09, pop: 485323 (95.70 %)) -> 2986729 (cost: 1.19069e-09, pop: 507110 (100.00 %))
Backward cost increment: 9.86e-12
Forward cost decrement: 0
Forward lookahead cost decrement: 0
```

litghost, 2019-11-18:
What this is show is the delta between two nodes in the final route, first showing the first pop of the relevant node, and second showing the "best" (e.g. lowest total cost).

The backward cost increment is: `nodes[N+1].back - nodes[N+1].back`
The forward cost decrement is: `nodes[N].forward - nodes[N+1].forward`
The lookahead cost decrement is `nodes[N].lookahead - nodes[N+1].lookahead`

The relationship between forward cost and lookahed cost is simply the A* factor:
`nodes[N].forward = A* factor * nodes[N].lookahead`

litghost, 2019-11-18:
For reference, this is the same analysis as above, but one iteration 1 (e.g. no congestion, max_crit = 0):

```
1. 1001908 "CLBLM_R_X29Y129/CLBLM_M_CQ", cost = 0.0, first explored at pop 1 (0.20 %)
2. 1003058 "CLBLM_R_X29Y129/CLBLM_M_CQ", cost = N/A, first explored at pop N/A
3. 1002902 "CLBLM_R_X29Y129/CLBLM_LOGIC_OUTS6", cost = 8.58244e-10, first explored at pop 4 (0.78 %)
4. 1002920 "INT_R_X29Y127/SS2END2", cost = 8.55285e-10, first explored at pop 9 (1.76 %)
5. 1003747 "INT_R_X29Y125/SS2END2", cost = 8.52325e-10, first explored at pop 13 (2.54 %)
6. 1004510 "INT_L_X28Y125/WL1END1", cost = 8.49366e-10, first explored at pop 19 (3.71 %)
7. 778643 "CLBLL_L_X28Y125/CLBLL_WR1END3", cost = 8.30623e-10, first explored at pop 20 (3.91 %)
8. 778750 "HCLK_R_X69Y130/HCLK_LVB12", cost = 8.17798e-10, first explored at pop 21 (4.10 %)
9. 1225033 "INT_R_X27Y101/LVB0", cost = 8.04974e-10, first explored at pop 22 (4.30 %)
10. 1197137 "BRKH_INT_X27Y99/BRKH_INT_LVB11", cost = 7.9215e-10, first explored at pop 23 (4.49 %)
11. 1521860 "INT_R_X27Y77/LVB0", cost = 7.79325e-10, first explored at pop 24 (4.69 %)
12. 1496071 "HCLK_R_X69Y78/HCLK_LVB10", cost = 7.66501e-10, first explored at pop 25 (4.88 %)
13. 1457630 "INT_R_X27Y53/LVB0", cost = 7.53676e-10, first explored at pop 26 (5.08 %)
14. 1994547 "BRKH_INT_X27Y49/BRKH_INT_LVB9", cost = 7.40852e-10, first explored at pop 27 (5.27 %)
15. 1963053 "INT_R_X27Y29/LVB0", cost = 7.28028e-10, first explored at pop 28 (5.47 %)
16. 3008558 "HCLK_R_X69Y26/HCLK_SS6E2", cost = 7.15203e-10, first explored at pop 29 (5.66 %)
17. 3001624 "INT_R_X27Y21/SS2END2", cost = 7.06325e-10, first explored at pop 30 (5.86 %)
18. 2985599 "INT_R_X27Y19/SS2END2", cost = 7.09284e-10, first explored at pop 73 (14.26 %)
19. 2986640 "INT_R_X27Y19/BYP_ALT3", cost = 7.26054e-10, first explored at pop 348 (67.97 %)
20. 2986295 "INT_R_X27Y19/BYP_BOUNCE3", cost = 7.24082e-10, first explored at pop 351 (68.55 %)
21. 2986558 "CLBLM_R_X27Y19/CLBLM_IMUX39", cost = 7.22109e-10, first explored at pop 353 (68.95 %)
22. 2986729 "CLBLM_R_X27Y19/CLBLM_L_D3", cost = 7.31974e-10, first explored at pop 511 (99.80 %)
23. 2996669 "CLBLM_R_X27Y19/CLBLM_L_D3", cost = 7.24822e-10, first explored at pop 512 (100.00 %)
24. 1002961 "INT_R_X29Y129/FAN_ALT5", cost = N/A, first explored at pop N/A
25. 1002966 "INT_R_X29Y129/FAN_BOUNCE5", cost = N/A, first explored at pop N/A
26. 1002945 "INT_R_X29Y129/BYP_ALT1", cost = N/A, first explored at pop N/A
27. 1002953 "INT_R_X29Y129/BYP_BOUNCE1", cost = N/A, first explored at pop N/A
28. 1002975 "INT_R_X29Y129/GFAN1", cost = N/A, first explored at pop N/A
29. 1002855 "CLBLM_R_X29Y129/CLBLM_IMUX28", cost = N/A, first explored at pop N/A
30. 1003025 "CLBLM_R_X29Y129/CLBLM_M_C4", cost = N/A, first explored at pop N/A
31. 1001858 "CLBLM_R_X29Y129/CLBLM_M_C4", cost = N/A, first explored at pop N/A

First pop 1002920 (cost: 8.55285e-10, pop: 9 (1.76 %)) -> 1003747 (cost: 8.52325e-10, pop: 13 (2.54 %))
Backward cost increment: 1.47973e-11
Forward cost decrement: 1.77573e-11
Forward lookahead cost decrement: 1.4797e-11

Best pop 1002920 (cost: 8.55285e-10, pop: 9 (1.76 %)) -> 1003747 (cost: 8.52325e-10, pop: 13 (2.54 %))
Backward cost increment: 1.47973e-11
Forward cost decrement: 1.77573e-11
Forward lookahead cost decrement: 1.4797e-11

First pop 1003747 (cost: 8.52325e-10, pop: 13 (2.54 %)) -> 1004510 (cost: 8.49366e-10, pop: 19 (3.71 %))
Backward cost increment: 1.47973e-11
Forward cost decrement: 1.77563e-11
Forward lookahead cost decrement: 1.4797e-11

Best pop 1003747 (cost: 8.52325e-10, pop: 13 (2.54 %)) -> 1004510 (cost: 8.49366e-10, pop: 19 (3.71 %))
Backward cost increment: 1.47973e-11
Forward cost decrement: 1.77563e-11
Forward lookahead cost decrement: 1.4797e-11

First pop 1004510 (cost: 8.49366e-10, pop: 19 (3.71 %)) -> 778643 (cost: 8.30623e-10, pop: 20 (3.91 %))
Backward cost increment: 3.45271e-11
Forward cost decrement: 5.32701e-11
Forward lookahead cost decrement: 4.4392e-11

Best pop 1004510 (cost: 8.49366e-10, pop: 19 (3.71 %)) -> 778643 (cost: 8.30623e-10, pop: 20 (3.91 %))
Backward cost increment: 3.45271e-11
Forward cost decrement: 5.32701e-11
Forward lookahead cost decrement: 4.4392e-11

First pop 778643 (cost: 8.30623e-10, pop: 20 (3.91 %)) -> 778750 (cost: 8.17798e-10, pop: 21 (4.10 %))
Backward cost increment: 6.4122e-11
Forward cost decrement: 7.6947e-11
Forward lookahead cost decrement: 6.4122e-11

[... 120 lines cut ...]
Best pop 2986640 (cost: 7.26054e-10, pop: 348 (67.97 %)) -> 2986295 (cost: 7.24082e-10, pop: 351 (68.55 %))
Backward cost increment: 9.865e-12
Forward cost decrement: 1.1837e-11
Forward lookahead cost decrement: 9.8647e-12

First pop 2986295 (cost: 7.24082e-10, pop: 351 (68.55 %)) -> 2986558 (cost: 7.22109e-10, pop: 353 (68.95 %))
Backward cost increment: 9.865e-12
Forward cost decrement: 1.1838e-11
Forward lookahead cost decrement: 9.86482e-12

Best pop 2986295 (cost: 7.24082e-10, pop: 351 (68.55 %)) -> 2986558 (cost: 7.22109e-10, pop: 353 (68.95 %))
Backward cost increment: 9.865e-12
Forward cost decrement: 1.1838e-11
Forward lookahead cost decrement: 9.86482e-12

First pop 2986558 (cost: 7.22109e-10, pop: 353 (68.95 %)) -> 2986729 (cost: 7.31974e-10, pop: 511 (99.80 %))
Backward cost increment: 9.865e-12
Forward cost decrement: 0
Forward lookahead cost decrement: 0

Best pop 2986558 (cost: 7.22109e-10, pop: 353 (68.95 %)) -> 2986729 (cost: 7.31974e-10, pop: 511 (99.80 %))
Backward cost increment: 9.865e-12
Forward cost decrement: 0
Forward lookahead cost decrement: 0
```

litghost, 2019-11-18:
What https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1047#issuecomment-555177138 shows is that lookahead is behaving okay in congestion free cases, except for one part of the path.

kmurray, 2019-11-18:
@litghost FYI, I've fixed two bugs related to the costing of non-configurable node sets:

1. Incorrect/under-predicted get_rr_cong_cost() for nodes in a non-configurable set (fixed in 83fdb1529)
2. Multiple counting of non-configurable set cost congestion when re-expanding nodes within the set (fixed in 68877d9a6)

Where (2) was the issue you pointed out earlier where we were re-adding the full congestion cost for the entire set when an element within it was re-expanded. 

The bugs in (1) and (2) pushed the cost in opposite directions, so their individual impact previously was likely somewhat muted.

litghost, 2019-11-18:
> The bugs in (1) and (2) pushed the cost in opposite directions, so their individual impact previously was likely somewhat muted.

I saw those changes, and they explain something I was seeing but couldn't explain around the non-config sets.  Anyways, I'm re-running to see if there is any impact.

litghost, 2019-11-18:
> Anyways, I'm re-running to see if there is any impact.

I'm seeing a somewhat faster convergence, so that is hopeful.  Individual iteration time is unchanged, but it might converge faster?

vaughnbetz, 2019-11-20:
Adding @YFWang97 @xuqinziyue @cindyhou to the discussion as they're working on Symbiflow quality.

vaughnbetz, 2019-11-20:
Adding @MohamedElgammal

litghost, 2019-11-20:
I'm working on a change that seems to help somewhat, here are the new results, `max_crit = 0.0`:
```
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 135: No routing path for connection to sink_rr 1908632, retrying with full device bounding box
Warning 136: 90 timing endpoints were not constrained during timing analysis
   1   14.3     0.0    0 9.0e+07    5458   16533    9576 ( 0.271%)  386913 ( 4.6%)   25.033 -2.400e+04    -25.033      0.000      0.000      N/A
   2   43.2     0.5    6 2.5e+08    4111   13541    5446 ( 0.154%)  407609 ( 4.8%)   26.676 -2.653e+04    -26.676      0.000      0.000      N/A
   3  110.3     0.6   36 5.8e+08    3475   11997    4406 ( 0.124%)  426418 ( 5.0%)   30.257 -2.857e+04    -30.257      0.000      0.000      N/A
   4  140.0     0.8   58 7.2e+08    3243   11703    2693 ( 0.076%)  442399 ( 5.2%)   29.924 -3.010e+04    -29.924      0.000      0.000      N/A
   5  131.1     1.1   79 6.9e+08    2921   11065     942 ( 0.027%)  452075 ( 5.3%)   30.139 -2.994e+04    -30.139      0.000      0.000      N/A
   6  113.0     1.4   57 5.9e+08    2798   10652     311 ( 0.009%)  458455 ( 5.4%)   31.120 -2.996e+04    -31.120      0.000      0.000      N/A
   7  101.9     1.9   23 5.2e+08    2728   10415      94 ( 0.003%)  460343 ( 5.4%)   30.669 -3.001e+04    -30.669      0.000      0.000      N/A
   8   89.8     2.4   12 4.7e+08    2688   10259      10 ( 0.000%)  461808 ( 5.4%)   30.628 -2.996e+04    -30.628      0.000      0.000      N/A
   9   85.4     3.1    4 4.4e+08    2674   10212       3 ( 0.000%)  461637 ( 5.4%)   30.144 -3.002e+04    -30.144      0.000      0.000      N/A
  10   80.4     4.1    3 4.2e+08    2667   10173       0 ( 0.000%)  461867 ( 5.4%)   30.144 -2.999e+04    -30.144      0.000      0.000       10
Restoring best routing
Critical path: 30.1439 ns
```

What I changes was the base cost length scaling.  The current code in master allows unlimited scaling.  In the 7-series arch, there are length 1 segments and segments that are ~44 units long.  This means the ratio of max(base_cost)/min(base_cost) ~= 44.  As a result when the router chooses a longer wire, it appears to be a very large base cost mispredict, and the router strongly avoid other wire types, even if they could help lower congestion.  

Overall, I believe my thesis is that base cost scaling should likely be limited to a compressed range (e.g. `1-4` or `1-2`).  I'll also explore the length/freq and freq base cost methods and see if they work better.

For reference, I believe the stratix 4 arch only has two wire lengths, 4 and 16, which is a base cost range of 4.

vaughnbetz, 2019-11-20:
Interesting.
We're getting some bad behaviour above in terms of re-routing: while there is little congestion, we are re-routing a ton of nets and connections, probably because we detect their timing is bad.  That's presumably because we're hacking the router a bit to set max_crit to 0, so it still thinks it's trying to get fast routes, but in fact we've crippled its timing optimization :(.
Could hack further to turn off the rip-up to improve timing condition (search route_timing.cpp).  Or could use the router in a more normal way by setting an sdc with a really huge clock period target, which should have the same effect of making all connections non-critical, but in a way the router understands and which should stop the excessive rip-up to try to improve timing.

litghost, 2019-11-20:
> Could hack further to turn off the rip-up to improve timing condition

There is a command line for that

litghost, 2019-11-20:
One thing @mithro suggested was to look at the bounding box for slow nets, and it does turn out that the slow nets have large bb's:

```
 Conn #1, from 798977 to 2999706 took 0.801417, (area: 5375, 43x125)
```
There are connections with large bounding boxes that route fast, but I haven't explored why this might be.

kmurray, 2019-11-20:
> We're getting some bad behaviour above in terms of re-routing: while there is little congestion, we are re-routing a ton of nets and connections, probably because we detect their timing is bad. [...]

Another thing to try would be running with `--timing_analysis off` which would target the router only at routability/wirelength and should avoid delay-based rip-up & re-route.

vaughnbetz, 2019-11-20:
If the lookahead works well and there isn't significant congestion, the bounding box size doesn't matter much since the router only explores a small part of the routing graph anyway. With significant congestion and/or lookahead mispredicts we explore a lot more, and the bounding box caps the worst-case behaviour. My guess would be that is the cause of the bi-modal behaviour.

If you don't route the clocks, how much faster does the router get?  Clock routing is a bit specialized (long wires, high fanout, net goes everywhere) and we're not yet using the 2-stage router, and the lookahead may not predict the paths well as it hasn't been tuned for them.  So an interesting experiment would be to turn off routing of the clocks and/or bin the cpu time into clock vs. regular routing to see if the clocks are an issue.

litghost, 2019-11-20:
> If you don't route the clocks, how much faster does the router get? Clock routing is a bit specialized (long wires, high fanout, net goes everywhere) and we're not yet using the 2-stage router, and [...]

Because none of the congestion resources are clocks, I don't believe clock nets are relevant.  None of the slow nets/connections I've examined are clock nets, they are all logic nets.

litghost, 2019-11-21:
> We're getting some bad behaviour above in terms of re-routing: while there is little congestion, we are re-routing a ton of nets and connections, probably because we detect their timing is bad.

Good catch!  When I turned off ripup and ran with `--max_criticality 0.0` runtime was only 120 seconds:

```
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 135: No routing path for connection to sink_rr 1908632, retrying with full device bounding box
Warning 136: 90 timing endpoints were not constrained during timing analysis
   1    7.2     0.0    0 5.4e+07    5458   16533   11188 ( 0.316%)  399246 ( 4.7%)   32.547 -3.058e+04    -32.547     -18.32     -3.744      N/A
   2    4.7     0.5    7 3.0e+07    4241   13707    8319 ( 0.235%)  412696 ( 4.9%)   31.042 -3.048e+04    -31.042      0.000      0.000      N/A
   3    4.8     0.6    6 3.1e+07    3781   12609    8576 ( 0.242%)  445907 ( 5.2%)   32.528 -3.210e+04    -32.528   -0.07983     -0.040      N/A
   4    4.9     0.8   13 3.1e+07    3388   11865    7537 ( 0.213%)  471534 ( 5.5%)   34.062 -3.271e+04    -34.062    -0.4314     -0.214      N/A
   5    4.9     1.1   27 3.2e+07    2951   11066    5004 ( 0.141%)  492053 ( 5.8%)   34.837 -3.344e+04    -34.837    -0.4432     -0.222      N/A
   6    4.8     1.4   35 3.1e+07    2434    9775    3290 ( 0.093%)  508486 ( 6.0%)   37.770 -3.452e+04    -37.770      0.000      0.000      N/A
   7    4.7     1.9   43 2.9e+07    1924    8638    2040 ( 0.058%)  522072 ( 6.1%)   36.676 -3.487e+04    -36.676      0.000      0.000      N/A
   8    3.7     2.4   34 2.2e+07    1289    6579     949 ( 0.027%)  530456 ( 6.2%)   36.407 -3.535e+04    -36.407      0.000      0.000      N/A
   9    2.6     3.1   36 1.6e+07     757    4166     389 ( 0.011%)  539021 ( 6.3%)   38.410 -3.575e+04    -38.410      0.000      0.000      N/A
  10    1.4     4.1   30 8028220     321    1880      98 ( 0.003%)  543265 ( 6.4%)   38.410 -3.601e+04    -38.410      0.000      0.000       19
  11    0.5     5.3   14 2498380      95     516      19 ( 0.001%)  544435 ( 6.4%)   41.372 -3.607e+04    -41.372      0.000      0.000       16
  12    0.3     6.9    2 1676148      22     115       4 ( 0.000%)  544671 ( 6.4%)   41.372 -3.611e+04    -41.372      0.000      0.000       14
  13    0.3     9.0    2 1335053       6      30       3 ( 0.000%)  544718 ( 6.4%)   41.372 -3.611e+04    -41.372      0.000      0.000       13
  14    0.3    11.6    0 1248041       3      13       1 ( 0.000%)  544918 ( 6.4%)   41.372 -3.611e+04    -41.372      0.000      0.000       14
  15    0.2    15.1    1  809083       2       4       0 ( 0.000%)  544940 ( 6.4%)   41.372 -3.611e+04    -41.372      0.000      0.000       14
Restoring best routing
Critical path: 41.3721 ns
# Routing took 46.46 seconds (max_rss 12698.1 MiB, delta_rss +0.0 MiB)
The entire flow of VPR took 125.24 seconds (max_rss 12698.1 MiB)
```

litghost, 2019-11-21:
I've disabled ripup on picosoc, and the runtime drops by more than half:
```
# Routing took 597.69 seconds (max_rss 12698.3 MiB, delta_rss +0.0 MiB)
```

litghost, 2019-11-21:
So I've found 3 changes that have dramatically improvement the runtime:
 - Disable ripup
 - Limit the scaling of base cost on length 4 orders of magnitude
 - Increase the pres fac cost scheduling to 3 (from 1.3)

I believe that last change is likely an intermix of the limiting of the scaling of the base cost, which effectively lowered base costs by too much.  So by increasing the pres fac cost schedule multiplier, it counter balanced the base cost drop.

Latest trace:
```
Confirming router algorithm: TIMING_DRIVEN.
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 135: No routing path for connection to sink_rr 1908632, retrying with full device bounding box
Warning 136: 90 timing endpoints were not constrained during timing analysis
   1   11.2     0.0    0 6.9e+07    5458   16533   11598 ( 0.328%)  735320 ( 8.7%)   19.877 -1.976e+04    -19.877      0.000      0.000      N/A
   2    4.6     0.5   91 2.6e+07    4342   13555    9981 ( 0.282%)  509995 ( 6.0%)   19.668 -2.123e+04    -19.668      0.000      0.000      N/A
   3    7.4     1.5    7 4.2e+07    4034   12926    8136 ( 0.230%)  539771 ( 6.4%)   19.801 -2.141e+04    -19.801      0.000      0.000      N/A
   4   53.8     4.5   64 3.0e+08    3404   11781    5063 ( 0.143%)  563073 ( 6.6%)   19.758 -2.289e+04    -19.758      0.000      0.000      N/A
   5  102.5    13.5  121 5.8e+08    2247    9376    1520 ( 0.043%)  601590 ( 7.1%)   19.680 -2.396e+04    -19.680      0.000      0.000      N/A
   6   69.3    40.5  146 3.8e+08     937    4298     303 ( 0.009%)  619851 ( 7.3%)   19.680 -2.427e+04    -19.680      0.000      0.000      N/A
   7   24.1   121.5  102 1.2e+08     254    1135      52 ( 0.001%)  624187 ( 7.3%)   19.750 -2.445e+04    -19.750      0.000      0.000      N/A
   8    6.1   364.5   24 2.6e+07      56     233       8 ( 0.000%)  625389 ( 7.4%)   20.271 -2.463e+04    -20.271      0.000      0.000      N/A
   9    0.7  1093.5    2 2984064       7      17       1 ( 0.000%)  625423 ( 7.4%)   20.760 -2.463e+04    -20.760      0.000      0.000      N/A
  10    0.1  3280.5    1    2791       1       3       0 ( 0.000%)  625417 ( 7.4%)   20.760 -2.463e+04    -20.760      0.000      0.000        9
Restoring best routing
Critical path: 20.76 ns
Successfully routed after 10 routing iterations.
Router Stats: total_nets_routed: 20740 total_connections_routed: 69857 total_heap_pushes: 1539843238 total_heap_pops: 290490454
```

Overall that isn't a bad result.  More circuits are required to be sure the new parameters and configuration is reasonable, but it seems to be working better than before!

Thoughts?

mithro, 2019-11-21:
> Critical path: 41.3721 ns
to
> Critical path: 20.1729 ns
to
> Critical path: 20.76 ns
?

vaughnbetz, 2019-11-21:
We wouldn't want to leave timing optimization off (which is basically what max_crit = 0 has done) but it's useful to see if the router speeds up when only trying to solve congestion (it should, and now it does) and to see how the lookahead is working for non-critical nets (since we made everything non-critical).  The long critical path should be an artifact of that, so we should leave timing optimization on in the real flow.

The fourth year design team working on artix7 symbiflow shows picosoc routing in 71 s (not including 20 s for loading files etc.) with default router parameter, but on the smaller ROI.  Critical path was 17.6 ns.

vaughnbetz, 2019-11-21:
FYI, we can probably save a little CPU time by making the first iteration have at least a small tie-break for congestion (don't set first_iteration_pres_cost to 0, set it to 1 or 0.5 or something above 0).  Could also set (all) criticalities to somewhat less than 0.99 in the first iteration (0.95? 0.9?) so we pay a little attention to congestion.  Would break ties that are nearly equal speed to spread out congestion right away, which would aid convergence, and also probably makes the lookahead a bit better (as some congestion is avoided right away, which helps avoid lookahead mispredicts due to congestion).  Not sure if this will be a measurable effect or not, but it may help a little and is easy to try.

litghost, 2019-11-21:
> FYI, we can probably save a little CPU time by making the first iteration have at least a small tie-break for congestion (don't set first_iteration_pres_cost to 0, set it to 1 or 0.5 or something [...]

I effectively did that be increasing the iteration 2 pres fac from `0.5` to `4` and used the default pres fac schedule, and the router completed in about 400 seconds.

vaughnbetz, 2019-11-21:
Makes sense.  Changing iteration 1 as I mentioned above may squeeze a little more out ...

closed 2025-05-16 by github-actions[bot]