verilog-to-routing/vtr-verilog-to-routing #1051: Congestion backwards cost missing from add_route_tree_node_to_heap

litghost, 2019-11-20:
#### Expected Behaviour

`add_route_tree_node_to_heap` should include congestion costs?

#### Current Behaviour

`add_route_tree_node_to_heap` only includes delay backward costs

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/route_timing.cpp#L1776

#### Possible Solution

Unclear if this behavior was intention or unintentional?

#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

litghost, 2019-11-20:
@vaughnbetz @kmurray

kmurray, 2019-11-20:
I believe what is being assumed here is that any node which is part of the existing route tree (i.e. used by other connections in the same net) is 'free' from a congestion perspective. This is why it doesn't have any (i.e. zero) congestion cost associated with it. 

Effectively this encourages the router to share nodes between connections. This is good for wirelength at the net level, as it encourages making a tree structure (common nodes) rather than a star structure (independent nodes). Setting the backward cost based on the Tdel of the node within the tree will make the router prefer starting to explore from low (rather high) delay nodes for critical connections.

@vaughnbetz any other comments?

(A look through the revision history and old releases of VPR (not under revision control) shows that this was added between the vpr_399 and vpr_422 releases.)

litghost, 2019-11-20:
Even in the absence of congestion, the backward costs should still have `(sum of base costs of previous nodes) * (1 - criticality)`

vaughnbetz, 2019-11-20:
This is an intentional behaviour; those wires are already part of the routing tree, so they are free from a congestion & base cost perspective.  They still have delay, so their timing cost is included, but not their base cost or any congestion multipliers.

closed 2019-11-25 by litghost