verilog-to-routing/vtr-verilog-to-routing #1056: Diamond route reduction

litghost, 2019-11-25:
#### Context

This is about matching vendor tool behavior, but I want to open an issue in case this should be solved in a general way.

#### Problem statement

In the 7-series fabric, `VCC` (e.g. logical true / 1) is available from the local switch boxes "HARD1" pin.  This is then connected to some nodes by default (e.g. an empty bitstream connects `VCC` to nodes.

Here is a concrete example:

Connections:
node `CLK_L1` needs to be driven to `VCC`
node `CTRL_L1` needs to be driven to `VCC`

Available switches:
node `HARD1` connects to `FAN_ALT5` (by default, buffered)
node `HARD1` connects to `FAN_ALT1` (by default, buffered)
node `FAN_ALT1` connects to `FAN_BOUNCE1` (by default, buffered)
node `FAN_ALT5` connects to `FAN_BOUNCE5` (by default, buffered)
node `FAN_ALT1` connects to `FAN_ALT5` (configurable, buffered switch)
node `FAN_BOUNCE1` connects to `CTRL_L1` (configurable, buffered)
node `FAN_BOUNCE5` connects to `CLK_L1` (configurable, buffered)

The connections that VPR produces for this part of the `VCC` net is:
`HARD1` -> `FAN_ALT1`
`FAN_ALT1` -> `FAN_BOUNCE1`
`FAN_ALT1` -> `FAN_ALT5`
`FAN_ALT5` -> `FAN_BOUNCE5`
`FAN_BOUNCE1` -> `CTRL_L1`
`FAN_BOUNCE5` -> `CLK_L1`

There is an equivalent connection, which has one less configuration which is:
`HARD1` -> `FAN_ALT1`
`FAN_ALT1` -> `FAN_BOUNCE1`
`FAN_BOUNCE1` -> `CTRL_L1`
`HARD1` -> `FAN_ALT5`
`FAN_ALT5` -> `FAN_BOUNCE5`
`FAN_BOUNCE5` -> `CLK_L1`

This second net is what Vivado produces when given this same net.

#### Solutions

It should always be possible to do fixup on the output routing to match the canonical routing from the vendor tool, but I'm wondering if VPR should find simple routing optimizations once routing is complete.

FYI @mithro @vaughnbetz @kmurray

vaughnbetz, 2019-11-26:
I don't see what the problem is. The two routings use the same number of resources (nodes); in fact the nodes used are the same. The VPR one has a slightly longer connection to the CLK_L1 (branches off FAN_ALT1 instead of HARD1) but as Vcc nets have infinite time to arrive at their destination it doesn't matter.
Does Vivado complain about the VPR routing?  If not, there's no problem that I can see.

litghost, 2019-11-27:
> Does Vivado complain about the VPR routing?

No, but one of subgoals is to have automated bitstream generation verification.  In this case, Vivado cannot express the `FIXED_ROUTE` that we use, which is less than ideal.  I agree VPR is not at fault here, I was just picking your brain on whether a VPR cleanup around this might be possible.  Given that this issue is simply an artifact of route expression, I'll like fix it up after VPR generates it output.

closed 2019-11-27 by litghost

vaughnbetz, 2019-11-27:
If you make criticality of this signal slightly above 0 you probably will reduce the odds of getting slightly longer connections (FAN_ALT1->FAN_ALT5 connection) by telling the router there is some timing impact (even though there isn't).  Not sure if that's a good idea or not, but you could try it by setting a min_crit as well as a max_crit in the router, and making min_crit small (say 0.05 or so) so you still focus mostly on congestion.
Modification would be around:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/f265da929a5722603ec693bf6bc05b2d39fa3cc3/vpr/src/route/route_timing.cpp#L973