verilog-to-routing/vtr-verilog-to-routing #1058: Official prebuilt binary download website is broken

rljacobson, 2019-11-27:
The links to download the official prebuilt releases eventually land here:
http://www.eecg.utoronto.ca/vtr/register.html

This form appears to be broken no matter if I complete the form or I click "Download without registration."  I get the following error (screenshot attached):

```
CGIWrap Error: User not found
CGIWrap was unable to find the user 'jluu' in the password file on this server.

Check the URL and try again.

Server Data:
Server Administrator/Contact: [no address given]
Server Name: www.eecg.toronto.edu
Server Port: 80
Server Protocol: HTTP/1.1
Virtual Host: www.eecg.toronto.edu
Request Data:
User Agent/Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_1) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/78.0.3904.108 Safari/537.36
Request Method: GET
Remote Host: h247.178.38.24.cable.wrrn.fullchannel.net
Remote Address: 24.38.178.247
Remote Port: 56584
Query String: name=&company=&email=&comments=&hidden=
Extra Path Info: /jluu/download_vtr.pl
Referring Page: http://www.eecg.utoronto.ca/vtr/register.html
```
[image]

kmurray, 2019-12-09:
Thanks for the report. 

@vaughnbetz can you look into this? It seems like the CGI script should be moved under the 'vtr' directory (rather than jluu which no longer exists).

vaughnbetz, 2019-12-09:
Working on this ... unfortunately I need to get the directories moved so I have permission to edit them. Asking our sysadmins for that ...

vaughnbetz, 2019-12-16:
Fixed. Thanks for flagging this.

closed 2019-12-16 by vaughnbetz