verilog-to-routing/vtr-verilog-to-routing #1062: Prepacker fails to find primitive

acomodi, 2019-12-11:
With latest master revision (https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/97ea31dec0fc6cda40914dee25ffae722fcbf9ef) packer fails with some SymbiFlow v2x tests (https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/utils/vlog/tests) with the following message:

```
# Packing
Begin packing '/home/alessandro/projects/symbiflow/symbiflow-arch-defs/build/utils/vlog/tests/muxes/use_mux.test.eblif'.

After removing unused inputs...
        total blocks: 5, total nets: 3, total inputs: 1, total outputs: 2
Begin prepacking.
# Packing took 0.00 seconds (max_rss 14.0 MiB, delta_rss +0.0 MiB)
Error 1:
Type: Packing
File: /home/alessandro/projects/symbiflow/vtr-symbiflow/vpr/src/pack/prepack.cpp
Line: 1189
Message: Failed to find any location to pack primitive of type '.names' in architecture
The entire flow of VPR took 0.00 seconds (max_rss 14.0 MiB)

```

#### Expected Behaviour

Packer completes successfully

#### Current Behaviour

Prepacker fails

#### Possible Solution

I have identified the problematic commit, which is this one: https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/3005451618913392b6e7a473de5d5cf07dc0a058

#### Steps to Reproduce

1. Unpack the tar.gz tarball: [reproduce_bug.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/3950077/reproduce_bug.tar.gz)
2. `cd reproduce-bug`
3. `vpr use_mux.arch.tiles.xml use_mux.test.eblif --pack --target_ext_pin_util 1.0,1.0`

#### Context

VPR is used in SymbiFlow to test the `v2x` (verilog2xml) utility. The utility translates verilog modules into xml architecture files which are then tested using VPR, to see if the architecture generated is compatible with the P&R tool.

#### Your Environment

* VTR revision used: https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/97ea31dec0fc6cda40914dee25ffae722fcbf9ef
* Operating System and version: Linux 4.15.0-65-generic, Ubuntu 18.04.3 LTS
* Compiler version: g++ (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0

kmurray, 2019-12-11:
Thanks for the report, I'll take a look!

kmurray, 2019-12-11:
The raw netlist considered is:
```
#Atom netlist generated by VPR
.model USE_MUX
.inputs \
    a \
    b \
    c
.outputs \
    o1 \
    o2

.names $false

.names $true
 1

.names $undef

.names logic_a o1
1 1

.names logic_a o2
1 1

.subckt LOGICBOX \
    I=a \
    O=logic_a

.end

.model LOGICBOX
.inputs \
    I
.outputs \
    O
.blackbox
.end
```

After cleaning the netlist revisions prior to 3005451 produce (as atom_netlist.cleaned.echo.blif when VPR is run with --echo_file on):
```
#Atom netlist generated by VPR
.model USE_MUX
.inputs \
    a
.outputs \
    o1 \
    o2

#Artificially inserted primary-output assigment buffer
.names o2 o1
1 1

.subckt LOGICBOX \
    I=a \
    O=o2

.end

.model LOGICBOX
.inputs \
    I
.outputs \
    O
.blackbox
.end
```
where the comment in front of the buffer indicates the buffer was actually absorbed (the atom netlist BLIF writer inserted a buffer to produce a valid BLIF so ABC's LEC could be used, as its not possible to represent a signal driving two primary outputs otherwise in BLIF).

The changes in 3005451 improved the robustness of buffer absorption to fix #1057, but resulted in more conservative handling, as a result the buffer LUT was not being absorbed:
```
#Atom netlist generated by VPR
.model USE_MUX
.inputs \
    a
.outputs \
    o1 \
    o2

.names o1 o2
1 1

.subckt LOGICBOX \
    I=a \
    O=o1

.end

.model LOGICBOX
.inputs \
    I
.outputs \
    O
.blackbox
.end
```
where the lack of a comment notes that the buffer is actually in the cleaned netlist. Since the architecture considered included no locations for LUTs the packer failed.

Traditionally we've avoided some cases of buffer absorption which can result in ambiguous primary input/output signal naming to ensure we can verify the post-implementation netlist with ABC's LEC (which relies on name matching the PIs/POs). However this should no longer be necessary as the netlist now track the primary output names separately so they should be maintained.

With 052fde6e0, I've removed the conservative cases to ensure the buffer is removed, which produces:
```
#Atom netlist generated by VPR
.model USE_MUX
.inputs \
    a
.outputs \
    o1 \
    o2

#Artificially inserted primary-output assigment buffer
.names o1__o2 o1
1 1

#Artificially inserted primary-output assigment buffer
.names o1__o2 o2
1 1

.subckt LOGICBOX \
    I=a \
    O=o1__o2

.end

.model LOGICBOX
.inputs \
    I
.outputs \
    O
.blackbox
.end
```
which again indicates that the buffer LUT is actually absorbed, allowing the circuit to be mapped to an architecture without any LUT primitives.

closed 2019-12-11 by kmurray

acomodi, 2019-12-12:
Great, thanks for the fix!