verilog-to-routing/vtr-verilog-to-routing #1066: Generating incomplete .fasm file

tanvir-a, 2019-12-13:
Recently I am working with VTR Flow to generate a small FPGA. I was facing some issues while generating the .fasm file. I tried a working version from an open-source project and the flow is working on the developer’s system. However, in my system, I found that the .fasm file is incomplete (the expected file has more information). Also, I tried with the debug and sanitizer option on and found the following runtime error message.

#### Current Behaviour

```
Serial number (magic cookie) for the routing is: -5755677
Circuit successfully routed with a channel w/home/tanvira/PRGA/vtr-verilog-to-routing/utils/fasm/src/fasm.cpp:590:28: runtime error: member access within null pointer of type 'const struct t_rt_node'
ASAN:DEADLYSIGNAL

=================================================================
==18961==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x559613f824ab bp 0x7fff47329950 sp 0x7fff473297f0 T0)
==18961==The signal is caused by a READ memory access.
==18961==Hint: address points to the zero page.
    #0 0x559613f824aa in walk_route_tree /home/tanvira/PRGA/vtr-verilog-to-routing/utils/fasm/src/fasm.cpp:590
    #1 0x559613f8317e in walk_routing /home/tanvira/PRGA/vtr-verilog-to-routing/utils/fasm/src/fasm.cpp:607
    #2 0x559613f835e8 in finish_impl /home/tanvira/PRGA/vtr-verilog-to-routing/utils/fasm/src/fasm.cpp:619
    #3 0x559613d97b13 in finish /home/tanvira/PRGA/vtr-verilog-to-routing/vpr/src/base/netlist_walker.h:45
    #4 0x559613d987ef in NetlistWalker::walk() /home/tanvira/PRGA/vtr-verilog-to-routing/vpr/src/base/netlist_walker.cpp:21
    #5 0x559613f4e80d in write_fasm /home/tanvira/PRGA/vtr-verilog-to-routing/utils/fasm/src/main.cpp:45
    #6 0x559613f5ba31 in main /home/tanvira/PRGA/vtr-verilog-to-routing/utils/fasm/src/main.cpp:81
    #7 0x7fb0f6715b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #8 0x559612d44ea9 in _start (/home/tanvira/PRGA/vtr-verilog-to-routing/build/utils/fasm/genfasm+0x1862ea9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/tanvira/PRGA/vtr-verilog-to-routing/utils/fasm/src/fasm.cpp:590 in walk_route_tree
==18961==ABORTING
```

Is the above runtime error cause this issue.

#### Your Environment

* VTR revision used: v8.0.0-rc2
* Operating System and version: Ubuntu 18.4
* Compiler version: GCC 7.4

kmurray, 2019-12-13:
Thanks for the report @tanvir-a.

Would it be possible for you to include a test-case which triggers this behaviour?

@litghost The stack trace appears to be in the FASM generation code, can you take a look?

tanvir-a, 2019-12-13:
Thanks for your quick reply. @kmurray

I checked the previous issue #1010 (issued by @leon575777642 ) which has merged in the master branch but not in the release. I modified the fasm.cpp, and the problem is solved. By the way, I am using PRGA project.

litghost, 2019-12-13:
> Thanks for your quick reply. @kmurray
> 
> I checked the previous issue #1010 (issued by @leon575777642 ) which has merged in the master branch but not in the release. I modified the fasm.cpp, and the problem is solved. By the way, I am using PRGA project.

Sounds like you have found the solution, and can close this issue?

closed 2019-12-13 by tanvir-a