verilog-to-routing/vtr-verilog-to-routing #1070: Past-ABC simulation error for soft multipliers implementation

georgkrylov, 2019-12-23:
ODIN simulator fails with a segmentation fault running the post-abc simulation for circuits that have multiplication operation, implemented on architectures having no hard blocks.
#### Expected Behaviour

The simulator should not produce a segmentation fault.
#### Current Behaviour

A crash!
#### Possible Solution

Since it works if either adders or multipliers are present on the architecture, the issue may be lying  in the place where the program crashes:
File:
$VTR_PATH/vtr-verilog-to-routing/ODIN_II/SRC/simulate_blif.cpp:1752
Function:
// TODO: Needs to be verified.
static void compute_generic_node(nnode_t *node, int cycle)
{
// yada yada yada
while (bit_map[0][lut_size] != 0) // <-- this line causes a crash

Alternatively, and less preferably, the issue may be caused by
the netlist generation for soft multipliers/adders or  maybe the abc optimization itself. 
#### Steps to Reproduce

1. create a task with the following config file contents:
##############################################
# Configuration file for running experiments
##############################################
# Path to directory of circuits to use
circuits_dir=benchmarks/verilog

# Path to directory of architectures to use
archs_dir=arch/timing

# Add circuits to list to sweep (basic bm)
circuit_list_add=two_multipliers.v

# Add architectures to list to sweep
arch_list_add=k6_frac_N10_40nm.xml

# Parse info and how to parse
parse_file=vpr_standard.txt

script_params_common=-ending_stage abc -use_odin_simulation 

2. The two_multipliers.v should be created in folder specified in task, and its contents are:

module top_module
(
    
    input [2:0] a1,a2,b1,b2,
    output [5:0] c1,c2
   
);
  assign c1 = a1*b1;
  assign c2 = a2*b2;

endmodule

3. Execute the task, see the failed:Odin simulator!

#### Context

I am trying to work with circuits  that have a limited hard blocks availability, as part of my coursework.
#### Your Environment

* VTR revision used:commit 2780988d58da6be454364e4a06cff8da8caf09e8, compiled with -DCMAKE_BUILD_TYPE=Debug -DWITH_BLIFEXPLORER=On
* Operating System and version: Linux user-Virtual-Machine-18 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux,
The problem also persists on other machine within non-virtualized environment
* Compiler version:
g++ (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

kmurray, 2020-01-29:
Thanks for the report.

@jeanlego Can you or someone on the ODIN team take a look?

@MohamedEldafrawy I recall you also looked into soft multiplier mapping. Do you happen to have any insights?

MohamedEldafrawy, 2020-01-29:
> Thanks for the report.
> 
> @jeanlego Can you or someone on the ODIN team take a look?
> 
> @MohamedEldafrawy I recall you also looked into soft multiplier mapping. Do you happen to have any insights?

Yes Kevin, I have ran into this before but it wasn't fixed. The code that implemts multiplies using only LUTs doesn't work. It seg faults as mentioned in this issue. Therefore in this case since the architecture has neither hard adders nor multipliers, ODIN will try to implement multiplies using LUTs only and therefore seg fault.

jeanlego, 2020-01-30:
Thanks for this @georgkrylov.
@sdamghan could you have a look?
@georgkrylov any chance you could add that in the PR that will include test cases for https://github.com/CAS-Atlantic/vtr-verilog-to-routing/pull/2?

georgkrylov, 2020-01-30:
@jeanlego I will create a separate PR with tests, sure.

closed 2025-05-18 by github-actions[bot]