verilog-to-routing/vtr-verilog-to-routing #1071: Error when doing power analysis

yonghaoXuGit, 2019-12-25:
#### Expected Behaviour

VTR should execute successfully with power analysis.

#### Current Behaviour

I want to do power analysis when executing run_vtr_flow.pl. So I use options -power and -cmo_tech,the cmos technology file I use is PTM_22nm/22nm.xml and benchmarks include sha.v,boundtop.v, or1200.v,arm_core.v,mkDelayWorker32B.v and stereovision3.v.But they all failed and report Error：“**Activity file does not contain signal...**”
#### Possible Solution

I guess if it is the reason that the FPGA architecture file or CMOS technology file can't match.
#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used: 8.0.0 rc1
* Operating System and version: Linux 4.15.0-66-generic Ubuntu 
* Compiler version: GCC 7.4.0

kmurray, 2020-01-29:
Thanks for the report. 

Can you provide the exact command you ran with run_vtr_flow.pl?

jgoeders, 2021-03-09:
This has been a while now, so I'm going to close the issue.  If you can provide the exact command + files to reproduce the issue I would be happy to re-open the issue and look into it.

closed 2021-03-09 by jgoeders