verilog-to-routing/vtr-verilog-to-routing #1083: xc6vlx240tff1156.xml

raminrasoulinezhad, 2020-01-24:
Hi, 

Unfortunately xc6vlx240tff1156.xml model is not working. Would you please help with that?  Thanks

command to run
```
./run_vtr_task.pl       task
```
Configuration file for running experiments in ./task/config/config.txt file.
```
circuits_dir=benchmarks/verilog
archs_dir=arch/timing
circuit_list_add=ch_intrinsics.v
arch_list_add=xc6vlx240tff1156.xml   <-- failed 
arch_list_add=k4_N4_90nm.xml        <-- worked

parse_file=vpr_standard.txt
pass_requirements_file=pass_requirements.txt 
script_params=-track_memory_usage
```

#### Expected Behaviour
work same as intel based models

#### Current Behaviour
failed: vpr (exited with return code 1) (took 0.43 seconds)

#### Your Environment

* VTR revision used: v8.0.0-rc2
* Operating System and version: Ubuntu 18
* Compiler version:

kmurray, 2020-01-29:
The `xc6vlx240tff1156.xml` model only works with the ISPD benchmark netlists (already techmapped) . It will not work with the VTR benchmarks (synthesized from verilog).

For an example of using `xc6vlx240tff1156.xml` see: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/tasks/regression_tests/vtr_reg_weekly/vpr_ispd/config/config.txt

where you'll need to run:
```
$ make get_ispd_benchmarks
```
from the VTR root to download the netlists first.

closed 2020-01-29 by kmurray

raminrasoulinezhad, 2020-01-31:
Thanks, 

After preparing the ISPD benchmarks, I ran the mentioned config.txt files. It works. However, after changing the architecture file from "ultrascale_ispd.xml" to "xc6vlx240tff1156.xml," the config file fails during VPR. 

So, I have two questions:

1- since the inputs are already tech mapped, it can not be applied to another architecture file, such as xc6vlx240tff1156.xml. Am I right?

2- Could you please tell me more about the process of tech mapping which is applied to "benchmarks/ispd_blif" files. Is it similar to the process that you did for the Stratix architectures using "vqm2blif" tool? Did you use ODIN?

Thanks a lot.

kmurray, 2020-01-31:
Oops, I got mixed up between `ultrascale_ispd.xml` and `xc6vlx240tff1156.xml`. Sorry for the confusion!

`ultrascale_ispd.xml` is derived from the ISPD contests.

`xc6vlx240tff1156.xml` [is derived from](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/arch/timing/xc6vlx240tff1156.xml#L1-L2) the V2B project: https://eddiehung.github.io/vtb.html
I believe that flow used YOSYS and/or a modified version of ODIN to do logic synthesis and map to the the primitives used by `xc6vlx240tff1156.xml`. At the moment I don't believe any of the tests in VTR actually use `xc6vlx240tff1156.xml` and it mostly serves as a reference.

> 1- since the inputs are already tech mapped, it can not be applied to another architecture file, such as xc6vlx240tff1156.xml. Am I right?

Yes that is correct.

> 2- Could you please tell me more about the process of tech mapping which is applied to "benchmarks/ispd_blif" files. Is it similar to the process that you did for the Stratix architectures using "vqm2blif" tool? Did you use ODIN?

The ISPD blifs were coverted using the vtr_flow/scripts/ispd2vtr.py script from the extended bookshelf format (which is already techmapped) provided with the ISPD ['16](http://www.ispd.cc/contests/16/l) and ['17](http://www.ispd.cc/contests/17/) contest.