verilog-to-routing/vtr-verilog-to-routing #1087: Overuse of heap in initialization

litghost, 2020-01-28:
#### Expected Behaviour

Dynamic allocation should be done deliberately, and avoided when possible.  1 allocation is better than N allocations, and N allocations is better than N*N allocations.

#### Current Behaviour

Dynamic allocation is being used in several places that is simply generating garbage.  Concrete examples:

- [my_atof_2D](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/libs/libvtrutil/src/vtr_token.cpp#L151) duplicates a const char *, simply to delimit a substring and call `vtr::atoi` on it.  Solution: have `vtr::atoi` use a `string_view`.
- [load_delay_annotations](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/pb_type_graph_annotations.cpp#L263-L267) allocates a bunch of arrays, including a 2D array with 1+N arrays, and then simply frees them.
- [load_delay_annotations](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/pb_type_graph_annotations.cpp#L315-L321) uses a std::set to check for pin usage, which generates 1 allocation per output pin.  A sorted vector would only require 1 allocation, and be faster!

As a concrete example, the function `load_delay_annotations` (and its descendants) generates around 1/3 of all allocation requests during `SetupVPR`.  Number of allocations done within `load_delay_annotations` is in excess of 21 million when loading the `stratixiv_arch.timing.xml` architecture.

To be clear, when not running with the heap profiler, this doesn't take too much time:
```
# Loading Architecture Description
# Loading Architecture Description took 0.80 seconds (max_rss 78.4 MiB, delta_rss +64.0 MiB)
# Building complex block graph
# Building complex block graph took 7.55 seconds (max_rss 620.5 MiB, delta_rss +542.1 MiB)
```

but given that the vast majority of those allocations are simply unneeded.

#### Possible Solution

Run https://gperftools.github.io/gperftools/heapprofile.html on VPR on designs (I started with `gsm_switch_stratixiv_arch_timing.blif`), and examine top `alloc_objects` (e.g. code that generates a lot of allocations).

#### Context

This was discovered when refactoring rr graph edge storage.  Because of the heap churn generated, it takes will days for the heap profiler to go from cold VPR start to finish allocating the rr graph.  While this isn't intrinsically a problem, the heap churn is generating useless work.  It is likely that a dedicate effort to be more deliberate with heap usage would improve runtime and lower heap churn.

litghost, 2020-01-28:
@kmurray / @vaughnbetz FYI

kmurray, 2020-01-29:
Yep the code which deals with the architecture delay annotations is messy and has been problematic for a while. 

If I recall correctly, it actually has O(N^2) behaviour at one point but that got reduced (86bda51bd7534acb407409bff37b0e9a63d36d2c). I'm sure its still far from optimal. Its one of those code areas which has been a 'would be nice to fix' but was relatively low on the priority list.

@litghost If you want to take a crack improving it go for it!

litghost, 2020-01-29:
> @litghost If you want to take a crack improving it go for it!

I haven't opened many PR's yet, I'm trying to get the amount of heap garbage during performance sensitive code significantly lower.  Once I've gotten the performance critical path cleaner (e.g. build rr graph, place delay computation, router lookahead computation, placement, routing), I'll start putting up PRs.

Examples:
 - https://github.com/litghost/vtr-verilog-to-routing/commit/78840cecea9cc0063a4d43e1820835e05dffc645
 - https://github.com/litghost/vtr-verilog-to-routing/commit/a37d0b7e82ef6d4cb91eaf0d9f4dbf8004afbdd8

litghost, 2020-01-30:
So I have a some preliminary number from https://github.com/litghost/vtr-verilog-to-routing , where I've eliminated most of the heap thrashing, and I'm starting to run some of the mediumish titan benchmarks:

Circuit (directrf_stratixiv_arch_timing.blif):

baseline:
```
# Loading Architecture Description took 1.09 seconds (max_rss 72.8 MiB, delta_rss +60.3 MiB)
# Building complex block graph took 10.34 seconds (max_rss 614.2 MiB, delta_rss +541.5 MiB)
# Load circuit took 79.28 seconds (max_rss 5094.1 MiB, delta_rss +4479.8 MiB)
# Clean circuit took 3.36 seconds (max_rss 5094.1 MiB, delta_rss +0.0 MiB)
# Compress circuit took 5.32 seconds (max_rss 5094.1 MiB, delta_rss +0.0 MiB)
# Verify circuit took 3.89 seconds (max_rss 5094.1 MiB, delta_rss +0.0 MiB)
# Build Timing Graph took 21.95 seconds (max_rss 5094.1 MiB, delta_rss +0.0 MiB)
# Load Timing Constraints took 0.07 seconds (max_rss 5094.1 MiB, delta_rss +0.0 MiB)
# Packing took 1452.81 seconds (max_rss 14395.1 MiB, delta_rss +9301.0 MiB)
Finished loading packed FPGA netlist file (took 133.974 seconds).
# Load Packing took 141.15 seconds (max_rss 14395.1 MiB, delta_rss +0.0 MiB)
## Build Device Grid took 2.53 seconds (max_rss 14395.1 MiB, delta_rss +0.0 MiB)
## Build routing resource graph took 332.31 seconds (max_rss 15703.3 MiB, delta_rss +1308.2 MiB)
# Create Device took 336.40 seconds (max_rss 15703.3 MiB, delta_rss +1308.2 MiB)
### Computing router lookahead map took 992.65 seconds (max_rss 15703.3 MiB, delta_rss +0.0 MiB)
### Computing delta delays took 4240.56 seconds (max_rss 29025.0 MiB, delta_rss +13321.8 MiB)
## Computing placement delta delay look-up took 5234.54 seconds (max_rss 29025.0 MiB, delta_rss +13321.8 MiB)
# Placement took 41168.18 seconds (max_rss 31270.2 MiB, delta_rss +15566.9 MiB)
# Routing took 13430.25 seconds (max_rss 34871.0 MiB, delta_rss +571.3 MiB)
Timing analysis took 102.721 seconds (73.9237 STA, 28.797 slack) (188 full updates: 172 setup, 0 hold, 16 combined).
The entire flow of VPR took 56739.72 seconds (max_rss 36061.0 MiB)
```

refactor_edges2:

```
# Loading Architecture Description took 1.38 seconds (max_rss 74.3 MiB, delta_rss +61.9 MiB)
# Building complex block graph took 11.64 seconds (max_rss 614.1 MiB, delta_rss +539.8 MiB)
# Load circuit took 86.37 seconds (max_rss 5099.3 MiB, delta_rss +4485.1 MiB)
# Clean circuit took 3.32 seconds (max_rss 5099.3 MiB, delta_rss +0.0 MiB)
# Compress circuit took 5.58 seconds (max_rss 5099.3 MiB, delta_rss +0.0 MiB)
# Verify circuit took 3.22 seconds (max_rss 5099.3 MiB, delta_rss +0.0 MiB)
# Build Timing Graph took 21.40 seconds (max_rss 5099.3 MiB, delta_rss +0.0 MiB)
# Load Timing Constraints took 0.08 seconds (max_rss 5099.3 MiB, delta_rss +0.0 MiB)
# Packing took 1761.02 seconds (max_rss 14396.4 MiB, delta_rss +9297.1 MiB)
Finished loading packed FPGA netlist file (took 383.474 seconds).
# Load Packing took 391.17 seconds (max_rss 14396.4 MiB, delta_rss +0.0 MiB)
## Build Device Grid took 2.78 seconds (max_rss 14396.4 MiB, delta_rss +0.0 MiB)
## Build routing resource graph took 381.94 seconds (max_rss 17733.5 MiB, delta_rss +3337.1 MiB)
# Create Device took 385.84 seconds (max_rss 17733.5 MiB, delta_rss +3337.1 MiB)
### Computing router lookahead map took 922.86 seconds (max_rss 17733.5 MiB, delta_rss +0.0 MiB)
### Computing delta delays took 4105.74 seconds (max_rss 34071.0 MiB, delta_rss +16337.5 MiB)
## Computing placement delta delay look-up took 5029.85 seconds (max_rss 34071.0 MiB, delta_rss +16337.5 MiB)
# Placement took 38243.54 seconds (max_rss 36327.6 MiB, delta_rss +18594.1 MiB)
# Routing took 14276.96 seconds (max_rss 39938.9 MiB, delta_rss +578.3 MiB)
Timing analysis took 103.88 seconds (75.0958 STA, 28.7845 slack) (188 full updates: 172 setup, 0 hold, 16 combined).
The entire flow of VPR took 55401.87 seconds (max_rss 41258.1 MiB)
```

cleanup_garbage2:

```
# Loading Architecture Description took 1.21 seconds (max_rss 74.2 MiB, delta_rss +61.6 MiB)
# Building complex block graph took 9.94 seconds (max_rss 614.3 MiB, delta_rss +540.1 MiB)
# Load circuit took 81.11 seconds (max_rss 5099.6 MiB, delta_rss +4485.3 MiB)
# Clean circuit took 3.19 seconds (max_rss 5099.6 MiB, delta_rss +0.0 MiB)
# Compress circuit took 5.22 seconds (max_rss 5099.6 MiB, delta_rss +0.0 MiB)
# Verify circuit took 3.27 seconds (max_rss 5099.6 MiB, delta_rss +0.0 MiB)
# Build Timing Graph took 20.42 seconds (max_rss 5099.6 MiB, delta_rss +0.0 MiB)
# Load Timing Constraints took 0.07 seconds (max_rss 5099.6 MiB, delta_rss +0.0 MiB)
# Packing took 1476.49 seconds (max_rss 14398.7 MiB, delta_rss +9299.1 MiB)
Finished loading packed FPGA netlist file (took 139.889 seconds).
# Load Packing took 148.74 seconds (max_rss 14398.7 MiB, delta_rss +0.0 MiB)
## Build Device Grid took 2.54 seconds (max_rss 14398.7 MiB, delta_rss +0.0 MiB)
## Build routing resource graph took 278.05 seconds (max_rss 17705.7 MiB, delta_rss +3307.0 MiB)
# Create Device took 281.66 seconds (max_rss 17705.7 MiB, delta_rss +3307.0 MiB)
### Computing router lookahead map took 867.78 seconds (max_rss 17705.7 MiB, delta_rss +0.0 MiB)
### Computing delta delays took 4129.70 seconds (max_rss 17705.7 MiB, delta_rss +0.0 MiB)
## Computing placement delta delay look-up took 4998.85 seconds (max_rss 17705.7 MiB, delta_rss +0.0 MiB)
```

The larger cleanup PR has finished, but you can already see a 40% reduction in max_rss.  I have further cleanup to do in the placer, which I'll be focusing on today, but the initial results are pretty promising.

An additional big advantage of completing this cleanup is that HeapProfiling on smaller titan designs is now completing in a reasonable amount of time, so future cleanups will know immediately when a thrash starts (because it will slow down dramatically).

FYI @hzeller / @HackerFoo

closed 2025-05-18 by github-actions[bot]