verilog-to-routing/vtr-verilog-to-routing #1101: Formatting changes for ODIN, global revamp

georgkrylov, 2020-02-03:
Hello everyone,

I am working on ODIN and contacted one of ODIN maintainers, @jeanlego, directly to review my improvements. I have got suggestions on cleaning up my code, which also involved style and  formatting changes. Looking for an automated formatting tool, as well as in search for strict guidelines I have noticed that VTR project already has a set of defined rules. The root of the repository contains a clang-format file, that describes coding standards, committed 8 months ago. I have discussed that with @jeanlego again and he suggested raising an issue that will promote open discussion.

This issue is to discuss if the changes are desired. One advantage of running clang-format on entire ODIN code is that the ODIN code-base will converge in style with the other projects that also use this format definition. The other advantage is in simplification of code-review/committing, since the style changes are easy to apply / suggest, by using standalone clang-format or vscode extension, which is quite popular code editor atm. 
This disadvantage is tremendous number of white-space changes that I am concerned will generate plenty of  potential merge conflicts. 

Dear project maintainers, do you think that if the change is done, the long term gain should outweigh the temporary inconvenience? I will create a PR shortly that will introduce/demonstrate the changes.

mithro, 2020-02-03:
Hi @georgkrylov,

The code formatting is currently used for vpr and automatically checked by Travis-CI. It would be awesome to have ODIN-II using the same formatting style. 

Autoformat changes should have the following properties;
 - [ ] Committed via the VTR Robot account (see https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/b56bb368c8d1de907a467a488e0ee7a654561190 for example)
 - [ ] Added to the [`.git-blame-ignore-revs`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.git-blame-ignore-revs)

I'm happy to help you do this if the ODIN-II developers are interested.

Tim '@mithro' Ansell

jeanlego, 2020-02-03:
I'd love to see this happen @mithro!
I agree that it would be preferable to have it done by the VTR Robot.
If it's not too much work for you, the changes would be more than welcome!

mithro, 2020-02-03:
To do a commit as the VTR Robot, you just do 
 `git commit --reset-author --author="VTR Robot <robot@verilogtorouting.org>"`

To get `make format` to be applied to ODIN-II, you just need to add it to the path at https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/6945966dbfcb5a62da7a5ea4a90e4cc3aaa002bd/CMakeLists.txt#L388-L397

I would recommend the following process;
 - [ ] Step 1 - Add 'ODIN-II' to the CMakeLists.txt auto format list.
 - [ ] Step 2 - Commit the result
 - [ ] Step 3 - Run `make format`
 - [ ] Step 4 - Commit the result, 
```bash
git commit -a --reset-author --author="VTR Robot <robot@verilogtorouting.org>" -m "🤖 - Automated code reformat.

I'm an auto code reformatting bot. Beep boop..."
```
 - [ ] Step 5 - Add commit to `.git-blame-ignore-revs`
 - [ ] Step 6 - Commit the result,
```bash
git commit -a --reset-author --author="VTR Robot <robot@verilogtorouting.org>" -m "🤖 - Ignore automated code reformat in blame.

Adding the reformatting change XXXXXXX (which I just committed!) to the ignore list.

I'm an auto code reformatting bot. Beep boop..."
```

kmurray, 2020-02-03:
FYI, there is a script which should to what @mithro described directly: dev/autoformat.py

jeanlego, 2020-02-03:
thanks!
tried it but I run into some issues with it (can't create file in /tmp)
went the manual route instead

jeanlego, 2020-02-03:
closing since #1105 got merged

closed 2020-02-03 by jeanlego