verilog-to-routing/vtr-verilog-to-routing #1114: Unitialized Variables in rr_graph_uxsdcxx_serializer.h

kmurray, 2020-02-06:
#### Expected Behaviour

VPR should build without warnings

#### Current Behaviour

With gcc-7:
```
n file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:18:0:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h: In function ‘void uxsd::load_channels(const pugi::xml_node&, T&, Context&, const std::function<void(const char*)>*, ptrdiff_t*) [with T = RrGraphSerializer; Context = void*]’:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h:1030:36: warning: ‘y_list_info’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         chan_width_->y_list[index] = info;
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:19:0:
/project/trees/vtr/vpr/src/route/gen/rr_graph_uxsdcxx.h:2693:21: note: ‘y_list_info’ was declared here
                 int y_list_info;
                     ^~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:18:0:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h:1026:25: warning: ‘y_list_index’ may be used uninitialized in this function [-Wmaybe-uninitialized]
             report_error(
             ~~~~~~~~~~~~^
                 "index %d on y_list exceeds y_list size %u",
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                 index, chan_width_->y_list.size());
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:19:0:
/project/trees/vtr/vpr/src/route/gen/rr_graph_uxsdcxx.h:2692:30: note: ‘y_list_index’ was declared here
                 unsigned int y_list_index;
                              ^~~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:18:0:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h:1003:36: warning: ‘x_list_info’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         chan_width_->x_list[index] = info;
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:19:0:
/project/trees/vtr/vpr/src/route/gen/rr_graph_uxsdcxx.h:2685:21: note: ‘x_list_info’ was declared here
                 int x_list_info;
                     ^~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:18:0:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h:999:25: warning: ‘x_list_index’ may be used uninitialized in this function [-Wmaybe-uninitialized]
             report_error(
             ~~~~~~~~~~~~^
                 "index %d on x_list exceeds x_list size %u",
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                 index, chan_width_->x_list.size());
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:19:0:
/project/trees/vtr/vpr/src/route/gen/rr_graph_uxsdcxx.h:2684:30: note: ‘x_list_index’ was declared here
                 unsigned int x_list_index;
                              ^~~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:18:0:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h: In function ‘void uxsd::load_block_types(const pugi::xml_node&, T&, Context&, const std::function<void(const char*)>*, ptrdiff_t*) [with T = RrGraphSerializer; Context = void*]’:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h:1161:41: warning: ‘pin_ptc’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         if (block_type_pin_index_to_name(tile, ptc) != value) {
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:19:0:
/project/trees/vtr/vpr/src/route/gen/rr_graph_uxsdcxx.h:3070:21: note: ‘pin_ptc’ was declared here
                 int pin_ptc;
                     ^~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:18:0:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h:1830:9: warning: ‘pin_class_type’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         switch (type) {
         ^~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:19:0:
/project/trees/vtr/vpr/src/route/gen/rr_graph_uxsdcxx.h:3149:31: note: ‘pin_class_type’ was declared here
                 enum_pin_type pin_class_type;
                               ^~~~~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:18:0:
/project/trees/vtr/vpr/src/route/rr_graph_uxsdcxx_serializer.h:1297:9: warning: ‘block_type_width’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         if (block_info.width != width) {
         ^~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:19:0:
/project/trees/vtr/vpr/src/route/gen/rr_graph_uxsdcxx.h:3214:21: note: ‘block_type_width’ was declared here
[... 184 lines cut ...]
         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from /project/trees/vtr/vpr/src/route/rr_graph_reader.cpp:19:0:
/project/trees/vtr/vpr/src/route/gen/rr_graph_uxsdcxx.h:2861:21: note: ‘switch_id’ was declared here
                 int switch_id;
                     ^~~~~~~~~
```

#### Possible Solution

Since this is in the generated code, it seems like the uxsdcxx generator will need to be updated to ensure these variables are always initialized.

#### Steps to Reproduce

1. Checkout  revision: 868126ec9545344db9b36a02534d7ca7814b1a97 (current master)
2. make vpr

#### Context

VTR should build warning clean, so that any newly introduced warnings are quickly identified and fixed. Keeping even otherwise innocuous warnings around makes it much more difficult for developers to do this.

#### Your Environment

* VTR revision used: 868126ec9545344db9b36a02534d7ca7814b1a97
* Operating System and version: Ubuntu 18.04
* Compiler version: gcc-7

kmurray, 2020-02-06:
@duck2 Any thoughts on this?

kmurray, 2020-02-06:
Interestingly, some further investigation shows that these warnings only show up if built with VTR_ENABLE_IPO_BUILD=off.

Likely when doing IPO/LTO the compiler defers the warnings until the final link stage. But we've suppressed those link-stage warnings for other reasons (#634)

litghost, 2020-02-06:
This is caused by a bug in gcc 7's static analyzer. I've tried added "[[noreturn]]" to attr_error, which would be called if the variable in question was in fact uninitialized.

duck2, 2020-02-06:
I'm seeing the same warnings on g++ 9. Maybe g++ can't follow our (rather convoluted) way of initializing, say, `switch_id`? In the currently generated code, if there is no `id` attribute, the code would throw. And else, if there is an `id` attribute which would cause `strtol` to report an `errno`, the code would again throw. So we know that we have an `int` to assign to `id` if the code didn't throw.

We can just zero-initialize the variables but I don't think this would be helping much in terms of code quality. It's still a solution, though.

litghost, 2020-02-06:
> We can just zero-initialize the variables but I don't think this would be helping much in terms of code quality. It's still a solution, though.

There is actually a runtime cost to zero initialization, so ideally we'd avoid it.

litghost, 2020-02-06:
> I'm seeing the same warnings on g++ 9. Maybe g++ can't follow our (rather convoluted) way of initializing, say, `switch_id`? In the currently generated code, if there is no `id` attribute, the code [...]

I dramatically simplified the checking (just a bool), and gcc still couldn't figure it out.  GCC is simply broken in this check.

kmurray, 2020-02-06:
> We can just zero-initialize the variables but I don't think this would be helping much in terms of code quality. It's still a solution, though.

I'd suggest giving this a try. It really isn't good to have these warnings (even if they're actually false positives). We can then measure if there is a significant run-time cost.

litghost, 2020-02-06:
As far as I can tell GCC cannot be trusted with detecting uninitialized variables in this case.  Overall my recommendation is to disable this code, link with examples: https://stackoverflow.com/questions/14132898/gcc-wuninitialized-wmaybe-uninitialized-issues 

I can also refactor the code to avoid zero initialization using `std::tuple`.

litghost, 2020-02-06:
> I can also refactor the code to avoid zero initialization using `std::tuple`.

This does not work, the same warning is generated.  Interestingly enough, the simple bool check works with the tuple addition, so the compiler isn't totally broken.  It just is unable to through follow pointers or bitset logic.

PR #1120 (2020-02-11): Fixes for new rr graph reader / writer
#### Description

Fixes #1114
Fixes #1121

#### Related Issue

#### Motivation and Context

#### How Has This Been Tested?

#### Types of changes

- [x] Bug fix (change which fixes an issue)
- [ ] New feature (change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)

#### Checklist:

- [ ] My change requires a change to the documentation
- [ ] I have updated the documentation accordingly
- [ ] I have added tests to cover my changes
- [x] All new and existing tests passed
closed 2020-02-11 by kmurray