verilog-to-routing/vtr-verilog-to-routing #1121: RR Graph Round-Trip Consistency Error

kmurray, 2020-02-07:
#### Expected Behaviour

The RR graph's generated by VPR should round-trip correctly with no loss/change in precision.

#### Current Behaviour

Some of the nightly regression tests (vpr_verify_rr_graph_bidir) which check this currently fail.

The offending precision change appears to be:

_arch.xml:_
```xml
    <segment freq="1.000000" length="4" type="bidir" Rmetal="11.064550" Cmetal="4.727860e-14">
```

_rr_graph.xml:_
```xml
<segment id="0" name="unnamed_segment_0"><timing C_per_meter="4.72786e-14" R_per_meter="11.0646"/>
```
where there has been round-off in the Rmetal/R_per_meter fields.

As a result VPR reports:
```
Error 1: 
Type: Routing
File: rr_graph.xml
Line: 85
Message: Architecture file does not match RR graph's segment R_per_meter
```

#### Possible Solution

The old RR graph writer was careful to ensure any float values were written out with enough precision to avoid any change in values (i.e. [std::numeric_limits<float>::digits10()](http://www.cplusplus.com/reference/limits/numeric_limits/)).

Perhaps the new RR graph reader needs to increase the precision at which it dumps floats in the XML format?

#### Steps to Reproduce

1. cd vtr_flow/tasks
2. ../scripts/run_vtr_task.pl regression_tests/vtr_reg_nightly/vpr_verify_rr_graph_bidir/
Producing:
```
vpr_verify_rr_graph_bidir: k4_n4_v7_bidir.xml/dsip.blif/common                                                           failed: vpr (exited with return code 1) (took 6.95 seconds)
vpr_verify_rr_graph_bidir: k4_n4_v7_bidir.xml/elliptic.blif/common                                                       failed: vpr (exited with return code 1) (took 13.00 seconds)
```

#### Context

Ran into this while trying to generate new golden results for the more thorough nightly regression.

#### Your Environment

* VTR revision used: latest master
* Operating System and version: Ubuntu 18.04
* Compiler version: gcc-7

kmurray, 2020-02-07:
@duck2 @litghost Any thoughts?

litghost, 2020-02-07:
> 2\. ../scripts/run_vtr_task.pl regression_tests/vtr_reg_nightly/vpr_verify_rr_graph_bidir/

Should these be in nightly (versus being in strong)?

kmurray, 2020-02-07:
> Should these be in nightly (versus being in strong)?

I was thinking the same. They run quick (<30 sec) and are really more functionality coverage tests, so they would likely be better suited to being in strong.

litghost, 2020-02-07:
Do want the fix in https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1120 or a new PR?

litghost, 2020-02-07:
> Do want the fix in #1120 or a new PR?

I went ahead and added it to #1120.

PR #1120 (2020-02-11): Fixes for new rr graph reader / writer
#### Description

Fixes #1114
Fixes #1121

#### Related Issue

#### Motivation and Context

#### How Has This Been Tested?

#### Types of changes

- [x] Bug fix (change which fixes an issue)
- [ ] New feature (change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)

#### Checklist:

- [ ] My change requires a change to the documentation
- [ ] I have updated the documentation accordingly
- [ ] I have added tests to cover my changes
- [x] All new and existing tests passed
closed 2020-02-11 by kmurray