verilog-to-routing/vtr-verilog-to-routing #1132: Stalls in congestion resolution should increase routing effort

HackerFoo, 2020-02-11:
#### Proposed Behaviour
VPR should increase routing effort in response to a lack of progress in congestion resolution.

#### Current Behaviour
There is a CONFLICTED router mode that can be enabled after a number of iterations, controlled by the `--congested_routing_iteration_threshold` flag. The default setting of this flag prevents the router from using this mode.

#### Possible Solution
VPR has a predictor that tracks the logarithmic progress of congestion resolution. This can be used to detect stalls and react by enabling the CONFLICTED router mode.

#### Context
I'm trying to clean up flags used by SymbiFlow, and so make VPR perform better using its default settings.

Similar to https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/658

kmurray, 2020-02-11:
This is a reasonable idea, but would need to be tuned carefully.

When I first put the conflicted mode in the experiments didn't really justify turning it on (from a QoR & run-time trade-off perspective). Although that was with a simplified iteration threshold based trigger.

Since turning on the conflicted mode increases search effort it also slows the router down (time per iteration increases). We'll want to avoid the case where a small blip in congestion resolution progress causes this to be turned on which slows the router down to the point where it no longer makes enough progress (in terms of iterations) for the routing predictor to fire (which lets users know their circuit isn't going to route).

HackerFoo, 2020-02-12:
I've created a flag with a conservative default. The `get_slope()` method of the routing predictor considers all iterations, so a sudden change won't have a large effect.