verilog-to-routing/vtr-verilog-to-routing #1134: Related clocks (PLL)

litghost, 2020-02-11:
#### Proposed Behaviour

SDC files should support `create_generated_clock` to allow for PLL/MMCM generated clocks to be defined.  This feature does not imply that VPR should understand the PLL/MMCM itself, but there does need to be a way to create synchronous relationships between clocks.  In particular important relationships are:
 - Clocks that have an integer fractional relationship
 - Clocks that have a phase offset relationship

#### Current Behaviour

I do not believe there is a way to describe related clocks at all.

#### Possible Solution

1. Add support for `create_generated_clock` to the SDC parser
2. Update `timing/read_sdc.cpp` to support creating clock relationships.

#### Context

DDR designs typically use multiple clocks with phase relationships.  With the ability to describe these relationships, the VPR STA is invalid.

litghost, 2020-02-11:
@kmurray I could use your help on this.  If this feature is already supported, how is it used?  If this feature doesn't exist, is this plan to implement it feasible?

litghost, 2020-02-11:
Issue tracking design being tested: https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1301

kmurray, 2020-02-11:
> SDC files should support `create_generated_clock` to allow for PLL/MMCM generated clocks to be defined. This feature does not imply that VPR should understand the PLL/MMCM itself, but there does [...]

Yep this makes complete sense. It would be good to support this.

However my understanding is that create_generated_clock is just a short-hand/syntatic-sugar which makes it easier to write the SDC files.

Off the top of my head I think it should be possible (although likely painful) to specify this by hand.

So something like:
```tcl
create_clock -name crystal -period 20 #Virtual 50Mhz base clock
create_generated_clock -multiply_by 2 -source crystal my_pll.out
```
Should be equivalent to:
```tcl
create_clock -name crystal -period 20 #Virtual 50Mhz base clock
create_clock -period 10 my_pll.out # 2x crystal
```

VPR does support the `-waveform` option of `create_clock`, which should allow the definition of phase-related clocks too:
```tcl
create_clock -name base -period 20 #Implicitly -waveform {0 10}
create_clock -name base_shifted -period 20 -waveform {5 15} #5ns phase shift vs base
```
The code to add to VPR to support create_generated_clock would (equivalently) do the same type of transformations.

There may be some scenarios where the explicit knowledge that the clocks are actually related in some way may be helpful (e.g. pessimism modelling, determining clock source latencies). But those are either currently unsupported or more secondary issues.

litghost, 2020-02-11:
> ```tcl
> create_clock -name crystal -period 20 #Virtual 50Mhz base clock
> create_clock -period 10 my_pll.out # 2x crystal
> ```

Ok, let me try that.  How does VPR know that these two clocks are the same virtual clock?

kmurray, 2020-02-11:
> Ok, let me try that. How does VPR know that these two clocks are the same virtual clock?

To clarify, VPR doesn't know they are the same, they will be treated as independent clocks. However the analysis results should be the same (e.g. worst case slacks on transfers between those two clocks).

kmurray, 2020-02-11:
Put another way:
```
create_clock -name crystal -period 20 #Virtual 50Mhz base clock
```
creates rising edges at 0, 20, 40,... ns.

While:
```
create_clock -period 10 my_pll.out # 2x crystal
```
creates rising edges at 0, 10, 20, 30, 40, ... ns

VPR already determines what the minimum time window is between those sequences of edges which is what are used as the actual inter-domain timing constraints.

litghost, 2020-02-11:
Okay, I believe we might not need anything then.  I believe the SDC we are using already captures the relative clock relationships.  The fact that VPR doesn't see any CDC paths must be another error in our timing model.

closed 2020-02-11 by litghost

kmurray, 2020-02-12:
>  The fact that VPR doesn't see any CDC paths must be another error in our timing model.

One thing to keep in mind is that VPR's _default_ timing constraints (if no SDC file is provided) [don't analyze inter-domain (i.e. CDC) paths](https://docs.verilogtorouting.org/en/latest/vpr/timing_constraints/#default-timing-constraints). But if you specify an SDC file they are analyzed by default (i.e. unless you use `set_false_path` or `set_clock_groups -exclusive`).