verilog-to-routing/vtr-verilog-to-routing #1136: Regression results are inconsistent accross platforms

jeanlego, 2020-02-12:
#### Proposed Behaviour
reproducible regression test results between machines

#### Current Behaviour
Google's CI (more specifically looking at vtr_reg_strong ) does not produce the same result as it does on Travis or locally.

#### Proposed Solution
After looking through the repo, it seems ABC makes heavy use of rand().
To fix this we would either have to patch ABC or push the fixes upstream onto ABC to use cross-platform number generators.

jeanlego, 2020-02-14:
VTR strong test failures just seems to be cpu time mismatches on fast running circuits (probably machine load).  Maybe we need to have very wide cpu ranges on very fast (1 s or so) circuits.

_Originally posted by @vaughnbetz in https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1144#issuecomment-586034389_

Make sense !

Any way we can disable QoR on runtime alone? I'll gladly help if I can

kmurray, 2020-02-18:
> Taking a stab in the dark here, After looking through the repo, it seems ABC makes heavy use of rand().

I don't think rand() is the issue as the actual build results are consistent, but not the run-times.

I suspect this large run-time bounds is due to the tests being run on VM's which are over-committed (e.g. more threads trying to run than real cores).

@mithro / @litghost Do you guys have any insight into this? Would it be possible to have the koroko tests be run on non-overloaded machines? I expect this run-time noise also bites us on larger benchmarks as well (although the percentage impact would hopefully be smaller). Fixing this would also greatly improve our ability to evaluate run-time/QoR changes (with large run-time bounce it is very difficult to evaluate whether we are actually improving/worsening anything, or if its just noise).

> Maybe we need to have very wide cpu ranges on very fast (1 s or so) circuits.

We should be able to increase the minimum threshold below which we don't issue an error using [RangeAbs](https://docs.verilogtorouting.org/en/latest/vtr/pass_requirements/#file-format) the for any run-time related metric in the VTR pass requirement files (e.g. [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/parse/pass_requirements/common/pass_requirements.vpr_pack_place.txt#L22-L23)).

Looking at the current run-time bounces a minimum absolute threshold of 10 seconds seems reasonable (in the example linked above the threshold is currently 2 seconds).

litghost, 2020-02-18:
Currently the vtr_reg_strong is run with -j16 (see https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.github/kokoro/presubmit/strong.cfg#L44), which is the number of cores on the VM's.  The placer is multithreaded, so I'd expect some variation from running with job number that high.  However, the sum of the entire run is likely fine, and so is the geomean of the run.  Individual variance is likely not meaningful, especially for smaller run size.

If we really care about CPU wallclock for small circuits, we really ought to be running them several times to account for cold vs hot disk caches, memory load, etc.

It is also worth noting that not all CPU's and memory subsystems are created equal.  NUMA affects are likely large on VPR designs that reside with a single memory bank, and access patterns between memory banks is going to be hard to account for.

What I'm trying to say is the current measurement system (run new code only, compare to committed result) likely doesn't have a meaningful result.  What might be more meaningful (and is what we do for PR's) is to run VPR at two commits (say a baseline commit and the new commit) on the same machine, sequentially, and compare between these.  This will take into account differences in CPU and memory performance that vary from host to host.

For the strong tests this likely won't increase the runtime dramatically.  For nightly and weekly tests, a 2x in runtime is a pretty steep cost.

litghost, 2020-02-18:
> Would it be possible to have the koroko tests be run on non-overloaded machines?

These machines are not overloaded, they are "dedicated" VMs.  What is true is that they are not necessarily always the same type of CPU and memory performance will vary based on a number of factors.

kmurray, 2020-02-19:
> The placer is multithreaded

Can you clarify this? My understanding is that only parallel part of VPR is the timing analyzer, but that we disable parallelism by default.

>  Individual variance is likely not meaningful, especially for smaller run size.

I agree small circuit run-times aren't particularly meaningful. However having individual circuit sanity checks is useful for detecting outliers (which can indicate bugs/unexpected behavior), although this is more useful for larger designs.

> It is also worth noting that not all CPU's and memory subsystems are not created equal. NUMA effects are likely large on VPR designs that reside with a single memory bank, and access patterns between memory banks is going to be hard to account for.

Yep I completely agree with this.

> What I'm trying to say is the current measurement system (run new code only, compare to committed result) likely doesn't have a meaningful result. 

I agree its not ideal, but it does serve as an automatic sanity check that we haven't severely changed QoR, and I fear that not having such tests would leave us dangerously running blind.

I'll also point out that the run-time variance seems to be a platform issue, as both TravisCI (also shared VMs) and BuildBot (dedicated machine) don't seem to have as much run-time variance as Koroko.

> What might be more meaningful (and is what we do for PR's) is to run VPR at two commits (say a baseline commit and the new commit) on the same machine, sequentially, and compare between these. This will take into account differences in CPU and memory performance that vary from host to host.

I agree this is the gold standard however as you've noted elsewhere it is time consuming. Another issue that this doesn't address is shared machine load (which can vary over time). That is still an issue with this methodology.

If someone has the time and resources to dedicate to automating this kind of QoR checking I'd be very interested.

> Currently the vtr_reg_strong is run with -j16 (see https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.github/kokoro/presubmit/strong.cfg#L44), which is the number of cores on the VM's.

I guess my question is whether these 16 VM cores correspond to 16 real cores on the actual machine? Or do they correspond to 8 cores + 2 way SMT/HyperThreading (if so we should reduce to the number of non-HT cores)? Or are they shared among a variety of VMs running on the actual machine (in which case we have limited control over other system load)?

litghost, 2020-02-19:
> Can you clarify this? My understanding is that only parallel part of VPR is the timing analyzer, but that we disable parallelism by default.

Sorry, I mean the timing analyzer in the placer.  I typically see VPR using multiple threads specifically when running the placer, and less so otherwise.  This may me an observation artifact, and not reality.

kmurray, 2020-02-19:
Hmm, I didn't realize the Kokoro configuration set VPR_NUM_WORKERS to something non-zero. Combined with the fact that a GCE [vCPU is a hyper-thread](https://cloud.google.com/compute/docs/faq#virtualcpu) means the previous configuration (before #1160) would run 16 instances of VPR, each using up to16 threads (so up to 256 threads total) on 8 'real' cores.

It correspondingly makes sense the run-times were a bit unstable (machine was overloaded).

In addition to #1160 (which reduces us down to 8 instances of VPR, so up to 128 threads), I think we should set VPR_NUM_WORKERS=1 (or equivalently leave it unspecified). That will ensure we don't overload the machine with more threads to run than real cores.

litghost, 2020-02-19:
> set VPR_NUM_WORKERS to something non-zero.

Won't this adversely affect nightly and weekly runtimes?  For large designs, the placer timing update tends not to overlap and serializing the timing analysis might have a negative impact.  Thoughts?

kmurray, 2020-02-19:
> Won't this adversely affect nightly and weekly runtimes? For large designs, the placer timing update tends not to overlap and serializing the timing analysis might have a negative impact. Thoughts?

There will be some impact (~10% placement time on Titan, see [Table VII here](http://www.eecg.utoronto.ca/~kmurray/tatum/fpt2018_tatum.pdf)), but the overall run-time effect is smaller (other stages like the packer/router spend less of their run-time on STA). 

Since the impact isn't too big, I think the trade-off of getting more consistent run-times is worth it.

kmurray, 2020-02-19:
Also, looking at #1163 and #1164 (for example) I don't see the Kokoro runs showing up in the status checks anymore? Is that expected, or something unrelated to this?

litghost, 2020-02-19:
> Also, looking at #1163 and #1164 (for example) I don't see the Kokoro runs showing up in the status checks anymore? Is that expected, or something unrelated to this?

We are looking into a fix.  Earlier this week there was a configuration change that likely messed up a permission somewhere.

mithro, 2022-02-24:
I believe this bug is now obsolete as it is 2022.

closed 2022-02-24 by mithro