verilog-to-routing/vtr-verilog-to-routing #1137: Setup and hold constraints from multiple clocks

litghost, 2020-02-12:
#### Proposed Behaviour

An input to a sub-circuit should be able to specify multiple clocks, and therefore multiple setup/hold constraints.

#### Current Behaviour

Each input port can only support one clock source, and therefore one setup/hold constraint.

#### Possible Solution

#### Context

kmurray, 2020-02-12:
Can you give an example (e.g. SDC file) which illustrates this? I don't quite follow the scenario.

litghost, 2020-02-12:
Example:
```
    (CELL
        (CELLTYPE "ISERDESE2_NUM_CE_1")
        (INSTANCE ILOGICE3)
        (TIMINGCHECK
            (HOLD CE1 (posedge CLK) (-0.065::-0.053))
            (SETUP CE1 (posedge CLK) (0.419::0.713))
            (HOLD CE1 (posedge CLKB) (-0.065::-0.053))
            (SETUP CE1 (posedge CLKB) (0.419::0.713))
        )
    )
```

litghost, 2020-02-12:
CLKB and CLK are clocks with 180 degree phase relationship.

litghost, 2020-02-12:
```
    (CELL
        (CELLTYPE "ISERDESE2")
        (INSTANCE ILOGICE3)
        (DELAY
            (ABSOLUTE
                (IOPATH CLKDIV Q1 (0.177::0.204)(0.568::0.653))
                (IOPATH CLKDIV Q2 (0.177::0.204)(0.568::0.653))
                (IOPATH CLKDIV Q3 (0.177::0.204)(0.568::0.653))
                (IOPATH CLKDIV Q4 (0.177::0.204)(0.568::0.653))
                (IOPATH CLKDIV Q5 (0.177::0.204)(0.568::0.653))
                (IOPATH CLKDIV Q6 (0.177::0.204)(0.568::0.653))
                (IOPATH CLKDIV Q7 (0.177::0.204)(0.568::0.653))
                (IOPATH CLKDIV Q8 (0.177::0.204)(0.568::0.653))
                (IOPATH OFB O (0.125::0.144)(0.327::0.376))
            )
        )
    )
    (CELL
        (CELLTYPE "ISERDESE2_INTERFACE_TYPE_OVERSAMPLE")
        (INSTANCE ILOGICE3)
        (DELAY
            (ABSOLUTE
                (IOPATH CLK Q1 (0.187::0.215)(0.498::0.573))
                (IOPATH CLK Q2 (0.187::0.215)(0.498::0.573))
                (IOPATH CLK Q3 (0.187::0.215)(0.498::0.573))
                (IOPATH CLK Q4 (0.187::0.215)(0.498::0.573))
                (IOPATH CLK Q5 (0.187::0.215)(0.498::0.573))
                (IOPATH CLK Q6 (0.187::0.215)(0.498::0.573))
                (IOPATH CLK Q7 (0.187::0.215)(0.498::0.573))
                (IOPATH CLK Q8 (0.187::0.215)(0.498::0.573))
            )
        )
    )
```

See `Figure 3-7: Logical View of ISERDESE2 Primitive in Oversample Mode` from https://www.xilinx.com/support/documentation/user_guides/ug471_7Series_SelectIO.pdf for diagrams and timing models.

litghost, 2020-02-12:
Another example: `Table 3-12: OSERDESE2 Switching Characteristics`

```
TOSDCK_T/TOSCKD_T T input Setup/Hold with respect to CLK
TOSDCK_T/TOSCKD_T T input Setup/Hold with respect to CLKDIV
```

litghost, 2020-02-12:
@mithro FYI

kmurray, 2020-02-12:
Ok, this is an architecture primitive modelling issue, and not an SDC support problem.

litghost, 2020-02-12:
> Ok, this is an architecture primitive modelling issue, and not an SDC support problem.

Sorry, ya!

kmurray, 2020-02-12:
In that case I think this should be supported. You just need to specify the 'edge' between the clock ports and related I/O ports, and then annotate the relevant T_setup/T_hold values in the pb_types.

See for instance [this scenario](https://docs.verilogtorouting.org/en/latest/tutorials/arch/timing_modeling/#multi-clock-sequential-block-with-internal-paths).

There are other (more complex) examples in the [Titan Stratix IV model](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/arch/titan/stratixiv_arch.timing.xml).

litghost, 2020-02-12:
Sorry, maybe I think I haven't been clear.  All of the examples you gave have 1 clock, so 1 setup/hold constraint.  I'm suggesting that there might be more than 1, so more than 1 clock would need to be specified in the model XML?

kmurray, 2020-02-12:
I guess I'm not following either :) 

> All of the examples you gave have 1 clock, so 1 setup/hold constraint.

Isn't that the 'Multi-clock Sequential block (with internal paths)' example [in the tutorial](https://docs.verilogtorouting.org/en/latest/tutorials/arch/timing_modeling/#multi-clock-sequential-block-with-internal-pathsl)? Multiple clocks with different Tsu/Thld?

litghost, 2020-02-12:
> Isn't that the 'Multi-clock Sequential block (with internal paths)' example [in the tutorial](https://docs.verilogtorouting.org/en/latest/tutorials/arch/timing_modeling/#multi-clock-sequential-block-with-internal-pathsl)? Multiple clocks with different Tsu/Thld?

That is still not enough, because in that case `we1` only goes to a FF under `clk1`.  Consider what would happen if `we1` went to a FF under `clk1` and another FF under `clk2`.

litghost, 2020-02-12:
> That is still not enough, because in that case `we1` only goes to a FF under `clk1`. Consider what would happen if `we1` went to a FF under `clk1` and another FF under `clk2`.

I guess we could duplicate the input port, and have the an internal pb-type connection to each of the duplicated ports :/

closed 2025-05-18 by github-actions[bot]