verilog-to-routing/vtr-verilog-to-routing #1138: Invalid Memory Access in Odin on strong_fpu_hard_block

kmurray, 2020-02-12:
#### Context

PR #1131 (for issue #1130) is trying to perform a regular sanitizer run of the vtr_reg_basic and vtr_reg_strong regression tests to detect undefined behaviour and memory errors.

The only outstanding issue is a bad memory access in ODIN that should be fixed.

Once fixed PR #1131 should begin passing the sanitizer test.

#### Steps to Reproduce

1. Build VTR with sanitizers enabled:
```bash
$ make BUILD_TYPE=debug CMAKE_PARAMS="-DVTR_ENABLE_SANITIZE=on" -j4
```

2. Run the strong_fpu_hard_block test case
```bash
$ cd vtr_flow/tasks
$ ../scripts/run_vtr_task.pl regression_tests/vtr_reg_strong/strong_fpu_hard_block_arch/
```
See that ODIN fails
```bash
regression_tests/vtr_reg_strong/strong_fpu_hard_block_arch: hard_fpu_arch_timing.xml/mm3.v/common                        failed: odin    (took 0.18 seconds)
Elapsed time: 0.3 seconds
```

3. Inspect the Odin Run Log to see the failure:
```bash
$ cd regression_tests/vtr_reg_strong/strong_fpu_hard_block_arch/latest/hard_fpu_arch_timing.xml/mm3.v/common
$ less odin.out
```
See the error message about buffer overflow from ASAN:
```
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

Reading Configuration file
Reading FPGA Architecture file
--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)/project/trees/vtr/libs/librtlnumber/src/include/internal_bits.hpp:367:52: runtime error: shift exponent 64 is too large for 32-bit type 'int'
=================================================================
==7490==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x61c00000e078 at pc 0x55a73f31bdef bp 0x7ffc417cefb0 sp 0x7ffc417cefa0
READ of size 8 at 0x61c00000e078 thread T0
    #0 0x55a73f31bdee in connect_hard_block_and_alias(ast_node_t*, char*, int, sc_hierarchy*) /project/trees/vtr/ODIN_II/SRC/netlist_create_from_ast.cpp:1730
    #1 0x55a73f35f0c1 in netlist_expand_ast_of_module(ast_node_t**, char*, sc_hierarchy*) /project/trees/vtr/ODIN_II/SRC/netlist_create_from_ast.cpp:728
    #2 0x55a73f36111f in netlist_expand_ast_of_module(ast_node_t**, char*, sc_hierarchy*) /project/trees/vtr/ODIN_II/SRC/netlist_create_from_ast.cpp:630
    #3 0x55a73f36111f in netlist_expand_ast_of_module(ast_node_t**, char*, sc_hierarchy*) /project/trees/vtr/ODIN_II/SRC/netlist_create_from_ast.cpp:630
    #4 0x55a73f36d4cf in convert_ast_to_netlist_recursing_via_modules(ast_node_t**, char*, sc_hierarchy*, int) /project/trees/vtr/ODIN_II/SRC/netlist_create_from_ast.cpp:386
    #5 0x55a73f36ed4b in create_netlist() /project/trees/vtr/ODIN_II/SRC/netlist_create_from_ast.cpp:187
    #6 0x55a73efbca5e in synthesize_verilog /project/trees/vtr/ODIN_II/SRC/odin_ii.cpp:116
    #7 0x55a73efbca5e in start_odin_ii(int, char**) /project/trees/vtr/ODIN_II/SRC/odin_ii.cpp:248
    #8 0x55a73efa7d50 in main /project/trees/vtr/ODIN_II/main.cpp:7
    #9 0x7f87db105b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #10 0x55a73efabff9 in _start (/project/trees/vtr/build/ODIN_II/odin_II+0xd96ff9)

0x61c00000e078 is located 8 bytes to the left of 1680-byte region [0x61c00000e080,0x61c00000e710)
allocated by thread T0 here:
    #0 0x7f87dd08dd38 in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xded38)
    #1 0x55a73f80cd5b in vtr::calloc(unsigned long, unsigned long) /project/trees/vtr/libs/libvtrutil/src/vtr_memory.cpp:42
    #2 0x55a73f8a009f  (/project/trees/vtr/build/ODIN_II/odin_II+0x168b09f)

SUMMARY: AddressSanitizer: heap-buffer-overflow /project/trees/vtr/ODIN_II/SRC/netlist_create_from_ast.cpp:1730 in connect_hard_block_and_alias(ast_node_t*, char*, int, sc_hierarchy*)
Shadow bytes around the buggy address:
  0x0c387fff9bb0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c387fff9bc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c387fff9bd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c387fff9be0: 00 00 fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c387fff9bf0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0c387fff9c00: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa[fa]
  0x0c387fff9c10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c387fff9c20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c387fff9c30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c387fff9c40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c387fff9c50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
```

Inspecting in GDB shows that [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/ODIN_II/SRC/netlist_create_from_ast.cpp#L1730) the variable `sc_spot_input_old` has value `-1` causing the invalid access.

#### Your Environment

* VTR revision used: sanitizer_fixes branch, although present on master as well.
* Operating System and version: Ubuntu 18.04
* Compiler version: GCC 7

jeanlego, 2020-02-12:
Thanks for the heads up, I'll have a look :)

closed 2020-02-13 by jeanlego

kmurray, 2020-02-13:
Thanks!

Closed #1138<https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1138> via #1140<https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1140>.

PR #1140 (2020-02-13): Fix invalid string cache access

#### Description
Fix invalid string cache access closing #1138 for #1131 
and add quick sanitize compile flags for Odin

#### Related Issue
closes #1138 
for #1131 

#### Motivation and Context
fix invalid memory access

#### How Has This Been Tested?
odin regression full

#### Types of changes

- [x] Bug fix (change which fixes an issue)
- [ ] New feature (change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)

#### Checklist:

- [ ] My change requires a change to the documentation
- [ ] I have [... cut]