verilog-to-routing/vtr-verilog-to-routing #1142: Compile warnings

jeanlego, 2020-02-13:
#### Current Behaviour
[256/343] Building CXX object vpr/CMakeFiles/libvpr.dir/src/device/check_rr_graph_obj.cpp.o                                                                                                                                                   
../vpr/src/device/check_rr_graph_obj.cpp: In function ‘bool check_rr_graph(const RRGraph&)’:                                                                                                                                                  
../vpr/src/device/check_rr_graph_obj.cpp:23:27: warning: potential null pointer dereference [-Wnull-dereference]                                                                                                                              
   23 |             result.first->second++;                                                                                                                                                                                                   
      |             ~~~~~~~~~~~~~~^~~~~~                                                                                                                                                                                                      
../vpr/src/device/check_rr_graph_obj.cpp:23:33: warning: potential null pointer dereference [-Wnull-dereference]                                                                                                                              
   23 |             result.first->second++;                                                                                                                                                                                                   
      |             ~~~~~~~~~~~~~~~~~~~~^~                                                                                                                                                                                                    
[323/343] Building CXX object vpr/CMakeFiles/libvpr.dir/src/route/router_lookahead_map.cpp.o                                                                                                                                                  
../vpr/src/route/router_lookahead_map.cpp: In function ‘void compute_router_src_opin_lookahead()’:                                                                                                                                            
../vpr/src/route/router_lookahead_map.cpp:426:10: warning: extra ‘;’ after in-class function definition [-Wextra-semi]                                                                                                                        
  426 |         };                                                                                                                                                                                                                            
      |          ^                                                                                                                                                                                                                            
      |          -                                                                                                                                                                                                                            

#### Steps to Reproduce
compile with gcc version 9.2.1

kmurray, 2020-02-18:
@tangxifan Can you take a look?

tangxifan, 2020-02-18:
For the warning at `check_rr_graph_obj.cpp:23:27`, I will use count() instead of insert(). The problem should be solved. Do you want me to check the warning at `router_lookahead_map.cpp:426:10` as well?

kmurray, 2020-02-18:
Thanks @tangxifan

> Do you want me to check the warning at `router_lookahead_map.cpp:426:10` as well?

I can take that one.

litghost, 2020-02-19:
It is worth noting that the `std::map::insert` cannot return an invalid iterator without throwing `bad_alloc`, so this warning is incorrect.  The fix in #1157 is not the correct solution because there are now two lookups into the b-tree.  If suppressing the warning is desired, the solution is:
```
VTR_ASSERT(result.first != edge_counter.end());
```

not to use `std::map::count`.  There is almost never a reason to use `std::map::count` followed by an access to the map.

tangxifan, 2020-02-19:
> It is worth noting that the `std::map::insert` cannot return an invalid iterator without throwing `bad_alloc`, so this warning is incorrect. See https://en.cppreference.com/w/cpp/container/map/insert
> 
> The fix in #1157 is not the correct solution because there are now two lookups into the b-tree. If suppressing the warning is desired, the solution is:
> 
> ```
> VTR_ASSERT(result.first != edge_counter.end());
> ```
> 
> not to use `std::map::count`. There is almost never a reason to use `std::map::count` followed by an access to the map.

I agree. This is not the most efficient way to fix-up the warning. But it does not lead to malfunction. What you propose looks good for me. I can apply the fix. @kmurray

kmurray, 2020-02-19:
> What you propose looks good for me. I can apply the fix.

Sounds good!

jeanlego, 2020-03-03:
Seem fixed in latest Travis build but a fallthrough showed up 
```
 89%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/pack/pb_type_graph.cpp.o
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/pack/pb_type_graph.cpp: In function ‘void alloc_and_load_interconnect_pins(t_interconnect_pins*, t_interconnect*, t_pb_graph_pin***, int, int*, t_pb_graph_pin***, int, int*)’:
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/pack/pb_type_graph.cpp:382:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
  382 |             VTR_ASSERT(num_output_sets == 1);
      |                  ^~
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/pack/pb_type_graph.cpp:384:9: note: here
  384 |         case MUX_INTERC:
      |         ^~~~
```
@kmurray want me to raise a separate issue?

kmurray, 2020-04-03:
I believe these have all been resolved, closing.

closed 2020-04-03 by kmurray