verilog-to-routing/vtr-verilog-to-routing #1172: Hi, I find some problems about switchblock description of VPR.

a3456410, 2020-02-27:
For some reason, I want to describe a custom switch block connection for my FPGA architecture, but not use the way which VTR 8.0 provide to choose custom SB type. 
I try to modify `rr_graph_sbox.cpp` file and add a new SB description in it, and use two FPGA xml files to test this SB. One is `k6_N10_40nm.xml`, which segments is unidir and its length is modified to 1 unit, and another is `k4_n4_v7_l1_bidir.xml`. The benchmark I use is `alu2.blif`.
I use graphics to check those two FPGA, and find that my new SB can be achieved correctly in `k4_n4_v7_l1_bidir.xml`, but be achieved wrongly in `k6_N10_40nm.xml`.

This is my SB in `k4_n4_v7_l1_bidir.xml`:
[image]

and this is in `k6_N10_40nm.xml`:
[image]
which connections of Top-to-Bottom, Bottom-to-Top, Left-to-Right, Right-to-Left are not changed.

#### Expected Behaviour

When the SB descriptions of Top-to-Bottom, Bottom-to-Top, Left-to-Right, Right-to-Left in `rr_graph_sbox.cpp` file are modified, those side-to-side connections in the SB should match the functions and change.

#### Current Behaviour

When I modify the SB descriptions of Top-to-Bottom, Bottom-to-Top, Left-to-Right, Right-to-Left in `rr_graph_sbox.cpp` file and use this SB in a unidir FPGA architecture, the connections of those directions in this SB are not changed actually. 
I also try to modify the classic SB descriptions in `rr_graph_sbox.cpp` file, such as Wiltion, and find that its connections of those four direction doesn't change in the unidir FPGA architecture.

#### Possible Solution

When the SB connections of those four directions are modified, this SB will be achieved in a bidir FPGA correctly, but will be achieved wrongly in a unidir FPGA. I guess the problem is in `rr_graph2.cpp`, which has conditional functions to choose the tracks are unidir or bidir. But I am not sure which section of code has bug in `rr_graph2.cpp`.

#### Steps to Reproduce

1. I add a new SB description in `rr_graph_sbox.cpp`, which name is WMSB. And I also add the WMSB Type in those file: `echo_arch.cpp`, `physical_types.h`, `read_xml_arch_file.cpp`, `ShowSetup.cpp`. Following are those files which I modify:
[wmsb_verify_file_vtr.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/4261050/wmsb_verify_file_vtr.zip)

2. Replace those files in VTR.

3. Use `make` command in the terminal to recompile VTR.

4. Modify segments length of `k6_N10_40nm.xml` to 1 unit. And modify the SB type of `k6_N10_40nm.xml` and `k4_n4_v7_l1_bidir.xml` to `wmsb`.

5. run the commands `vpr k6_N10_40nm.xml alu2.blif --disp on` and  `vpr k4_n4_v7_l1_bidir.xml alu2.blif --disp on`.

#### Context

For my thesis, I am using a IP auto generator of FPGA architectures which is based on VPR, which is called OpenFPGA.
[https://sites.google.com/site/pegaillardon/research/openfpga](url)

OpenFPGA also has the same problem I mentioned above, so I want to know how to solve it and then modify the code of OpenFPGA.

#### Your Environment

* VTR revision used: `commit 7f10959`
* Operating System and version: `Ubuntu 18.04.4 LTS`
* Compiler version:
`g++ (Ubuntu 8.3.0-6ubuntu1~18.04.1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

gcc (Ubuntu 8.3.0-6ubuntu1~18.04.1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

kmurray, 2020-03-02:
Since you've gone in and modified the VPR RR graph generator yourself there isn't much support we can provide (you wrote the code and we don't have the resources to debug it). All we can do is provide limited advice.

It sounds like you may have only edited the part of the RR graph generator which creates bi-directional RR graphs. If you want your SB to also work for uni-directional RR graphs you'd also need to modify the part of the RR graph generator which builds uni-directional RR graphs.

If you're using OpenFPGA you may also want to discuss this with @tangxifan.

tangxifan, 2020-03-02:
I have the same advice as @kmurray 
The RR graph builder has two different entries when building the connection inside switch blocks for bi-directional and uni-directional architecture.
I would suggest you set a breakpoint in your rr_graph_sbox.cpp and see if these codes are indeed executed in your run.

a3456410, 2020-03-09:
Hi, thanks for your suggestions!
I tried to modify the range from the line 2307 to the line 2334 of ` rr_graph2.cpp` file, which described the connection functions of from sides and opposite to sides, and the SB generation result was successful. So I think my problem was solved.

closed 2020-03-09 by a3456410