verilog-to-routing/vtr-verilog-to-routing #1180: Odin micro regression test failures

kmurray, 2020-03-03:
The VTR Buildbot is currently reporting build failures for the odin_reg_micro regression, as shown [here](http://builds.verilogtorouting.org:8080/builders/strong/builds/1211/). (These were previously masked by other issues with Buildbot which have recently been resolved).

There are two issues to resolve:

1. Fix the build failures in the Odin tests
2. Determine why the same failures weren't occurring in the TravisCI/Kokoro automated tests

kmurray, 2020-03-03:
@jeanlego Could you please take a look?

For issue (2), are we not testing odin_reg_micro on TravisCI/Kokoro? And if not, should we be?

jeanlego, 2020-03-03:
thanks! I'll defer this to @sdamghan and @Shreyv. 
Could you guys have a look? these are expected failures but seem to return success on buildbot.

sdamghan, 2020-03-06:
Sure, I'll have a look ASAP!

Shreyv, 2020-03-09:
Sure I will have a look at it

jeanlego, 2020-03-21:
any update @sdamghan ?

jeanlego, 2020-03-23:
http://builds.verilogtorouting.org:8080/waterfall
Fixed with #1230 closing since it is resolved

closed 2020-03-23 by jeanlego