verilog-to-routing/vtr-verilog-to-routing #1240: Direct inter-block connections identified but warnings are still given during placement delta delay lookup computation

leon575777642, 2020-03-27:
I'm using a custom architecture file very similar to `k6_frac_N10_frac_chain_mem32K_40nm.xml` (attached below). Specifically, the architecture has a carrychain basically identical to the one in the flagship architecture (`cin` on the top of the CLB, `cout` on the bottom of the CLB, `fc` set to 0 in <tile> tag). When running vpr with this architecture, the carrychain is successfully identified and utilized during packing, placement and routing. However, unlike the flagship architecture, when computing the placement delta delay look-up, vpr still tries to find routable path from a few other block output pins to the `cin` pins of all CLB blocks. This takes a significant amount of time (90% of runtime for small designs like `stereovision3.v`).

I think the most probable reason is I missed something here, because the flagship architecture does not have this issue.

#### Expected Behaviour
Placement delta delay look-up computation should ignore pins with fc set to 0.

#### Current Behaviour
Placement delta delay look-up computation does not ignore pins with fc set to 0.

#### Steps to Reproduce
Run `vtr_flow/scripts/run_vtr_flow.pl vtr_flow/benchmarks/verilog/stereovision3.v arch.scal.xml`.

#### Your Environment

* VTR revision used: 8612c27

[arch.scal.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/4394751/arch.scal.zip)

leon575777642, 2020-03-27:
Tried with other benchmarks. Computing delta delay look-up takes about 70-80% of the total runtime.

kmurray, 2020-03-27:
A quick work around appears to be re-ordering the pins in the architecture file so the cin/cout come last in the `<tile>` and `<pb_type>` definitions. For instance:
```xml
		<tile name="clb" capacity="1" width="1" height="1">
			<clock name="clk" num_pins="1"></clock>
			<input name="in0" num_pins="6"></input>
			<output name="out0" num_pins="2"></output>
			<input name="in1" num_pins="6"></input>
			<output name="out1" num_pins="2"></output>
			<input name="in2" num_pins="6"></input>
			<output name="out2" num_pins="2"></output>
			<input name="in3" num_pins="6"></input>
			<output name="out3" num_pins="2"></output>
			<input name="in4" num_pins="6"></input>
			<output name="out4" num_pins="2"></output>
			<input name="in5" num_pins="6"></input>
			<output name="out5" num_pins="2"></output>
			<input name="in6" num_pins="6"></input>
			<output name="out6" num_pins="2"></output>
			<input name="in7" num_pins="6"></input>
			<output name="out7" num_pins="2"></output>
			<input name="cin" num_pins="1"></input> <!-- moved to end of list -->
			<output name="cout" num_pins="1"></output>
...
```

That reduces stereovision3's delay look-up time from 60 to 8 seconds on my machine (with a debug build).

kmurray, 2020-03-27:
In the mean time I'm looking into how to avoid this more generally.

The existing code tried to avoid this by looking for the pin class with the most pins (which is usually not a cin/cout). However in this case the architecture has no pin-equivalence, so each pin gets it's own class, and it effectively falls back to picking based on the architecture file order.

leon575777642, 2020-03-27:
Thanks for the quick response. This is really helpful.

I'm not using pin equivalence now because I haven't figured out how to correctly generate bitstream for them. This is on my TODO list as well.

commit b49dacd6 (2020-03-30): vpr: Skip fc=0 pins during placement delay lookup calculation

The previously code could choose such pins if no ports had pin
equivalence (i.e. each was a unique class of size one), and an
fc=0 pin appeared first in the tile/pb_type port list. This
would cause high run-times during placement delay calculation,
since the router would need to search the entire RR graph
before giving up (since no such path exists).

This commit ensures that we only consider pin classes which are a
accessible to/from the general inter-block routing network (i.e.
skipping those with fc=0).

Closes #1240
commit ea7b7abd (2020-03-30): vpr: Skip fc=0 pins during placement delay lookup calculation

The previously code could choose such pins if no ports had pin
equivalence (i.e. each was a unique class of size one), and an
fc=0 pin appeared first in the tile/pb_type port list. This
would cause high run-times during placement delay calculation,
since the router would need to search the entire RR graph
before giving up (since no such path exists).

This commit ensures that we only consider pin classes which are a
accessible to/from the general inter-block routing network (i.e.
skipping those with fc=0).

Closes #1240
PR #1241 (2020-03-30): vpr: Skip fc=0 pins during placement delay lookup calculation
The previously code could choose such pins if no ports had pin
equivalence, and an fc=0 pin appeared first in the tile/pb_type
port list. This would cause high run-times during placement delay
calculation, since the router would need to search the entire RR graph
before giving up (since no such path exists).

This commit ensures that we only consider pin classes which are a
accessible to/from the general inter-block routing network (i.e.
skipping those with fc=0).

Closes #1240
commit 73819dc0 (2020-03-30): vpr: Skip fc=0 pins during placement delay lookup calculation

The previously code could choose such pins if no ports had pin
equivalence (i.e. each was a unique class of size one), and an
fc=0 pin appeared first in the tile/pb_type port list. This
would cause high run-times during placement delay calculation,
since the router would need to search the entire RR graph
before giving up (since no such path exists).

This commit ensures that we only consider pin classes which are a
accessible to/from the general inter-block routing network (i.e.
skipping those with fc=0).

Closes #1240
closed 2020-03-30 by kmurray