verilog-to-routing/vtr-verilog-to-routing #1313: FASM: Unexpected fasm_prefix added to output features

leon575777642, 2020-05-22:
In a corner case, the `fasm_prefix` of an inner pb_type is prepended to a feature belonging to an outer pb_type.

Specifically, in the `arch.xml` attached below, there are two `cluster` pb_types inside the top-level pb_type `clb`, and each of the `cluster` pb_types has a `fasm_prefix`. In the top-level pb_type `clb`, there are a few muxes connecting `clb` inputs and `cluster` outputs to `cluster` inputs. Now the bug is, when such a mux is configured to connect the output of a `cluster` pb_type to any input of the SAME `cluster` pb_type, the prefix of the `cluster` pb_type is prepended to the output feature of the mux. This is not expected, because the mux does not belong to the `cluster` pb_type. Note that the prefix of the `cluster` pb_type is NOT prepended when the mux is configured to connect a `clb` input to a `cluster` input, or a `cluster` output to the input of the other `cluster` pb_type.

My guess is, in the internal data structure, when both ends of a connection belong to the same pb_type, the fasm_prefix is emitted. I looked at the fasm code briefly and couldn't come up with a quick fix.

#### Expected Behaviour
`fasm_prefix` features should not be prepended to features outside its owner pb_type.

#### Current Behaviour
`fasm_prefix` feature is prepended to features outside its owner pb_type in a corner case.

#### Possible Solution
Not sure.

#### Steps to Reproduce
Please refer to the attached tarball. The fasm file is generated from the `.blif`, `.net`, `.place`, and `.route` files attached. An example of the bug mentioned above can be found in line 276-277, and line 284-286. In these lines, `x3.y6.b0` is the prefix of the clb tile (the x and y values in the prefix are not the position in VPR grid. This tile is actually located at x=5, y=7), `b0` and `b69` are the prefixes of the `cluster` pb_types mentioned above, while `b152`, `b153`, `b175`, `b176`, and `b177` are features related to mux `mux_cluster_i0_in_3` and `mux_cluster_i1_in_3`.

#### Context
I'm using FASM for bitstream generation in my PRGA project.

#### Your Environment

* VTR revision used: v8.0.0-rc1-1464-g8612c27 (commit 8612c27)
* Operating System and version: Springdale Linux release 7.8 (Verona)
* Compiler version: GNU 6.3.1 on Linux-3.10.0-1062.12.1.el7.x86_64 x86_64

#### Attachments

[bugreport.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/4669250/bugreport.zip)

litghost, 2020-05-22:
Can you please supply the output FASM that you expected, versus what you got?

Please make an effort to reduce the test case to something small if possible, e.g. a small circuit.

leon575777642, 2020-05-22:
Hi @litghost , thanks for your response. I've digged more into the source code and come up with a fix.

The bug is in this line:

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/7465954b5cb7addbd8fa79b590287cd0867d370e/utils/fasm/src/fasm.cpp#L682

Here it's using the parent pb graph node of the mux input when searching the prefix. However, if the mux input is an output of an inner pb_type, then the prefix of the inner pb_type is emitted along the way, causing this bug. I'll work out a quick fix and link the pull request here.

leon575777642, 2020-05-22:
Hi @litghost please review the draft pull request.

litghost, 2020-11-17:
Fix in PR #1314 has been merged.

closed 2020-11-17 by litghost