verilog-to-routing/vtr-verilog-to-routing #1317: VTR cannot build on Cygwin due to 'posix_memalign' not declared

Bill-hbrhbr, 2020-05-25:
#### Environment
OS: Windows 10, 64 bits
Cygwin: 3.1.4, x86_64
GCC/G++: 9.3.0
Cmake: 3.14.5

#### Expected Behaviour
Trying to build VTR on Windows Cygwin. The build should be successful since the function **posix_memalign** should be declared in stdlib.h

#### Current Behaviour
~/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_memory.h:67:12: error: ‘posix_memalign’ was not declared in this scope
   67 |     return posix_memalign(ptr_out, align, size);

#### Possible Solution
Use the standard C library functions **_aligned_malloc** and **_aligned_free** instead.
#define posix_memalign(p, a, s) (((*(p)) = _aligned_malloc((s), (a))), *(p) ?0 :errno)

Also, **posix_memalign** return pointers are passed to the regular **free** function, so maybe define:
#ifdef _WIN32
#define posix_memalign_free _aligned_free
#else
#define posix_memalign_free free
#endif

#### Steps to Reproduce
1.  Clone the VTR project and build with the environment settings mentioned above.
2.  I also encountered **M_PI not defined** before this error. I added a macro to relevant src files for a simple fix.

This seems to be a common problem. Let me know if you need extra information on my system settings.

cr1901, 2020-11-01:
I am also running into this issue. I'll try the solution @Bill-hbrhbr suggests and report back.

>Use the standard C library functions _aligned_malloc and _aligned_free instead.
>#define posix_memalign(p, a, s) (((*(p)) = _aligned_malloc((s), (a))), *(p) ?0 :errno)

`_aligned_malloc` and `_aligned_free` appear to be [Microsoft-specific](https://docs.microsoft.com/en-us/cpp/c-runtime-library/reference/aligned-free?view=msvc-160), not standard C.

>Also, posix_memalign return pointers are passed to the regular free function, so maybe define:
>#ifdef _WIN32
>#define posix_memalign_free _aligned_free
>#else
>#define posix_memalign_free free
>#endif

I would personally leave `posix_memalign_free` alone on *nix systems since it's guaranteed to be there. Something shorter like:

```
#ifdef _WIN32
#define posix_memalign_free _aligned_free
#endif
```

vaughnbetz, 2020-11-02:
@cr1901 : It would be good to land this one, so if you can check out those options and report back that would be great.

PR #1582 (2020-11-19): Native Windows Compile Fixes For `vpr`
#### Description
This PR adds some fixes so that `mingw-w64` `gcc` can successfully compiler `vpr`. _Only `vpr` is known to compile as of this PR._

Specifically:

* Use `_aligned_malloc` and `_aligned_free` on Windows instead of 
* Change `long` to `uintptr_t` or `intptr_t` based on context in `libvqm`.
* Do not include `sys/mman.h` header in `libvtrcapnproto`.

#### Related Issue
This PR should close #1317, but includes additional fixes so that compilation succeeds using `gcc` targeting native windows exes.

#### Motivation and Context
Per @vaughnbetz i [... cut]
closed 2020-11-19 by vaughnbetz