verilog-to-routing/vtr-verilog-to-routing #1322: ODIN produces mult primitives that don't exist in the architecture file

MahshadJ, 2020-05-28:
I was following the Quickstart Tutorial to run a circuit through ODIN and ABC. I was using LU8PEEng.v for the circuit file and k6_frac_N10_frac_chain_mem32K_40nm.xml for the architecture file. I was able to generate the odin.blif, abc_no_clock.blif, and the pre-vpr.blif files successfully. However, when running the last step which was vpr, I got an error saying "failed to find matching architecture model for mult_24_24_48". I retried the process with the same files as in the tutorial and there was no problem, so I was sure I was running it properly. After some debugging, it seems that ODIN is producing mult primitives that don't exist in the architecture file.

#### Expected Behaviour

VPR would run successfully because after running ODIN, there shouldn't be any primitives in the odin.blif file that weren't in the original architecture file.

#### Current Behaviour

Currently, after running the circuit and architecture files through ODIN, and abc, ODIN seems to be producing mult variables that weren't in the original architecture file. (mult_24_24_48)

#### Steps to Reproduce

1. Follow the Quickstart Guide
2. Run ODIN with:
    i. LU8PEEng.v
    ii. k6_frac_N10_frac_chain_mem32K_40nm.xml
(will generate LU8PEEng.odin.blif)
3. Run ABC (will generate LU8PEEng.abc_no_clock.blif)
4. Restore Clocks (will generate LU8PEEng.pre-vpr.blif)
5. Run vpr (this is where the error is generated)
Error Message: Failed to find matching architecture model found for mult_24_24_48

#### Debugging

I looked at the blif file generated after running ODIN to see if there is a "mult_24_24-48", and there was. I then checked the architecture file itself and there was a mult_36 and a mult_18 but no mult_24_24_48.

#### Context

I was trying to run vpr and test some gui features on a complex circuit.

#### Your Environment

* VTR revision used: 8
* Operating System and version: MacOS
* Machine: pchenry

kmurray, 2020-05-28:
@jeanlego Any ideas? Are there specific options ODIN requires to control synthesizing multipliers?

I would have expected ODIN to pick up from the architecture file what (if any) multipliers exist in the architecture, and only generate multiplier primitives if the architecture supports them.

jeanlego, 2020-05-29:
> I would have expected ODIN to pick up from the architecture file what (if any) multipliers exist in the architecture, and only generate multiplier primitives if the architecture supports them.

This is odd. It does, it reads the architecture to find multipliers available and their port sizes. 
mult_24_24-48 could only show up if it's specified in the architecture file. I'll dog into this more and come back to you on this :) 
Thanks for resporting this!

closed 2025-05-19 by github-actions[bot]