verilog-to-routing/vtr-verilog-to-routing #1325: Enhance VPR lookahead map to account for CHAN to IPIN costs

acomodi, 2020-06-01:
#### Expected Behaviour

Router correctly predicts the expected cost with correct lookahead map values

#### Current Behaviour

Router mispredicts expected costs to reach a target node from an intermediate one due to incorrectness of the lookahead map.

#### Possible Solution

This issue is related to the Series 7 fabrics.
The Lookahead map consists of expected costs that guide the router in finding the best path to a target node. In case the lookahead is incorrect, the router efficiency is drastically reduced, as it would incur in many backtracking steps due to wrong paths explorations.

Regarding SymbiFlow, a new kind of lookahead map has been used, that makes use of the so called connection boxes. These are additional data structures that hold information on the connection between IPINs and the routing nodes.
In particular, they expose the `site pin delays`, which are not taken into account in the current lookahead map. These delays should be added to the expected cost, as they differ for the various IPINs of a tile.
In fact, a LUT in a CLB tile does not have all the IPINs with the same delay values, but, depending on the IPIN selected, the delay changes.

This delay can be found on the edge connecting a CHAN to an IPIN of the RR graph.

Currently, the lookahead map has a mechanism to capture the delay between the SOURCE/OPIN nodes to the first CHAN[XY] nodes, and this delay is added to the expected cost.

A possible solution for the CHAN --> IPIN delay, is to adopt a similar mechanism:
1. An additional lookahead is to be generated, which captures the delay and congestion cost to get from the last CHAN[XY] node of the wire to the IPIN/SINK node
1. Modify the expected cost function to lookup to the target node, and, if it is a SINK node, apply the extra cost to the final expected cost. This would be equivalent to adding the site_pin_delay of the connection box map.

There is an additional potential issue with the Series 7 fabric which can be summarized as follows.
Before reaching the routing infrastructure, the OPIN/IPIN are connected to a series of two nodes with a single edge each between them. This can be visualized in the image below:
[image].
It is yet to understand whether this could be an issue in generating a correct lookahead map for the Series 7 family.

#### Context

Symbiflow and Series 7 RR graph and lookahead computation.

acomodi, 2020-06-01:
@vaughnbetz @kmurray @litghost FYI

acomodi, 2020-06-02:
While debugging the router lookahead map behaviour, I have noticed that the `get_expected_cost` function that queries the lookahead map returns null values and all the queries end up with an expected cost equal to 0.

```
E.g.:
Expanding to node 1361583 (RR node: 1361583 type: CHANY location: (99,72) <-> (99,73) track: 188 len: 1 longline: 0 seg_type: NW2BEG dir: BI_DIR capacity: 1)                                                                                                              
EXPECTED DELAY: 0.000000    EXPECTED CONGESTION: 0.000000, deltaX: 6, deltaY: 76    SEG INDEX: 46: TYPE: 1                                                                                                                                                                       
                                                                                                                                                                                                                                                                               
Lookahead from NW2BEG (RR node: 1361583 type: CHANY location: (99,72) <-> (99,73) track: 188 len: 1 longline: 0 seg_type: NW2BEG dir: BI_DIR capacity: 1) to BLK-TL-CLBLM_R[0].CLBLM_L_A3[0] (RR node: 2916545 type: SINK location: (105,150) class: 52 capacity: 1) is           finite, expected_cost = 0.000000, to->R_upstream = 0.001185  
```

I have double-checked the `f_wire_cost_map` values after the calculation of the lookahead and, for the same (CHAN index, SEG index, delta_x, delta_y), the delay value is non-0.

```
E.g. (from the NW2BEG segment at delta (6,76))
6,76:   9.78e-09
```

This is most likely the reason why the lookahead map performs very poorly on Series 7 rr graphs.
I'll need to further investigate why the expected cost is constantly equal to 0.

acomodi, 2020-06-02:
Turns out that there was an issue while printing these values, fixing the VTR_LOG strings showed up correct expected cost values, hence this is not an issue as previously thought.

acomodi, 2020-06-02:
By further investigating, I think there is something odd happening during the lookahead map-based routing. It seems that the total cost gets reduced by a factor of 100 as soon as the router gets to expand the routing nodes.

Below there are some examples of expansion of the same net, one is with using the lookahead map and the second is obtained by using the connection box lookahead map:

Lookahead map:
```
   8271 Computing src/opin lookahead took 0.02 seconds (max_rss 3247.8 MiB, delta_rss +0.0 MiB)                                                                                                                                                                                           
   8272   Adding node  2916578 to heap from init route tree with cost 7.06777e-10 (RR node: 2916578 type: SOURCE location: (105,150) class: 85 capacity: 1)                                                                                                                               
   8273   Routing to 2916543 as normal net (BB: 0,0 x 118,163)                                                                                                                                                                                                                            
   8274   Popping node 2916578 (cost: 7.06777e-10)                                                                                                                                                                                                                                        
   8275     Better cost to 2916578                                                                                                                                                                                                                                                        
   8276     New total cost: 7.06777e-10                                                                                                                                                                                                                                                   
   8277     New back cost: 0                                                                                                                                                                                                                                                              
   8278       Setting path costs for associated node 2916578 (from -1 edge 4294967295)                                                                                                                                                                                                    
   8279       Expanding node 2916578 edge 22302881 -> 2916673                                                                                                                                                                                                                             
   8280       Expanding to node 2916673 (RR node: 2916673 type: OPIN location: (105,150) pin: 85 pin_name: BLK-TL-CLBLM_R.CLBLM_L_BMUX[0] capacity: 1)                                                                                                                                    
   8281   Popping node 2916673 (cost: 7.08283e-10)                                                                                                                                                                                                                                        
   8282     Better cost to 2916673                                                                                                                                                                                                                                                        
   8283     New total cost: 7.08283e-10                                                                                                                                                                                                                                                   
   8284     New back cost: 1.50615e-12                                                                                                                                                                                                                                                    
   8285       Setting path costs for associated node 2916673 (from 2916578 edge 22302881)                                                                                                                                                                                                 
   8286       Expanding node 2916673 edge 22302975 -> 2916486                                                                                                                                                                                                                             
   8287       Expanding to node 2916486 (RR node: 2916486 type: CHANY location: (104,150) track: 269 len: 0 longline: 0 seg_type: OUTPINFEED dir: BI_DIR capacity: 1)                                                                                                                     
   8288   Popping node 2916486 (cost: 6.53669e-12)                                                                                                                                                                                                                                     [... cut ...]
 1501     New total cost: 7.86802e-10                                                                                                                                                                                                                                                     
 1502     New back cost: 4.51845e-12   
```

I believe that the difference in the two outputs is the fact that in the connection box map, the expected cost comes already with the CHAN -> IPIN delay added, which lets the total cost stabilize way sooner and allowing the router to proceed on the correct path.

acomodi, 2020-06-03:
It seems that adding the site_pin_delay (CHAN --> IPIN delay info from the switch), helps in improving the lookahead map with a reduction in the number of backtracking performed by the router.

The major issue seems to be the fact that, when expanding a node that has many edges (e.g. a node that is on the interconnect tile and from which there are several possible routes), the lookahead does not contain correct values for the expected delay.

Below are two outputs for the same net during routing.

Connection box Lookahead Map:
```
Loading routing resource graph took 5.82 seconds (max_rss 3917.8 MiB, delta_rss +2476.7 MiB)
  RR Graph Nodes: 2928865
  RR Graph Edges: 22354091
  Adding node  1114728 to heap from init route tree with cost 0 (RR node: 1114728 type: SOURCE location: (71,61) class: 84 capacity: 1)
  Routing to 1114717 as normal net (BB: 0,0 x 118,163)
  Popping node 1114728 (cost: 0)
    Better cost to 1114728
    New total cost: 0
    New back cost: 0
      Setting path costs for associated node 1114728 (from -1 edge 4294967295)
      Expanding node 1114728 edge 8855706 -> 1114823
      Expanding to node 1114823 (RR node: 1114823 type: OPIN location: (71,61) pin: 84 pin_name: BLK-TL-CLBLM_R.CLBLM_L_B[0] capacity: 1)
  Popping node 1114823 (cost: 1.50615e-12)
    Better cost to 1114823
    New total cost: 1.50615e-12
    New back cost: 1.50615e-12
      Setting path costs for associated node 1114823 (from 1114728 edge 8855706)
      Expanding node 1114823 edge 8855800 -> 1114627
      Expanding to node 1114627 (RR node: 1114627 type: CHANY location: (70,61) track: 178 len: 0 longline: 0 seg_type: OUTPINFEED dir: BI_DIR capacity: 1)
Lookahead cost: 6.59349e-10
  Popping node 1114627 (cost: 7.94231e-10)
    Better cost to 1114627
    New total cost: 7.94231e-10
    New back cost: 3.0123e-12
      Setting path costs for associated node 1114627 (from 1114823 edge 8855800)
      Expanding node 1114627 edge 8855673 -> 1114464
      Expanding to node 1114464 (RR node: 1114464 type: CHANY location: (70,61) track: 301 len: 0 longline: 0 seg_type: OUTPINFEED dir: BI_DIR capacity: 1)
Lookahead cost: 6.57843e-10
  Popping node 1114464 (cost: 7.9393e-10)
    Better cost to 1114464
    New total cost: 7.9393e-10
    New back cost: 4.51845e-12
      Setting path costs for associated node 1114464 (from 1114627 edge 8855673)
      Expanding node 1114464 edge 8853884 -> 462784
      Expanding to node 462784 (RR node: 462784 type: CHANX location: (59,61) <-> (70,61) track: 90 len: 11 longline: 0 seg_type: WW4BEG dir: BI_DIR capacity: 1)
Lookahead cost: 7.92178e-10
      Expanding node 1114464 edge 8853885 -> 1084917
      Expanding to node 1084917 (RR node: 1084917 type: CHANY location: (70,55) <-> (70,61) track: 136 len: 6 longline: 0 seg_type: NN6BEG dir: BI_DIR capacity: 1)
Lookahead cost: 7.9657e-10
      Expanding node 1114464 edge 8853886 -> 1112946
      Expanding to node 1112946 (RR node: 1112946 type: CHANY location: (69,61) <-> (69,65) track: 181 len: 4 longline: 0 seg_type: SW6BEG dir: BI_DIR capacity: 1)
Lookahead cost: 7.92652e-10
      Expanding node 1114464 edge 8853887 -> 1113050
      Expanding to node 1113050 (RR node: 1113050 type: CHANY location: (69,61) <-> (69,62) track: 332 len: 1 longline: 0 seg_type: SW2BEG dir: BI_DIR capacity: 1)
Lookahead cost: 6.62361e-10
      Expanding node 1114464 edge 8853888 -> 1113109
      Expanding to node 1113109 (RR node: 1113109 type: CHANY location: (69,61) track: 373 len: 0 longline: 0 seg_type: WL1BEG dir: BI_DIR capacity: 1)
Lookahead cost: 6.57843e-10
      Expanding node 1114464 edge 8853889 -> 1113113
      Expanding to node 1113113 (RR node: 1113113 type: CHANY location: (69,61) track: 369 len: 0 longline: 0 seg_type: WR1BEG dir: BI_DIR capacity: 1)
Lookahead cost: 6.62361e-10
      Expanding node 1114464 edge 8853890 -> 1113992
      Expanding to node 1113992 (RR node: 1113992 type: CHANY location: (69,60) <-> (69,61) track: 282 len: 1 longline: 0 seg_type: NW2BEG dir: BI_DIR capacity: 1)
Lookahead cost: 6.60855e-10
      Expanding node 1114464 edge 8853891 -> 1114014
      Expanding to node 1114014 (RR node: 1114014 type: CHANX location: (70,60) <-> (74,60) track: 82 len: 4 longline: 0 seg_type: NE6BEG dir: BI_DIR capacity: 1)
Lookahead cost: 8.0251e-10
      Expanding node 1114464 edge 8853892 -> 1114042
      Expanding to node 1114042 (RR node: 1114042 type: CHANX location: (70,60) <-> (73,60) track: 137 len: 3 longline: 0 seg_type: NE2BEG dir: BI_DIR capacity: 1)
Lookahead cost: 6.60855e-10
      Expanding node 1114464 edge 8853893 -> 1114184
      Expanding to node 1114184 (RR node: 1114184 type: CHANY location: (70,60) <-> (70,61) track: 129 len: 1 longline: 0 seg_type: NL1BEG dir: BI_DIR capacity: 1)
[... 75 lines cut ...]
Lookahead cost:   5.18e-10
      Expanding node 1114464 edge 8853893 -> 1114184
      Expanding to node 1114184 (RR node: 1114184 type: CHANY location: (70,60) <-> (70,61) track: 129 len: 1 longline: 0 seg_type: NL1BEG dir: BI_DIR capacity: 1)
Lookahead cost:   5.17e-10
      Expanding node 1114464 edge 8853894 -> 1114188
      Expanding to node 1114188 (RR node: 1114188 type: CHANY location: (70,60) <-> (70,61) track: 124 len: 1 longline: 0 seg_type: NR1BEG dir: BI_DIR capacity: 1)
Lookahead cost:   5.17e-10
      Expanding node 1114464 edge 8853895 -> 1114324
[... cut ...]
I think that this is due to the fact that the `router_cost_map` is built in a way that it selects the `SMALLEST` value for each delta_x, delta_y locations, hence the almost constant values when expanding the nodes. IMO this is related to the fact that, the routing cost map should have a value at least for each segment type instead of only having a x, y coordinate systems. This way, the higher values do not get trimmed off, allowing to have a more diverse and correct lookaheads for a specific location.

@kmurray Does this make sense?

kmurray, 2020-06-03:
> IMO this is related to the fact that, the routing cost map should have a value at least for each segment type instead of only having a x, y coordinate systems

I believe that is the case the the map lookahead, which is indexed in 4 dimensions as:
[orientation][segment_type][dx][dy]

See for instance: 
- https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/router_lookahead_map.h#L36-L37
- https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/router_lookahead_map.cpp#L220

> @kmurray Does this make sense?

The rest is somewhat hard to follow since I don't understand what the different segment types arein this architecture. Perhaps you could produce some kinds of plots which capture the effects you are seeing? And how they related to the spatial layout of the RR graph? 

Also it looks like all your segment types are bi-direction (BIDIR)? Are they really bi-direction? Usually modern FPGAs use uni-directional routing.

acomodi, 2020-06-08:
@kmurray I will try to get some diagrams prepared to easily get an overall view on how the RR graph looks like, to get a better understanding on what is going on.

> Also it looks like all your segment types are bi-direction (BIDIR)? Are they really bi-direction? Usually modern FPGAs use uni-directional routing.

Regarding BI-DIR segment types I am unsure whether we can move them to be unidirectional, I would need to test this, probably @litghost could provide some more insight on this matter.

By analysing the current behaviour and the connection box one, I think that the poor results obtained so far are due to the following issues:

- There are quite a lot of segment types that do not have entries in the cost map, or at least they have very sparse cost entries throughout the grid. Probably, a better sampling method to capture more costs is the right solution. On a side note, assigning a high-cost value for zero-cost entries seems to help the router in finding the correct path.

- Connection box-based lookahead makes use of penalties for holes in the cost map. The penalty is calculated by copying the closest nearby cost of an invalid cost entry, prior to a penalty cost addition. For the lookahaed map, only a copy of the entry is added to fill the hole. The problem though is also that in the lookahead map, there appear to be many holes as well, after the filling function is being called, suggesting that it would be required to enhance also this part of the lookahead generation.

- As already previously discussed, the CHAN --> IPIN delay should be captured as well. The best way to do this would be to expand wire types samples for a constant low value N so that there are a maximum of N expansions, that would lead to the IPIN locations. Than, similarly to the SRC/OPIN --> CHAN lookahead, an additional lookahead would be produced to capture the CHAN --> IPIN delays, and this would happen in constant time, allowing the process.

litghost, 2020-06-08:
> * There are quite a lot of segment types that do not have entries in the cost map, or at least they have very sparse cost entries throughout the grid. Probably, a better sampling method to capture [...]

@HackerFoo spent a good deal of time developing the current connection box lookahead sampling method.  This method should be suitable for the lookahead map, and should help and/or solve missing entries.  That code likely needs to be made into a library and/or function that both the connection box lookahead and map lookahead can use as a sampling pattern.

closed 2025-05-20 by github-actions[bot]