verilog-to-routing/vtr-verilog-to-routing #1328: Automatic API documentation

rw1nkler, 2020-06-02:
#### Proposed Behaviour

Currently, there is no explicit standardization in the code comments, which makes it impossible to include the API description in the main documentation.  Converting the current code comments to the Doxygen style will allow for generating automatic API documentation. Doxygen description can be used standalone or can be imported to Sphinx using the [Breathe](https://breathe.readthedocs.io/en/latest/) plugin.

Being able to point to specific functions and list the most important classes will improve the main documentation.

#### Current Behaviour

Automatic API documentation cannot be generated.

#### Possible Solution

- Convert the code comments either automatically or manually.
- Provide the description of the new comment style

#### Context

VTR is an important element of the [SymbiFlow](https://github.com/SymbiFlow) toolchain. Every improvement in the project documentation is helpful and might be useful for people involved in both projects.

rw1nkler, 2020-06-02:
I'm interested in helping with this task

kmurray, 2020-06-02:
I agree this would be a good upgrade. I've looked briefly at Breathe before and it seems like the right way to integrate with sphinx and readthedocs.

There are effectively two layers of documentation in VTR: user facing (i.e. those using and running the tools), and developer facing (those making code changes). I think we want to keep that division pretty clean. Most users won't be interested in the internal documentation details (and we don't want to overwhelm them). But having browse-able documentation will be helpful for developers.

Another important consideration is that there are *no stable APIs* within the various VTR tools (i.e. VPR). We'll want to make sure that is very clear, as some will take API documentation as the declaration of a stable interface otherwise.

I'd suggest the following general plan of attack:

1. Integrate breath/doxygen support into VTR combined with read the docs. (i.e. automatically run doxygen and link with breath on read the docs)

2. Gradually bring in documentation for key components, likely focusing on key data structures (VPR contexts, Netlists, RR graph). Likely will just be function/class definitions as code comments won't be in appropriate formats yet.

3. Gradually re-work code comments to produce better documentation.

vaughnbetz, 2020-06-03:
Agreed.  This is a good idea, and a practical plan.  Documenting the key data structures you list is the best place to start, and the most important.

vaughnbetz, 2025-05-05:
Closing as this plan is well underway now.

closed 2025-05-05 by vaughnbetz