verilog-to-routing/vtr-verilog-to-routing #1333: Better messages for vpr graphics support missing packages / not supported

vaughnbetz, 2020-06-03:
We disable graphics in vpr if we don't find the necessary packages. But we can make the messages better / more helpful.

1. cmake currently gives easygl error messages (at least in the 8.0 release) if it can't find the x11 development headers. It should only be giving ezgl error messages, as easygl is obseleted.  I can't find the lines that print this in the master (trunk) so perhaps it's only in 8.0 release, or perhaps I'm missing it.  See the attached screenshot submitted by a user.

2. If graphics aren't build but --disp on is specified (asking for graphics) we should give a warning that says graphics aren't compiled in, and are disabled. 

[image]

kmurray, 2020-06-03:
> 1. cmake currently gives easygl error messages (at least in the 8.0 release) if it can't find the x11 development headers. It should only be giving ezgl error messages, as easygl is obseleted.  I [...]

@vaughnbetz I believe VTR 8 branched before we upgraded from EasyGL to EZGL.

>     2\. If graphics aren't build but --disp on is specified (asking for graphics) we should give a warning that says graphics aren't compiled in, and are disabled.

Yes this would be an excellent improvement.

MahshadJ, 2020-06-03:
@vaughnbetz which flag enables and disables graphics? is it VPR_USE_EZGL or EZGL_BUILD_EXAMPLES?

kmurray, 2020-06-03:
For (2), I believe we'd want to print out a messaged based on the 'NO_GRAPHICS' define (see vpr/src/draw.cpp). Specifically, we should probably put it in the `#else` section of `init_graphics_state()`, and print the warning provided any of the --save_graphics  on/--graphics_commands/--disp on  options have been specified (i.e. informing the user that they won't see graphics since VPR wasn't compiled with graphics support.

You can see something similar in the debug logging code [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/route_timing.cpp#L1946-L1948). Although we'd want to provide a more detailed and actionable message, since this is something we expect non-expert users to encounter.

MahshadJ, 2020-06-03:
Yes, I know I've already done that, I just wanted to test it. So I wanted to disable graphics and run --disp on to see if it does give me the warning.

vaughnbetz, 2020-06-03:
VPR_USE_EZGL not being set looks to be the control variable for running without graphics libraries.  So I'd test that one.  If it's easy to also test NO_GRAPHICS being set I'd test that too (should be set automatically when VPR_USE_EZGL isn't set).

closed 2025-05-20 by github-actions[bot]