verilog-to-routing/vtr-verilog-to-routing #1334: Memory leak when packing disable code causes failures

kmurray, 2020-06-03:
#### Expected Behaviour

Code should be free of memory errors.

#### Current Behaviour

When with sanitizers enabled there is an [error](https://source.cloud.google.com/results/invocations/a90d3085-ffd1-42fe-a51d-907094009e6f/targets/foss-fpga-tools%2Fverilog-to-routing%2Fupstream%2Fcontinuous%2Fstrong_sanitized/log):
```
pack_disable:             k6_frac_N10_40nm_disable_packing.xml/mult_5x6.blif/common                                      failed: expected to fail with 'exited with return code 1' but was 'exited with return code 23' (took 1.74 seconds)
```
where 23 usually indicates a memory leak detected by ASAN.

The resulting [log file](
https://storage.cloud.google.com/vtr-verilog-to-routing/artifacts/prod/foss-fpga-tools/verilog-to-routing/upstream/continuous/strong_sanitized/30/20200603-055813/vtr_flow/tasks/regression_tests/vtr_reg_strong/strong_pack_disable/latest/k6_frac_N10_40nm_disable_packing.xml/mult_5x6.blif/common/vpr.out
) shows:
```
# Load Timing Constraints took 0.00 seconds (max_rss 228.9 MiB, delta_rss +0.0 MiB)
Error 1: 
Type: Packing
File: /tmpfs/src/github/vtr-verilog-to-routing/vpr/src/pack/prepack.cpp
Line: 1205
Message: Failed to find any location to pack primitive of type '.names' in architecture

=================================================================
==16776==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 80 byte(s) in 1 object(s) allocated from:
    #0 0x7efc22bcb532 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x99532)
    #1 0x1873396 in alloc_and_load_pack_molecules(t_pack_patterns*, std::multimap<vtr::StrongId<atom_block_id_tag, int, -1>, t_pack_molecule*, std::less<vtr::StrongId<atom_block_id_tag, int, -1> >, std::allocator<std::pair<vtr::StrongId<atom_block_id_tag, int, -1> const, t_pack_molecule*> > >&, std::unordered_map<vtr::StrongId<atom_block_id_tag, int, -1>, t_pb_graph_node*, std::hash<vtr::StrongId<atom_block_id_tag, int, -1> >, std::equal_to<vtr::StrongId<atom_block_id_tag, int, -1> >, std::allocator<std::pair<vtr::StrongId<atom_block_id_tag, int, -1> const, t_pb_graph_node*> > >&, int) /tmpfs/src/github/vtr-verilog-to-routing/vpr/src/pack/prepack.cpp:857
    #2 0x1374266 in try_pack(t_packer_opts*, t_analysis_opts const*, t_arch const*, t_model const*, t_model const*, float, std::vector<t_lb_type_rr_node, std::allocator<t_lb_type_rr_node> >*) [clone .constprop.1411] /tmpfs/src/github/vtr-verilog-to-routing/vpr/src/pack/pack.cpp:99
    #3 0x497892 in vpr_pack /tmpfs/src/github/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:568
    #4 0x497892 in vpr_pack_flow /tmpfs/src/github/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:487
    #5 0x497892 in vpr_flow /tmpfs/src/github/vtr-verilog-to-routing/vpr/src/base/vpr_api.cpp:349
    #6 0x497892 in main /tmpfs/src/github/vtr-verilog-to-routing/vpr/src/main.cpp:62
    #7 0x7efc1f47982f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #8 0x4a5828 in _start (/tmpfs/src/github/vtr-verilog-to-routing/build/vpr/vpr+0x4a5828)
```

#### Possible Solution

We should ensure all memory is freed correctly. This will involve tracking down where the relevant memory is allocated and ensuring it gets freed in case an error/exception is thrown so as not to leak. 

I suspect this isn't directly related to the packing disable code (although I haven't dug into it to verify). Likely this is a latent issue in the packer, as the packing code wasn't originally written to clean-up its memory when it fails internally (as it does in this case).

#### Steps to Reproduce

1. Build VTR with VTR_ENABLE_SANITIZE=ON and VTR_ASSERT_LEVEL=3
2. Run strong_pack_disable test

#### Context

We test for memory issues in a special build (strong snaitizied). Any failure should be fixed.

#### Your Environment

Kokoro

kmurray, 2020-06-03:
@tangxifan Could you take a look?

tangxifan, 2020-06-03:
Sure. I should be able to patch it this week.

tangxifan, 2020-06-08:
Hello, I have patched in the [pull request](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1344)

kmurray, 2020-06-08:
Fixed with #1344.

closed 2020-06-08 by kmurray