verilog-to-routing/vtr-verilog-to-routing #1338: Nand gate 1 bit error

emacdo12, 2020-06-03:
#### Expected Behaviour

If a nand gate is called upon, it should produce the appropriate output based on it's inputs.

#### Current Behaviour

An error occurs while parsing.

#### Possible Solution

#### Steps to Reproduce

This bug is replicated in a PR I made that includes benchmarks for gates (binary). Specifically, nand_wire.vh. 

The error is:
```
 Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

Using Lut input width of: -1
Verilog: nand_wire.vh
--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file regression_test/benchmark/verilog/binary/nand_wire.vh to parse list
Adding file "wire_test.v" to parse list
Warning::PARSE_TO_AST wire_test.v:1 error in parsing: (syntax error, unexpected vAND, expecting vSYMBOL_ID or '#')
Optimizing module by AST based optimizations
Error::PARSE_TO_AST Parser found (1) errors in your syntax, exiting
/home/emacdo12/workspace/vtr-verilog-to-routing/ODIN_II/SRC/odin_error.cpp:27: void verify_delayed_error(odin_error): Fatal error
Aborted (core dumped)
``` 

#### Context

Just a little bug that needs to be fixed.

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

jeanlego, 2020-07-06:
could you link to the file here to be able to reproduce

jeanlego, 2020-07-16:
this is actually a preprocessor error. closing

closed 2020-07-16 by jeanlego