verilog-to-routing/vtr-verilog-to-routing #1357: Markdown files fail to link into documentation

kmurray, 2020-06-11:
#1354 seems to have introduced some build issues with the documentation. In particular where we were linking to .md files in the docs.

For instance, the developer guide (from README.developers.md) no longer appears in the main docs at: https://docs.verilogtorouting.org/en/latest/dev/

Before:
[image]

After:
[image]

With this PR I also see a number of (likely related) warnings when building on either my machine or RTD:
```
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/index.rst:43: WARNING: toctree contains reference to document 'CHANGELOG' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/index.rst:43: WARNING: toctree contains reference to document 'CHANGELOG' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/index.rst:43: WARNING: toctree contains reference to document 'CHANGELOG' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/index.rst:43: WARNING: toctree contains reference to document 'CHANGELOG' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/arch/reference.rst:611: WARNING: undefined label: arch_fc (if the link has no caption the label must precede a section header)
WARNING: dvipng command 'dvipng' cannot be run (needed for math display), check the imgmath_dvipng setting
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/index.rst:43: WARNING: toctree contains reference to document 'CHANGELOG' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/index.rst:43: WARNING: toctree contains reference to document 'CHANGELOG' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/index.rst:43: WARNING: toctree contains reference to document 'CHANGELOG' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/index.rst:43: WARNING: toctree contains reference to document 'CHANGELOG' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/building/index.rst:4: WARNING: toctree contains reference to document 'building/building' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/dev/contributing/index.rst:3: WARNING: toctree contains reference to document 'dev/contributing/contributing' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/dev/index.rst:6: WARNING: toctree contains reference to document 'dev/developing' that doesn't have a title: no link will be generated
/home/kmurray/trees/vtr/doc/src/dev/index.rst:6: WARNING: toctree contains reference to document 'dev/license' that doesn't have a title: no link will be generated
```

Not sure what the issue is, but my guess is switching to python3/sphinx3. We rely on https://github.com/SymbiFlow/sphinxcontrib-markdown-symlinks to link in the .md docs into sphinx developed by @acomodi. Perhaps it doesn't like/work with python/sphinx3?

### To reproduce:

Before:
git checkout 2a597e01f
cd doc
pip install -r requirements.txt

After:
git checkout 48c2c4f0d5
pip3 install -r requirements.txt
make html

kmurray, 2020-06-11:
@rw1nkler can you look into this?

kmurray, 2020-06-11:
@vaughnbetz @sfkhalid FYI

vaughnbetz, 2020-06-11:
Thanks Kevin.  @rw1nkler: this has resulted in a lot of inaccessible documentation and broken links so it's an important one to resolve quickly.

mithro, 2020-06-11:
@acomodi

rw1nkler, 2020-06-11:
@kmurray @vaughnbetz The fix is available in the https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1358

closed 2020-06-11 by kmurray