verilog-to-routing/vtr-verilog-to-routing #1369: ODIN_II: overriding paramters doesn't work

emacdo12, 2020-06-16:
#### Expected Behaviour

If the proper syntax is in place, it is possible to override the parameters in a seperate model with new values. There are two ways to do it, defparam and model #(.paramter(new_parameter)) model (ports[]); I have included test cases for both.
#### Current Behaviour

Odin_II doesn't do the overried (defparam) or can't handle it and aborts.
#### Possible Solution

#### Steps to Reproduce

All these cases have been verified through quartus.

``` 
module simple_op(in1,in2,in3,out1,out2);
    parameter msb1 = 3;
    parameter msb2 = 2;
    parameter lsb  = 0;
    
    input  [msb1:lsb] in1;
    input  [msb2:lsb] in2;
    input  [msb2:lsb] in3;
    output [msb1:lsb] out1;
    output [msb2:lsb] out2;

    assg #(.msb(msb1),.lsb(lsb)) assg(in1,out1);

    addi #(.msb(msb2),.lsb(lsb)) addi(in2,in3,out2);
    
endmodule 

module assg(in,out);
    parameter msb = 8;
    parameter lsb = 1;

    input  [msb:lsb] in;
    output [msb:lsb] out;

    assign out = in;
endmodule

module addi(in1,in2,out);
    parameter msb = 4;
    parameter lsb = 2;
    
    input  [msb:lsb] in1;
    input  [msb:lsb] in2;
    output [msb:lsb] out;

    assign out = in1 + in2;
endmodule
```
```
GLOBAL_SIM_BASE_CLK in1 in2 in3
0 0000 000 000
1 0001 101 010
0 0010 110 001
1 0011 001 101
0 1101 010 100
1 1001 100 001
```
```
out1 out2
0000 000
0001 111
0010 111
0011 110
1101 110
1001 101
```
```
✔ {15:20}[]~/workspace/vtr-verilog-to-routing/ODIN_II:[ML/DNN ✭|… 30⚑ 1] ➭ ./odin_II -V regression_test/benchmark/verilog/ML/parameter_override.v -t regression_test/benchmark/verilog/ML/parameter_override_input -T regression_test/benchmark/verilog/ML/parameter_override_output 
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

Using Lut input width of: -1
Verilog: parameter_override.v
--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file regression_test/benchmark/verilog/ML/parameter_override.v to parse list
Optimizing module by AST based optimizations
Converting AST into a Netlist. Note this netlist can be viewed using GraphViz (see documentation)
==========================
Detected Top Level Module: 	simple_op
==========================
Segmentation fault (core dumped)
```
defparam: 

```
module simple_op(in1,in2,out1,out2);
	parameter width1 = 2;
	parameter width2 = 4;
	
	input  [width1-1:0] in1;
	input  [width2-1:0] in2;
	output [width1-1:0] out1;
	output [width2-1:0] out2;
	
	assg m1 (in1,out1);
	assg m2 (in2,out2);
	
endmodule
	
module assg(in,out);
	parameter width = 3;
		
	input  [width-1:0] in;
	output [width-1:0] out;
		
	assign out = in;
endmodule 
	
module params;
    defparam
		simple_op.m1.width = 2,
		simple_op.m2.width = 4;
endmodule 
```
```GLOBAL_SIM_BASE_CLK in1 in2
0 00 0000
1 01 1100
0 10 1010
1 11 0110
0 10 1001
```
```
out1 out2
00 0000
01 1100
10 1010
11 0110
10 1001
```
```
Segmentation fault (core dumped)
✘ {15:25}[]~/workspace/vtr-verilog-to-routing/ODIN_II:[ML/DNN ✭|… 30⚑ 1] ➭ ./odin_II -V regression_test/benchmark/verilog/ML/defparam.v -t regression_test/benchmark/verilog/ML/defparam_input -T regression_test/benchmark/verilog/ML/defparam_output 
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

Using Lut input width of: -1
Verilog: defparam.v
--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file regression_test/benchmark/verilog/ML/defparam.v to parse list
Warning::AST defparam.v:29 there are no ports for the module (params)
	all logic will be dropped since it is not driving an output
29: endmodule 
Optimizing module by AST based optimizations
Converting AST into a Netlist. Note this netlist can be viewed using GraphViz (see documentation)
==========================
Detected Top Level Module: 	simple_op
==========================
Performing Optimizations of the Netlist
Performing Partial Map to target device
Outputting the netlist to the specified output format
Warning::NETLIST defparam.v:9 This output is undriven (simple_op^out2~3) and will be removed
9: 	output [width2-1:0] out2;
Warning::NETLIST Net simple_op^out2~3 driving node simple_op^out2~3 is itself undriven.
Successful High-level synthesis by Odin
	Blif file available at default_out.blif

	==== Stats ====
Number of <FF_NODE> node:                 0
Number of <INPUT_NODE> node:              5
Number of <OUTPUT_NODE> node:             6
Number of <CLOCK_NODE> node:              0
Number of <GENERIC> node:                 0
Total estimated number of lut:            0
Total number of node:                     0
Longest path:                             2
Average path:                             2

Elaboration Time: 1.3ms
--------------------------------------------------------------------

Reading top level module
Reading blif netlist...
100%|==================================================>| Total time: 0.6ms    
-------------------------------------
Netlist Simulation Begin
Simulation starts 
Beginning simulation. Output_files located @: .
Simulating 5 existing vectors from "regression_test/benchmark/verilog/ML/defparam_input".

default_out.blif:
  Nodes:           21
  Connections:     11
  Threads:         1
  Degree:          0.52
  Stages:          3
  Nodes/thread:    21(100.00%)

100%|==================================================>| Total time: 0.3ms    

Warning::OUTPUT_BLIF Vector 1 mismatch:
	0X1 0X4  in output_vectors
	01 1100 in regression_test/benchmark/verilog/ML/defparam_output
Warning::OUTPUT_BLIF Vector 2 mismatch:
	0X2 0X2  in output_vectors
	10 1010 in regression_test/benchmark/verilog/ML/defparam_output
Warning::OUTPUT_BLIF Vector 4 mismatch:
	0X2 0X1  in output_vectors
	10 1001 in regression_test/benchmark/verilog/ML/defparam_output
Error::OUTPUT_BLIF Vector files differ.
/home/emacdo12/workspace/vtr-verilog-to-routing/ODIN_II/SRC/simulate_blif.cpp:223: void simulate_netlist(netlist_t*): Fatal error
Aborted (core dumped)
```
In this last test case you can see that where the output vector differs is the most significant bit indicating that the defparam didn't motify the parameters of the assg model. If it had, the msb would have been passed through. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

jeanlego, 2020-07-16:
this is odd, we have test that work for this form, only hierarchical overrides are not yet supported

sdamghan, 2021-02-02:
@hmoein87 Would you please have a look at this?

sdamghan, 2021-05-22:
@HMoein87 :  @ArashAhmadian is one of UoT students working on improving the regression test. He has recently faced the same issue you have been working on last semester. 
In [64-16bit-fixed-JACM.v.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6526667/64-16bit-fixed-JACM.v.txt), Odin_II throws an error at line 1057. When we use a constant value, like 10, instead of depth at line 1057, Odin_II continues working properly. Please provide an appropriate solution to this aspect of the issue for him.
Moreover, please update this branch and create a new PR in the main VTR repository so that we shall merge it.

HMoein87, 2021-06-22:
seems the code is not complete. it has some declaration errors.

closed 2025-05-23 by github-actions[bot]