verilog-to-routing/vtr-verilog-to-routing #1377: Flag "--enable_timing_computations" removed

litghost, 2020-06-24:
#### Expected Behaviour

`--enable_timing_computations option` enabled disabling of timing calculations.

#### Current Behaviour

Flag is now removed.

#### Possible Solution

#### Steps to Reproduce

1. 
2. 

#### Context

`--enable_timing_computations` option was removed in https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/1b2a445308a822d78156e0627497dde9315510a7

This flag was being used to create dummy rr graphs that could be patched later.  These dummy graphs sometimes had non-sense timing values, which resulted in the tatum throwing exceptions.

litghost, 2020-06-24:
It is possible that `--enable_timing_computations off --place_algorithm bounding_box` is now equivalent to simply `--place_algorithm bounding_box`, testing that now.

kmurray, 2020-06-29:
> It is possible that `--enable_timing_computations off --place_algorithm bounding_box` is now equivalent to simply `--place_algorithm bounding_box`, testing that now.

That seems like a reasonable behavior.

One thing which may come up is whether `--place_algorithm bounding_box` constructs an RR graph for placement. I think it may not (since it isn't strictly needed for non-timing driven placement, since no delta delays need to be calculated).
If that is the case, we could just unconditionally build the RR graph before placement (regardless of --place_algorithm), the non-timing driven placer code path isn't our primary use case at this point so the 'extra' run-time of building/loading the RR graph shouldn't be an issue.

closed 2025-05-23 by github-actions[bot]