verilog-to-routing/vtr-verilog-to-routing #1378: Placer crashing

litghost, 2020-06-24:
#### Expected Behaviour

Placer should not crash when running.

#### Current Behaviour

```
/tmpfs/src/github/vtr-verilog-to-routing/vpr/src/place/timing_place.h:196 PlacerTimingCosts: Assertion 'num_connections == 0 || num_level_before_leaves < num_connections' failed (Level before should have fewer nodes than connections (to ensure using the smallest binary tree)).
```

#### Possible Solution

Still investigating, recent changes in the placer (near https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/7f39b64da4a3c6f301d5bd4bb3ab92f45f9bfb72) appear to be the changes that caused the issue.

#### Steps to Reproduce

Working on reduced tested case, TBD.

1. 
2. 

#### Context

At a minimum the symbiflow-arch-defs ice40 arch started generating this exception when updating to the lastest VTR master.

vaughnbetz, 2020-06-25:
Bill, can you take a look?  It appears to be in the new code that adds up timing costs in a hierarchical way to minimize round off.  Keith is working on a small test case, but says making a circuit with a single input connected to a single output appears to cause the problem.  So maybe there is a degenerate case where very small circuits break this code (maybe an out of bounds error for such a case?).

vaughnbetz, 2020-06-25:
So this may not be debuggable without more info, but if you inspect the code perhaps you'll see it.
Also may indicate a missing test case (extremely small, one net circuit).

litghost, 2020-06-25:
When I tried to replicate this outside of CI, the issue did not appear.  I'm confused about why, but for now I'm going to close this issue.  I will re-open once I get a replication :/

litghost, 2020-06-25:
I've been able to replicate this issue!  GCC 5 and GCC 6 generate the exception, and clang-7 does not!  This hints at possibly some kind of undefined behavior being present?

litghost, 2020-06-25:
Ya, I believe the assertion fires when `num_connections = 1`.  Let me add a vtr_reg_basic wire test, and see if it I can get a replication.

Bill-hbrhbr, 2020-06-25:
It seems to me that this for loop in timing_place.cpp:

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place.h#L165-L180 

is skipped entirely when there is only one unignored net within the netlist. This problem disappears when there are two or more unignored nets.

For example, the following code generates the error, even if there are 3 nets in total (2 of them have constant drivers)

	module ex(input w1, w2, w3, output reg z1, z2, z3);
	assign z1 = 1;
	assign z2 = w3;
	assign z3 = 1;
	endmodule

Still trying to figure out the reason

Bill-hbrhbr, 2020-06-26:
Sorry about my previous comment on the for loop. I misused VTR_LOG. The looped isn't skipped.
I think the error occurred due to this function `num_nodes_in_level`:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/place/timing_place.h#L374-L377
I think it should be 
`return (1 << (ilevel));`
for this heap data structure. It seems like root has index 0.
I think **num_nodes_up_to_level** is correct.

closed 2020-06-30 by Bill-hbrhbr