verilog-to-routing/vtr-verilog-to-routing #1387: Calculation of RR node congestion cost

Bill-hbrhbr, 2020-06-30:
#### Description
The RR node congestion cost should be calculated as a product of individual cost components:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/route/route_common.cpp#L730-#L732

however, in draw.cpp, the cost is calculated as a sum:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/draw/draw.cpp#L1202-#L1204

Not sure if this is a bug or an intended modification to make the graphics look nicer. We should add comments if it is the latter case.

@kmurray @vaughnbetz

vaughnbetz, 2020-06-30:
The route_common code is for sure correct.  I suspect the draw code is just a bug.  Just to double check: Kevin, was this intentional to make drawing look nicer?

kmurray, 2020-07-01:
>  I suspect the draw code is just a bug.

Yep that looks like a bug. To fix it, I'd suggest using `get_single_rr_cong_cost()` in draw.cpp so we have only a single definition of how costs are calculated.

Bill-hbrhbr, 2020-07-01:
Ok, I will update this in a future pull request.

vaughnbetz, 2025-05-25:
I'm pretty sure @Bill-hbrhbr fixed this.

closed 2025-05-25 by vaughnbetz