verilog-to-routing/vtr-verilog-to-routing #1394: Improve block selection / colouring in VPR GUI

vaughnbetz, 2020-07-03:
Block selection via the search dialog currently only accepts clustered block names, and clustered block IDs.  
1. Ideally it would also accept primitive names (atom netlist names) and highlight the block that contains that primitive.  
2. Longer term, it would be good if we could also search for/highlight individual primitives using the intra-block visualization feature.

#### Proposed Behaviour

1. We could search through the atom netlist looking for names as well, and highlight the cluster containing that atom if found. This shouldn't be too hard.  (Action for Mahshad).

2. When you refactor the UI Mahshad, I think it would be good to allow a user to set the colour of a block, rather than just select it. This would involve storing the color in draw_state.

3. This code can draw the internals of a cluster. 
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/draw/intra_logic_block.cpp
  
In particular, this routine shows flylines between atoms (primitives), so ideally selecting atoms would highlight only those atoms.  This is a bigger change that would involve extracting / making usable some of the code in this file, and connecting it to an enhanced selection dialog.
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/f13e8bf72fb309f97efac47f6e046294956ab9d6/vpr/src/draw/intra_logic_block.cpp#L463

kmurray, 2020-07-03:
Useful information for this is found in `g_vpr_ctx.atom().lookup`.

Specifically, the function [atom_clb()](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/base/atom_lookup.h#L65) will return the the CLB block associated with a specific atom block.

vaughnbetz, 2025-05-25:
This was completed.