verilog-to-routing/vtr-verilog-to-routing #1409: Odin: While keyword not supported

emacdo12, 2020-07-07:
#### Expected Behaviour

While should be supported. Underneath the supported keywords in the flex it's labelled.

#### Current Behaviour

There's an error that while statements aren't supported.

#### Possible Solution

#### Steps to Reproduce

```
`define WIDTH 3

module simple_op(out);
    
  output reg [`WIDTH-1:0] out;
	integer i;

  always @(*) begin
		i = 0;
    while (i <`WIDTH) begin
      out[i] = 1;
      i = i+1;
    end
  end

endmodule 
```

```✘ {19:15}[]~/workspace/vtr-verilog-to-routing/ODIN_II:[keywords|⚑ 1] ➭ ./odin_II -V regression_test/benchmark/verilog/keywords/while/while_loop.v
--------------------------------------------------------------------
Welcome to ODIN II version 0.1 - the better High level synthesis tools++ targetting FPGAs (mainly VPR)
Email: jamieson.peter@gmail.com and ken@unb.ca for support issues

Using Lut input width of: -1
Verilog: while_loop.v
--------------------------------------------------------------------
High-level synthesis Begin
Parser starting - we'll create an abstract syntax tree. Note this tree can be viewed using Grap Viz (see documentation)
Adding file regression_test/benchmark/verilog/keywords/while/while_loop.v to parse list
Error::AST while_loop.v:13 While statements are NOT supported
13:     end
/home/emacdo12/workspace/vtr-verilog-to-routing/ODIN_II/SRC/parse_making_ast.cpp:1039: ast_node_t* newWhile(ast_node_t*, ast_node_t*, int): Fatal error
Aborted (core dumped)
```

#### Context

Error arrose while creating benchmarks for keywords.

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

sdamghan, 2021-05-26:
@nanjekyejoannah Would you please provide an update on this issue?

nanjekyejoannah, 2021-05-27:
@sdamghan, I have extensions to the work I presented during the seminar,  that I am working on in time for publication. Once I have all results for this extension work, I will avail the PR. Thanks for the reminder.